ouster-lidar/ouster-ros #171: Some questions used in ros1

Sisyphus-neo, 2023-07-22:
Hello, at present, I have completed the installation of the ros1 driver system according to the documents and completed the visualization, but I do not know how to use other functions of sdk, such as some functions of the cli client to deal with point cloud, how to use some functions of sdk? And the other thing is if I want to do something with the point cloud and visualize it which file should I do it in? In terms of data storage, the bag file recorded on the tutorial does not seem to be a pointcloud2 message type, I do not know how to deal with it, and it is currently saved directly as a pcd file.

Samahu, 2023-07-22:
@Sisyphus-neo when using the ros driver, if you connect to a sensor or replay a recorded file the driver would automatically display the point cloud in RVIZ. You should see a window like this:
[image]
RVIZ here is rendering the topic `/ouster/points` which is of the type PointCloud2.

If you use the supplied `recored.launch` launch file to record a bag file as described in this section https://github.com/ouster-lidar/ouster-ros#recording-mode this produces a bag file which contains the raw packets sent by the sensor. If your intention however to have PointCloud2 messages saved into the bag file, I don't have an easy command for it but you could easily achieve that by directly invoking the rosbag commnad while the driver is running as follows :
```bash
rosbag record -O output.bag /ouster/points
```

Sisyphus-neo, 2023-07-23:
Ok, if I want to do some preliminary filtering on the point cloud, can I directly modify a file in the ros1 driver? Because I ran another file to try to filter the radar data in real time, but there was an error with no point cloud input.
[image]

Sisyphus-neo, 2023-07-23:
/#include<ros/ros.h>
#include<pcl/io/pcd_io.h>
#include<pcl/point_cloud.h>
#include<pcl/filters/passthrough.h>  
#include<pcl/filters/voxel_grid.h> 
#include<pcl_conversions/pcl_conversions.h>
#include<pcl/filters/statistical_outlier_removal.h>
#include<sensor_msgs/PointCloud2.h>

typedef pcl::PointXYZI PointType;   //可自己修改点云格式
bool received_cloud = false;
pcl::PointCloud<PointType>::Ptr pcl_cloud(new pcl::PointCloud<PointType>);
pcl::PointCloud<PointType>::Ptr new_cloud(new pcl::PointCloud<PointType>);

void pointCloudCallback(const sensor_msgs::PointCloud2ConstPtr& cloud_msg)
{
    // 将 sensor_msgs::PointCloud2 转换为 pcl::PointCloud<PointType>::Ptr 格式
    
    pcl::fromROSMsg(*cloud_msg, *pcl_cloud);
}

void newCloudCallback(const sensor_msgs::PointCloud2ConstPtr& cloud_msg)
{
    // 将接收到的 sensor_msgs::PointCloud2 转换为 pcl::PointCloud<PointType>::Ptr 格式
   
    pcl::fromROSMsg(*cloud_msg, *new_cloud);

    // 将布尔变量设置为 true，表示接收到了新的点云数据
    static bool received_cloud = true;
}

ros::Publisher read_pcd_pub;
ros::Publisher pointCloud_pub;

std::string pcd_doc_path;
std::string output_frame_id;
std::string pointCloud_pubTopic;
std::string output_pcd_path;

double pass_x_min, pass_x_max, pass_y_min, pass_y_max, pass_z_min, pass_z_max;  
double voxel_size;
int sor_nearby_number;
double sor_thresh_value;
double x_rotate_value, y_rotate_value, z_rotate_value;
double x_trans_value, y_trans_value, z_trans_value;

//角度制转弧度制
double rad(double d)
{
	return d * 3.1415926 / 180.0;
}

int main(int argc,char **argv)
{
    ros::init(argc,argv,"read_pcd");
    ros::NodeHandle nh;

/********************************参数初始化**********************************************/
    nh.param<double>("pass_x_min",pass_x_min,1.0);
    nh.param<double>("pass_x_max",pass_x_max,2.0);
    nh.param<double>("pass_y_min",pass_y_min,1.0);
    nh.param<double>("pass_y_max",pass_y_max,2.0);
    nh.param<double>("pass_z_min",pass_z_min,1.0);
    nh.param<double>("pass_z_max",pass_z_max,2.0);
    nh.param<double>("voxel_size",voxel_size,0.01);
    nh.param<int>("sor_nearby_number",sor_nearby_number,30);
    nh.param<double>("sor_thresh_value",sor_thresh_value,1.0);
    nh.param<double>("x_rotate_value",x_rotate_value,0.0);
    nh.param<double>("y_rotate_value",y_rotate_value,0.0);
    nh.param<double>("z_rotate_value",z_rotate_value,0.0);
[... 123 lines cut ...]
        //read_pcd_pub.publish(input_cloud);
        pointCloud_pub.publish(output_cloud);
        
        ros::spinOnce();

         if (received_cloud)
        {
            // 获取当前时间戳
            ros::Time timestamp = ros::Time::now();

            // 使用时间戳作为文件名
            std::stringstream ss;
            ss << output_directory << "cloud_" << timestamp << ".pcd";

            // 保存点云数据到文件
            pcl::io::savePCDFileBinary(ss.str(), *new_cloud);

            // 将布尔变量设置为 false，等待接收新的点云数据
            received_cloud = false;
        }
        loop_rate.sleep();
    }

    return 0;
}

Sisyphus-neo, 2023-07-23:
The above code is what I originally used for real-time filtering, but it always shows no point cloud input

closed 2023-07-24 by Sisyphus-neo