ouster-lidar/ouster-ros #172: ROSTopic -> Pcap

L-Reichardt, 2023-07-24:
Is there a way to record rostopics, and in a post-processing step convert them to .pcap files?
In my application, multiple sensors are used with ROS, but for Oxts-Navigation integration I need the Ouster data as .pcap. To make this "cleaner" I prefer to keep all the data in a single rosbag.

Samahu, 2023-07-24:
The `record.launch` launch file described in https://github.com/ouster-lidar/ouster-ros#recording-mode offers you the ability to record the lidar packets (unprocessed or raw packets). You could edit the launch file to add any additional topics that you want to include in the same bag file.

I am a little bit confused when you say that you need Ouster Lidar as a pcap file but later you mention that a rosbag would be cleaner.

closed 2024-08-15 by Samahu

L-Reichardt, 2024-08-16:
I went a different route, by recording the packets message into the Rosbag. The Rosbag is then post processed into .pcaps and pointsclouds from the pcap using the Ouster SDK.