ouster-lidar/ouster-ros #173: Cartesian coordinate system

Sisyphus-neo, 2023-07-24:
How to project the point cloud data of LiDAR onto the Cartesian coordinate system in ROS1 environment

Samahu, 2023-07-24:
If you are subscribing to the **PointCloud2** message through the `/ouster/points` topic then the fields `x, y, z` represent the projected Cartesian coordinates.

Sisyphus-neo, 2023-07-25:
It seems that the imu of ouster cannot output attitude information. I want to see whether the attitude information of external imu is needed for the transformation of laser radar coordinate to Local tangent plane coordinates, and then coordinate transformation can be performed

Samahu, 2023-07-25:
You can achieve this by making the sensor a child to another object through a fixed link that has the altitude transformation. This mainly a ROS setup that can be achieved using the [TF package](http://wiki.ros.org/tf).

Samahu, 2024-10-17:
The question is more of a pure ros question which can find better answers at https://discourse.ros.org/

closed 2024-10-17 by Samahu