ouster-lidar/ouster-ros #174: Unable to launch `os_driver` node after update

leonardogargani, 2023-07-24:
**Describe your question**

Today I updated my local copy of this repository and I became unable to obtain data from my OS-1-64 lidar.

Running `ros2 launch ouster_ros driver.launch.py` I obtain the following error:
```
[INFO] [launch]: All log files can be found below /home/airlab/.ros/log/2023-07-24-17-23-54-796551-airlab-9842
[INFO] [launch]: Default logging verbosity is set to INFO
/home/airlab/bunker_sensors_ros2/install/ouster_ros/share/ouster_ros
/home/airlab/bunker_sensors_ros2/install/ouster_ros/share/ouster_ros/config/viz.rviz
/home/airlab/bunker_sensors_ros2/install/ouster_ros/share/ouster_ros
/home/airlab/bunker_sensors_ros2/install/ouster_ros/share/ouster_ros/config/viz.rviz
[INFO] [rviz2-1]: process started with pid [9854]
[INFO] [os_driver-2]: process started with pid [9856]
[rviz2-1] [INFO] [1690212235.091298863] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1690212235.091338737] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-1] [INFO] [1690212235.104852258] [rviz2]: Stereo is NOT SUPPORTED
[os_driver-2] [INFO] [1690212235.138739316] [ouster.os_driver]: TIME_FROM_ROS_TIME timestamp mode specified. IMU and pointcloud messages will use ros time
[os_driver-2] [INFO] [1690212235.138801296] [ouster.os_driver]: Will send UDP data to 192.168.1.101
[os_driver-2] [INFO] [1690212235.207991526] [ouster.os_driver]: Sensor os1-xxxxxxxxxxxx.local configured successfully
[os_driver-2] [INFO] [1690212235.208091860] [ouster.os_driver]: Starting sensor os1-xxxxxxxxxxxx.local initialization...
[os_driver-2] [2023-07-24 17:23:55.208] [ouster::sensor] [info] initializing sensor: os1-xxxxxxxxxxxx.local with lidar port/imu port: 7502/7503
[rviz2-1] [INFO] [1690212235.246414989] [rviz2]: Stereo is NOT SUPPORTED
[os_driver-2] [INFO] [1690212246.896064604] [ouster.os_driver]: No metadata file was specified, using: os1-xxxxxxxxxxxx-metadata.json
[os_driver-2] [INFO] [1690212246.896372716] [ouster.os_driver]: Wrote sensor metadata to os1-xxxxxxxxxxxx-metadata.json
[os_driver-2] [INFO] [1690212246.896398075] [ouster.os_driver]: ouster client version: 0.8.1+d730798-release
[os_driver-2] product: OS-1-64, sn: xxxxxxxxxxxx, firmware rev: v2.4.0
[os_driver-2] lidar mode: 1024x10, lidar udp profile: LEGACY
[os_driver-2] [INFO] [1690212246.897362299] [ouster.os_driver]: reset service created
[os_driver-2] [INFO] [1690212246.897838660] [ouster.os_driver]: get_metadata service created
[os_driver-2] [INFO] [1690212246.898126568] [ouster.os_driver]: get_config service created
[INFO] [launch.user]: os_driver activating...
[os_driver-2] [INFO] [1690212246.898402976] [ouster.os_driver]: set_config service created
[ERROR] [os_driver-2]: process has died [pid 9856, exit code -11, cmd '/home/airlab/bunker_sensors_ros2/install/ouster_ros/lib/ouster_ros/os_driver --ros-args -r __node:=os_driver -r __ns:=/ouster --params-file /home/airlab/bunker_sensors_ros2/install/ouster_ros/share/ouster_ros/config/driver_params.yaml'].
```

I also tried using the *sensor.launch.xml* launch file, but the result did not change.
I then decided to revert back to the previous commit that was working for me, but again the same error was shown.
All the tests above lead to the same result both with ROS Humble and ROS Iron, using the *ros2* branch.

I can say for sure that the lidar has no hardware/firmware problem and it is properly connected to the computer, since I am able to visualize pointclouds using the [ouster-cli](https://static.ouster.dev/sdk-docs/cli/overview.html) utility.

To make an attempt while using ROS Humble/Iron, since *ouster-cli* is ROS-independent, I used  [ouster-lidar/ouster-ros](https://github.com/ouster-lidar/ouster-ros) drivers and the pointclouds were correctly displayed on Rviz2.

For completeness, here is the content of *driver_params.yaml*:
```yaml
ouster/os_driver:
  ros__parameters:
    sensor_hostname: 'os1-xxxxxxxxxxxx.local'
    udp_dest: '192.168.1.101'
    mtp_dest: ''
    mtp_main: false
    lidar_mode: '1024x10'
    timestamp_mode: 'TIME_FROM_ROS_TIME'
    udp_profile_lidar: 'LEGACY'
    metadata: ''
    lidar_port: 7502
    imu_port: 7503
    sensor_frame: os_sensor
    lidar_frame: os_lidar
    imu_frame: os_imu
    point_cloud_frame: os_lidar 
    proc_mask: IMG|PCL|IMU|SCAN
    scan_ring: 0
    use_system_default_qos: false
```

**Platform:**
- Ouster Sensor: *OS-1-64*
- Ouster Firmware Version: *2.4.0*
- ROS version/distro: *humble*, *iron*
- Operating System: *Ubuntu 22.04 LTS*
- Machine Architecture: *x64*

Samahu, 2023-07-24:
I didn't understand this comment of yours:

>To make an attempt while using ROS Humble/Iron, since ouster-cli is ROS-independent, I used [ouster-lidar/ouster-ros](https://github.com/ouster-lidar/ouster-ros) drivers and the pointclouds were correctly displayed on Rviz2.

Were you able to visualize the point cloud with RVIZ or not?

Samahu, 2023-07-24:
Unfortunately the logs are not telling much on why the `os_driver` is failing. Try setting the log level to "debug" by adding the line `arguments=['--ros-args', '--log-level', 'debug']` to **os_driver** within the launch file and hopefully we get more information.

leonardogargani, 2023-07-24:
> I didn't understand this comment of yours:
>
>
> Were you able to visualize the point cloud with RVIZ or not?

My bad, I linked the wrong repository. What I tried and worked without problems was [ros-drivers/ros2_ouster_drivers](https://github.com/ros-drivers/ros2_ouster_drivers), while [ouster-lidar/ouster-ros](https://github.com/ouster-lidar/ouster-ros) had no success.

leonardogargani, 2023-07-24:
> Unfortunately the logs are not telling much on why the `os_driver` is failing. Try setting the log level to "debug" by adding the line `arguments=['--ros-args', '--log-level', 'debug']` to **os_driver** within the launch file and hopefully we get more information.

This is what I get with by setting the log level to "debug" and running `ros2 launch ouster_ros driver.launch.py`:
```
[INFO] [launch]: All log files can be found below /home/airlab/.ros/log/2023-07-24-19-17-52-167236-airlab-3685
[INFO] [launch]: Default logging verbosity is set to INFO
/home/airlab/bunker_sensors_ros2/install/ouster_ros/share/ouster_ros
/home/airlab/bunker_sensors_ros2/install/ouster_ros/share/ouster_ros/config/viz.rviz
/home/airlab/bunker_sensors_ros2/install/ouster_ros/share/ouster_ros
/home/airlab/bunker_sensors_ros2/install/ouster_ros/share/ouster_ros/config/viz.rviz
[INFO] [rviz2-1]: process started with pid [3697]
[INFO] [os_driver-2]: process started with pid [3699]
[os_driver-2] [DEBUG] [1690219072.309120440] [rclcpp]: signal handler installed
[os_driver-2] [DEBUG] [1690219072.309372453] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '0' services
[os_driver-2] [DEBUG] [1690219072.309232957] [rclcpp]: deferred_signal_handler(): waiting for SIGINT/SIGTERM or uninstall
[os_driver-2] [DEBUG] [1690219072.309812062] [os_driver]: Load library libos_driver_component.so
[os_driver-2] [DEBUG] [1690219072.326017920] [os_driver]: Instantiate class rclcpp_components::NodeFactoryTemplate<ouster_ros::OusterDriver>
[os_driver-2] [DEBUG] [1690219072.326060320] [rcl]: Couldn't parse arg 0 (/home/airlab/bunker_sensors_ros2/install/ouster_ros/lib/ouster_ros/os_driver) as a remap rule in its deprecated form. Error: Expected lexeme type (19) not found, search ended at index 76, at ./src/rcl/lexer_lookahead.c:239
[os_driver-2] [DEBUG] [1690219072.326436159] [rcl]: Initializing node 'os_driver' in namespace ''
[os_driver-2] [DEBUG] [1690219072.326447076] [rcl]: Using domain ID of '47'
[os_driver-2] [DEBUG] [1690219072.331727101] [rcl]: Initializing publisher for topic name '/rosout'
[os_driver-2] [DEBUG] [1690219072.331745237] [rcl]: Expanded and remapped topic name '/rosout'
[os_driver-2] [DEBUG] [1690219072.332153809] [rcl]: Publisher initialized
[os_driver-2] [DEBUG] [1690219072.332174857] [rcl]: Node initialized
[os_driver-2] [DEBUG] [1690219072.332425874] [rcl]: Initializing service for service name 'os_driver/get_parameters'
[os_driver-2] [DEBUG] [1690219072.332431489] [rcl]: Expanded and remapped service name '/ouster/os_driver/get_parameters'
[os_driver-2] [DEBUG] [1690219072.333227038] [rmw_fastrtps_cpp]: ************ Service Details *********
[os_driver-2] [DEBUG] [1690219072.333232540] [rmw_fastrtps_cpp]: Sub Topic rq/ouster/os_driver/get_parametersRequest
[os_driver-2] [DEBUG] [1690219072.333234462] [rmw_fastrtps_cpp]: Pub Topic rr/ouster/os_driver/get_parametersReply
[os_driver-2] [DEBUG] [1690219072.333235818] [rmw_fastrtps_cpp]: ***********
[os_driver-2] [DEBUG] [1690219072.333254390] [rcl]: Service initialized
[os_driver-2] [DEBUG] [1690219072.333270319] [rcl]: Initializing service for service name 'os_driver/get_parameter_types'
[os_driver-2] [DEBUG] [1690219072.333274371] [rcl]: Expanded and remapped service name '/ouster/os_driver/get_parameter_types'
[os_driver-2] [DEBUG] [1690219072.333431488] [rmw_fastrtps_cpp]: ************ Service Details *********
[os_driver-2] [DEBUG] [1690219072.333435940] [rmw_fastrtps_cpp]: Sub Topic rq/ouster/os_driver/get_parameter_typesRequest
[os_driver-2] [DEBUG] [1690219072.333438085] [rmw_fastrtps_cpp]: Pub Topic rr/ouster/os_driver/get_parameter_typesReply
[os_driver-2] [DEBUG] [1690219072.333439850] [rmw_fastrtps_cpp]: ***********
[os_driver-2] [DEBUG] [1690219072.333454733] [rcl]: Service initialized
[os_driver-2] [DEBUG] [1690219072.333464390] [rcl]: Initializing service for service name 'os_driver/set_parameters'
[os_driver-2] [DEBUG] [1690219072.333468235] [rcl]: Expanded and remapped service name '/ouster/os_driver/set_parameters'
[os_driver-2] [DEBUG] [1690219072.333593077] [rmw_fastrtps_cpp]: ************ Service Details *********
[os_driver-2] [DEBUG] [1690219072.333595891] [rmw_fastrtps_cpp]: Sub Topic rq/ouster/os_driver/set_parametersRequest
[os_driver-2] [DEBUG] [1690219072.333597555] [rmw_fastrtps_cpp]: Pub Topic rr/ouster/os_driver/set_parametersReply
[os_driver-2] [DEBUG] [1690219072.333598829] [rmw_fastrtps_cpp]: ***********
[os_driver-2] [DEBUG] [1690219072.333606781] [rcl]: Service initialized
[os_driver-2] [DEBUG] [1690219072.333612972] [rcl]: Initializing service for service name 'os_driver/set_parameters_atomically'
[os_driver-2] [DEBUG] [1690219072.333616526] [rcl]: Expanded and remapped service name '/ouster/os_driver/set_parameters_atomically'
[os_driver-2] [DEBUG] [1690219072.333733821] [rmw_fastrtps_cpp]: ************ Service Details *********
[os_driver-2] [DEBUG] [1690219072.333737483] [rmw_fastrtps_cpp]: Sub Topic rq/ouster/os_driver/set_parameters_atomicallyRequest
[os_driver-2] [DEBUG] [1690219072.333739085] [rmw_fastrtps_cpp]: Pub Topic rr/ouster/os_driver/set_parameters_atomicallyReply
[os_driver-2] [DEBUG] [1690219072.333740388] [rmw_fastrtps_cpp]: ***********
[os_driver-2] [DEBUG] [1690219072.333749452] [rcl]: Service initialized
[os_driver-2] [DEBUG] [1690219072.333755182] [rcl]: Initializing service for service name 'os_driver/describe_parameters'
[... cut ...]
[os_driver-2] [DEBUG] [1690219084.093410557] [rcl]: Expanded and remapped topic name '/ouster/points'
[os_driver-2] [DEBUG] [1690219084.093851666] [rcl]: Publisher initialized
[ERROR] [os_driver-2]: process has died [pid 3699, exit code -11, cmd '/home/airlab/bunker_sensors_ros2/install/ouster_ros/lib/ouster_ros/os_driver --ros-args --log-level debug --ros-args -r __node:=os_driver -r __ns:=/ouster --params-file /home/airlab/bunker_sensors_ros2/install/ouster_ros/share/ouster_ros/config/driver_params.yaml'].
```

Samahu, 2023-07-24:
I do suspect that the fact you are using "OS-1-64" might have to do with it. Since this is the main thing I don't usually test the driver against - although I don't see why this might be the case -. So I would need to double check on that. Kindly can you open a support ticket through https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86 to help expiate coming up with a fix for this issue.

Samahu, 2023-07-24:
One more question, do you have more than one sensor connected to the same machine? If so could you make sure that the other sensor is either unplugged or in STANDBY or at least uses different ports.

Samahu, 2023-07-24:
I didn't have an OS-1-64 but test the diver on an OS-2-32 and it ran without any issues. So we could remove the possibility that this might be caused by the number of beams.

leonardogargani, 2023-07-25:
After further troubleshooting, I managed to identify what was wrong.

The problem was in the version of PCL installed on the computer (1.13.1).
After reverting back to 1.12.1, which is the one shipped by default with Humble, everything works as expected.

I think there is no need to open the support ticket. Thank you for your help!