ouster-lidar/ouster-ros #175: Reduce Vertical resolution for improving performance of the driver

shubham-shahh, 2023-07-27:
**Describe your question**
Hi, I am using OS0-128 lidar (REV-D) with the latest version of the driver. When the resolution is set to 512x10 and the packets are set to Legacy, I am receiving data at ~6-7Hz. but, I am using an OS0-32 lidar (Rev-6), and I am getting the data at the desired rate. I am willing to make sure I get data at 10Hz at 512x10 config. how can I reduce the amount of data processed by the driver? either by limiting the number of rings to 32 or reducing the points in some other practical ways.

**Platform:**
- Ouster Sensor? OS-32
- Ouster Firmware Version? V.2.5.2
- ROS version/distro? Melodic
- Operating System? Linux
- Machine Architecture? Arm

shubham-shahh, 2023-07-31:
> Have you switched to using `driver.launch` from `sensor.launch`?

I am using sensor.launch, but I saw the "scan_ring" param in driver.launch, I will try using driver.launch and update the issue

thanks

shubham-shahh, 2023-08-14:
Hi @Samahu, I looked through the code, I am looking to reduce the vertical resolution of the Pointcloud from let's say, 128 rings to 32 rings, I was unable to do that with the existing implementation.

thanks

shubham-shahh, 2023-08-16:
> @shubham-shahh please test #191 which shows how to reduce the vertical resolution.

Hi, thanks for quick solution, I'll test and update the thread.

thanks

shubham-shahh, 2023-08-28:
Hi @Samahu I tried the changes you mentioned, I am getting 32 rings as the output, but in the output cloud, the rings look distorted, below I have attached a screenshot for reference.
thanks
[image]

shubham-shahh, 2023-08-31:
> Hi @Samahu I tried the changes you mentioned, I am getting 32 rings as the output, but in the output cloud, the rings look distorted, below I have attached a screenshot for reference.
> thanks
> [image]
> 

Hi @Samahu any update on this?

Samahu, 2023-08-31:
@shubham-shahh I got side tracked with another request.. I don't have an update but was planning to take another stab at it tomorrow. I have a question for you though. You described the point cloud as distorted. It is hard for me to guess how the point cloud should appear for you but does this distortion only happen with the branch I linked or the point cloud also appear distorted to you in general even when using the mainline? We recently implemented a change in which we publish the point cloud in its destaggered form (for additional details refer to #188 and #199). So maybe what you are observing is the net effect of this change. All I am doing in the linked PR is allocate smaller buffer and skip a certain amount of rings determined by this hard-coded value: https://github.com/ouster-lidar/ouster-ros/blob/ceff71c634b0ad11197c6e9c26e33deb19472af8/src/point_cloud_processor.h#L23
If you set this value to 1 then this branch becomes effective the same as the mainline code.

PR #445 (2025-03-07): [ROS1] Expose the vertical beam reduction param
## Related Issues & PRs
- Closes #175
- Include #191 
- ROS2: #444 

## Summary of Changes
* Expose the vertical beam reduction value

## Validation
```bash
roslaunch ... driver.launch sensor_hostname:=<source_url> v_reduction:=2
```
or
```bash
roslaunch ... replay.launch bag_file:=<source_url> v_reduction:=2
```
and observe the number of beams are reduced by the specified factor with respect to the original beam count of the sensor.
closed 2025-03-07 by Samahu