ouster-lidar/ouster-ros #178: [ROS2] using latched metadata topic for playback

wodtko, 2023-07-28:
**Is your feature request related to a problem?**
When using the driver/cloud nodes in live mode (sensor connected), a service, a publisher, and a file are created to distribute metadata information.
However, when running in playback mode, the replay_node requires a metadata parameter to be set and, consequently, requires the user to store the metadata in a file.

**Describe the solution you'd like**
I would favor a way of using the latched topic when replaying data.
The latched topic can be recorded using rosbags and there would not be the need for keeping track of metadata files in addition to the rosbag.
More specifically, I was thinking of evaluating the `metadata` parameter and only if a valid value is found, the file option is used.
Otherwise a new parameter is evaluated, perhaps something like `metadata_topic` and the relay node acquires the metadata information via subscribing to the latched topic.

I'm aware of the issue that, in general, topics may not be received while transitioning between lifecycle states when a SingeThreadedExecutor is used. 
I can think of three different ways to overcome this:
1. a second executor can be used (like TF_tree is listening to messages while bussy waiting) which is no longer required after receiving the information
2. the node remains in the `configured` state as long as no metadata was received
3. the node may be switched to active, but will not publish PCs until metadata information was received.

Perhaps the first solution comes with the largest overhead, but would be the best way from my point of view.
Here timeouts and logging messages could be set.

Any other recommendations or comments on such a behavior?

**Targeted Platform:**
- Ouster Sensor? N/A
- Ouster Firmware Version? N/A
- ROS version/distro? ROS2 [all distros]
- Operating System? N/A
- Machine Architecture? N/A

Samahu, 2023-07-28:
Hi @wodtko, are you using the latest driver? in the current [README](https://github.com/ouster-lidar/ouster-ros/tree/ros2#recording-mode) I clearly state that the metadata file is optional since package version 8.1 and later. You no longer need to set the metadata when recording or when replaying (assuming the bag file was recorded with a recent version of the driver 8.1+).

wodtko, 2023-07-28:
Hey @Samahu, thanks a lot for the fast reply.
I was not using the provided launch files, but rather my own launch files (python instead of xml).
I just compared launch files and double checked the cpp files.

Given the current ros2 branch (ouster-ros version 0.9.2, I suppose) :
When using the example in the [README](https://github.com/ouster-lidar/ouster-ros/tree/ros2#recording-mode) I would assume that this [file](https://github.com/ouster-lidar/ouster-ros/blob/ros2/ouster-ros/launch/replay.composite.launch.xml) is actually launching things.
Here the replay node is only executed if the metadata parameter is set.

When looking at the code, the parameter is always declared and in [on_configure](https://github.com/ouster-lidar/ouster-ros/blob/9cbb840fe9e343e2122d1bc77007aa8d7a5baad7/ouster-ros/src/os_replay_node.cpp#L27) it is required that the file exists and contains metadata in order to succeed.

Am I looking at the wrong version? I think its the same as linked in your answer.

closed 2023-07-28 by wodtko