ouster-lidar/ouster-ros #179: What is the rationale for logarithmic reflectivity representation mapping post 100% reflectivity?

heemonsu, 2023-07-28:
I was curious to know about the rationale to pick the logarithmic scaling of the reflectivity values between the 101 - 255 representation. To me, this representation allows for a much higher range, at the cost of less resolution compared to a fully linear representation (where 255 value would mean 255% reflectivity). 

I assume this decision was made for a particular reason, is it really necessary to be able to account for such a high range of reflectivity, what objects/returns could even cause that?

kairenw, 2023-07-28:
Hi @heemonsu - can you contact [Ouster's customer support ](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86) at the link? 

Github issues is for questions pertaining to the SDK.

closed 2023-07-28 by kairenw

heemonsu, 2023-07-28:
Yup, makes sense! Thank you for the clarification!