ouster-lidar/ouster-ros #183: Want to use single line lidar data

WyjFrogi, 2023-08-07:
I am using a 64-line lidar, and get the lidar data through ROS. Now I want to use the single-line lidar data. Can I configure the parameters directly in the launch file? If it is not possible to configure parameters, can I directly operate on the output PointCloud2 format data of node /os1_cloud_node/points to obtain single-line radar data?

Samahu, 2023-08-07:
Have you tried the `SCAN` setting for `proc_mask`? When this is enabled it publishes a topic named `/ouster/scan` which has the `LaserScan` message type. This message only contains a single line or ring of the different beams. You can select which beam to use using the `scan_ring` parameter.

WyjFrogi, 2023-08-08:
> 
Thanks, but I didn't find proc_mask. It may be that my ROS-SDK version is too low. I found this parameter in the latest version of the code. Can I use it directly after I pull the latest code and compile it, because I don't know the ouster lidar Is there any version dependency.

Samahu, 2023-08-08:
Yes, for ROS1 you need to be on version 0.9.0. Once you pull the latest you can use it as follows:
```bash
roslaunch ouster_ros driver.launch sensor_hostname:=<sensor host or ip> proc_mask:="SCAN"
```
This would only enable LaserScan messages on the `/ouster/scan` topic.
You can maintain the rest of topics simply by changing the flag:
```bash
roslaunch ouster_ros driver.launch sensor_hostname:=<sensor host or ip> proc_mask:="SCAN|IMU|PCL|IMG"
```
Additionally you should have greater performance and stability when upgrading to the latest version.

WyjFrogi, 2023-08-09:
Ok, it is true that my ros-sdk version is too low. I have successfully run the single-line lidar mode through the latest sdk, thank you very much!

Samahu, 2023-08-09:
Perfect!

closed 2023-08-09 by Samahu