ouster-lidar/ouster-ros #185: timestamp with replay.launch

HaroldMurcia, 2023-08-07:
**Describe your question**
I am interester in recording LiDAR data together with other sensors in a rosbag with the following sh script:

first I run the launch of the ouster-ros as well as the other sensors.

```
<!-- Ouster LiDAR ... -->
  <include file="$(find minnie_ros1)/launch/ouster_os1.launch">
      <arg name="sensor_hostname" value="os-122313000337.local"/>
      <arg name="udp_profile_lidar" value="RNG19_RFL8_SIG16_NIR16_DUAL"/>
      <arg name="lidar_mode" value="2048x10"/>
      <arg name="timestamp_mode" value="TIME_FROM_ROS_TIME"/>  
      <arg name="viz" value="false"/>
      <arg name="tf_prefix" value=""/>
</include>
```

At this point I check the timestamps of the topics.

$ rostopic echo /ouster/points/header

```
seq: 0
stamp: 
  secs: 1691436592
  nsecs: 432981990
frame_id: "os_sensor"
---
```

```
$ rostopic echo /ekf/imu
header: 
  seq: 5962
  stamp: 
    secs: 1691436627
    nsecs: 963497638
  frame_id: "minnie/base_link"
orientation: 
  x: 0.001350082573480904
  y: -0.003177694045007229
  z: 0.09410891681909561
  w: 0.9955559372901917
orientation_covariance: [0.01745, 0.0, 0.0, 0.0, 0.01745, 0.0, 0.0, 0.0, 0.15708]
angular_velocity: 
  x: -0.00951102189719677
  y: 0.00586409168317914
  z: 0.0031170141883194447
angular_velocity_covariance: [0.0004, 0.0, 0.0, 0.0, 0.0004, 0.0, 0.0, 0.0, 0.0004]
linear_acceleration: 
  x: 0.0713324174284935
  y: 0.025194313377141953
  z: 9.813617706298828
linear_acceleration_covariance: [0.0004, 0.0, 0.0, 0.0, 0.0004, 0.0, 0.0, 0.0, 0.0004]

```

When doing rostopiec echo /ouster/points/header on the machine (real time) , I notice that the LiDAR timestamp matches the time of the other sensors. Then I make the rosbag record:

```
rosbag record "/ouster/metadata" "/ouster/points" "/ouster/points2" "/ekf/imu" "/tf" "/tf_static"
```

But when I run play the rosbag file, I notice that the timestamp changes, it does not coincide with the unix time nor with that of the other systems. It has a local timestamp.

```
roslaunch ouster_ros replay.launch viz:=false
rosbag play 2023-08-07-21-34-19.bag --clock
rostopic echo /ouster/points/header
```

```
seq: 20
stamp: 
  secs: **1410**
  nsecs: 573860088
frame_id: "os_sensor"
---
seq: 21
stamp: 
  secs: **1410**
  nsecs: 673879656
frame_id: "os_sensor"
```

**Q**uestion: How can I keep the original timestamp when using replay.launch ? 

NOTE:/ If I remove the additional sensor /ekf/imu, then the timestamp is correct.

**Platform (please complete the following information):**
- Ouster Sensor? OS1-128
- ROS version/distro?  Noetic
- Operating System? Ubuntu 20.04.6 LTS - Focal
- Machine Architecture? 16 GB RAM, Intel(R) Core(TM) i7-6700 CPU @ 3.40GHz
- Profile: RNG19_RFL8_SIG16_NIR16_DUAL

HaroldMurcia, 2023-08-08:
I have noticed that in my execution (before record), when I do rostopic echo to the points, the time corresponds to the timestamp.

```
  <!-- Ouster LiDAR ... -->
      <include file="$(find ouster_ros)/launch/driver.launch">
          <arg name="sensor_hostname" value="os-122313000337.local"/>
          <arg name="udp_profile_lidar" value="RNG19_RFL8_SIG16_NIR16_DUAL"/>
          <arg name="lidar_mode" value="2048x10"/>
          <arg name="timestamp_mode" value="TIME_FROM_ROS_TIME" /> 
          <arg name="viz" value="false"/>
          <arg name="tf_prefix" value=""/>
          <arg name="metadata" value="/home/hfmurciamo/bags/metadata.json"/>
      </include>
```

```
$  rostopic echo /ouster/points/header
---
seq: 69
stamp: 
  secs: 1691493287
  nsecs: 927358681
frame_id: "os_sensor"
---
```

However, when I echo to the /ouster/metadata topic in the same run, the timestamp does not match (it changes to  TIME_FROM_INTERNAL_OSC). Is this normal?

```
$ rostopic echo /ouster/metadata 
data: "{\n    \"beam_intrinsics\": \n    {\n        \"beam_altitude_angles\": \n       \
  \ [\n            20.88,\n            20.63,\n            20.33,\n            19.96,\n\
  \            19.68,\n            19.41,\n            19.09,\n            18.74,\n\
  \            18.44,\n            18.16,\n            17.85,\n            17.49,\n\
  \            17.18,\n            16.9,\n            16.58,\n            16.23,\n\
  \            15.92,\n            15.63,\n            15.3,\n            14.94,\n\
  \            14.62,\n            14.34,\n            14,\n            13.64,\n \
  \           13.32,\n            13.02,\n            12.67,\n            12.31,\n\
  \            12,\n            11.69,\n            11.36,\n            10.98,\n \
  \           10.66,\n            10.35,\n            9.99,\n            9.64,\n \
  \           9.31,\n            8.99,\n            8.64,\n            8.29,\n   \
  \         7.96,\n            7.63,\n            7.28,\n            6.91,\n     \
  \       6.57,\n            6.25,\n            5.9,\n            5.53,\n        \
  \    5.21,\n            4.86,\n            4.5,\n            4.14,\n           \
  \ 3.8,\n            3.46,\n            3.12,\n            2.76,\n            2.4,\n\
  \            2.07,\n            1.71,\n            1.35,\n            1.01,\n  \
  \          0.67,\n            0.31,\n            -0.04,\n            -0.38,\n  \
  \          -0.74,\n            -1.1,\n            -1.44,\n            -1.77,\n \
  \           -2.14,\n            -2.49,\n            -2.84,\n            -3.17,\n\
  \            -3.56,\n            -3.89,\n            -4.24,\n            -4.57,\n\
  \            -4.94,\n            -5.28,\n            -5.63,\n            -5.97,\n\
  \            -6.33,\n            -6.68,\n            -7.01,\n            -7.35,\n\
  \            -7.7,\n            -8.07,\n            -8.39,\n            -8.73,\n\
  \            -9.09,\n            -9.44,\n            -9.77,\n            -10.1,\n\
  \            -10.46,\n            -10.79,\n            -11.12,\n            -11.44,\n\
  \            -11.82,\n            -12.16,\n            -12.46,\n            -12.79,\n\
  \            -13.16,\n            -13.49,\n            -13.79,\n            -14.12,\n\
  \            -14.49,\n            -14.82,\n            -15.12,\n            -15.43,\n\
  \            -15.8,\n            -16.12,\n            -16.44,\n            -16.73,\n\
  \            -17.11,\n            -17.42,\n            -17.72,\n            -18.02,\n\
  \            -18.39,\n            -18.71,\n            -18.98,\n            -19.28,\n\
  \            -19.65,\n            -19.95,\n            -20.23,\n            -20.54,\n\
  \            -20.88,\n            -21.2,\n            -21.46\n        ],\n     \
  \   \"beam_azimuth_angles\": \n        [\n            4.16,\n            1.34,\n\
  \            -1.46,\n            -4.25,\n            4.15,\n            1.35,\n\
  \            -1.45,\n            -4.24,\n            4.16,\n            1.36,\n\
  \            -1.44,\n            -4.23,\n            4.16,\n            1.37,\n\
  \            -1.43,\n            -4.23,\n            4.17,\n            1.36,\n\
  \            -1.42,\n            -4.22,\n            4.17,\n            1.38,\n\
  \            -1.43,\n            -4.22,\n            4.19,\n            1.39,\n\
[... 71 lines cut ...]
  \            8,\n            -8,\n            -24,\n            24,\n          \
  \  8,\n            -8,\n            -24,\n            24,\n            8,\n    \
  \        -8,\n            -24,\n            24,\n            8,\n            -8,\n\
  \            -24,\n            24,\n            8,\n            -8,\n          \
  \  -24,\n            24,\n            8,\n            -8,\n            -24,\n  \
  \          24,\n            8,\n            -8,\n            -24,\n            24,\n\
  \            8,\n            -8,\n            -24,\n            24,\n          \
  \  8,\n            -8,\n            -24,\n            24,\n            8,\n    \
  \        -8,\n            -24,\n            24,\n            8,\n            -8,\n\
  \            -24,\n            24,\n            8,\n            -8,\n          \
  \  -24\n        ],\n        \"pixels_per_column\": 128,\n        \"udp_profile_imu\"\
  : \"LEGACY\",\n        \"udp_profile_lidar\": \"RNG19_RFL8_SIG16_NIR16_DUAL\"\n\
[... cut ...]
  \            0,\n            1,\n            38.195,\n            0,\n         \
  \   0,\n            0,\n            1\n        ]\n    },\n    \"sensor_info\": \n\
  \    {\n        \"build_date\": \"2023-02-09T04:49:03Z\",\n        \"build_rev\"\
  : \"v3.0.1\",\n        \"image_rev\": \"ousteros-image-prod-bootes-v3.0.1+20230209044733\"\
  ,\n        \"initialization_id\": 1629377,\n        \"prod_line\": \"OS-1-128\"\
  ,\n        \"prod_pn\": \"860-105010-07\",\n        \"prod_sn\": \"122313000337\"\
  ,\n        \"status\": \"RUNNING\"\n    }\n}"
---

```

Samahu, 2023-09-09:
@HaroldMurcia Sorry for late response but the option **TIME_FROM_ROS_TIME** is a ros driver construct.  The sensor is not aware of ROS or what ROS time is. When this option is set it won't alter the current clock on the sensor, what happens really is that the driver simply uses ROS time as the timestamp of generated ROS messages (Imu, PointCloud2. and LaserScan). So whatever you have as the active clock on the sensor will remain the same on the sensor and that's what get's captured in sensor_metadata.

Refer to this section of the code if you want more details:
https://github.com/ouster-lidar/ouster-ros/blob/6a7693c551a34fc9f8dc47c8dda09cc64e2c57c0/src/lidar_packet_handler.h#L112

I will mark this as closed but feel free to re-open or start new issue if you have more concerns regarding this.

closed 2023-09-09 by Samahu