ouster-lidar/ouster-ros #186: point_cloud_frame parameter does not seem to affect laserscan data

relffok, 2023-08-09:
**Describe the bug**
The driver allows to set the `point_cloud_frame` parameter, citing the description: 

    # point_cloud_frame[optional]: which frame of reference to use when
    # generating PointCloud2 or LaserScan messages, select between the values of
    # lidar_frame and sensor_frame.

However, if the parameter is set to `os_sensor`, the header of the Laserscan message is changed, but the scan itself does not seem to be affected. Am I misunderstanding something here?

**To Reproduce**
Steps to reproduce the behavior (steps below are just an example):
1. source ros environment
2. compile the project workspace
3. source the project workspace
4. adapt the point_cloud_frame parameter to os_sensor in the [parameter configuration](https://github.com/ouster-lidar/ouster-ros/blob/ros2/ouster-ros/config/driver_params.yaml#L55)
5. ros2 launch ouster_ros driver.launch.py
6. open rviz and visualize both, PointCloud and Laserscan
7. PointCloud and Laserscan are not aligned (turned 180 degree)

**Platform (please complete the following information):**
- Ouster Sensor? **OS-1**
- ROS version/distro? **humble**
- Operating System? **Linux**

Samahu, 2023-08-09:
Hi @relffok , you are correct the `frame_id` for **LaserScan** message changes when the point_cloud_frame value changes.
The `LaserScan` message is rather a new and an incomplete feature. I am think that we should not change the `frame_id` for LaserScan since we don't actually perform any transformation. That would be my go to solution/fix (and also update the comment for the `point_cloud_frame` parameter).

relffok, 2023-08-10:
I agree, this is one possible solution. However, I am wondering why you would want to have the PointCloud transformed to the other frame but not the LaserScan?

Without having checked out the source code, I am surprised that the LaserScan is not transformed to the `point_cloud_frame`. Is it not "extracted" from the actual PointCloud? If it was, it just needs to be extracted *after* the transformation of the PointCloud into the other frame (`os_sensor`).

Samahu, 2023-08-10:
Currently for the PCL point cloud message the frame selection only affects the xyz caluclated values. We don't apply any transformations to intensity, reflection values, etc that are included in `ouster_ros::Point` object. These values are copied as is from the LidarScan message. The same applies to LaserScan message which has 1D arrays of range and intensities. These values are copied as is regardless of the current frame_id. I think one could argue that we should apply transformation according to the referenced frame but may I ask what would be the use case?