ouster-lidar/ouster-ros #190: ROS2 driver pointcloud data lost

ahuazuipiaoliang, 2023-08-11:
**Describe the bug**
sometimes the pointcloud data lost when use the ROS2 driver for OS1 128 REV7

**To Reproduce**
Steps to reproduce the behavior (steps below are just an example):
1. source ros environment
2. compile the project workspace
3. source the project workspace
4. ros launch ouster_ros sensor.launch.xml
5. automatically open Rviz2
6. observe the issue ...

**Screenshots**
[image]
abnormal data
[image]
normal data

**Platform (please complete the following information):**
- Ouster Sensor? 
OS1 128 REV7
- Ouster Firmware Version? 
[image]
- ROS version/distro? 
Humble
- Operating System?
Ubuntu 22.04
- Machine Architecture? 
x86
- git commit hash (if not the latest).
13171a84b986cae489f644e86fb4a57a4b373cf9

ahuazuipiaoliang, 2023-08-11:
more abnormal screenshots
[image]
[image]
[image]

Samahu, 2023-08-11:
Hi @ahuazuipiaoliang, we do have have some upcoming changes the would help with the issue. In the meantime, could you try to increase the receive buffer  on our system and see if that helps out:
```bash
sudo sysctl -w net.core.rmem_default=16777216
sudo sysctl -w net.core.rmem_max=16777216
```

ahuazuipiaoliang, 2023-08-12:
@Samahu I found the data lost problem was caused by the hardware, since I switched to a new LiDAR the problem disappeared. I will close the issue.

Samahu, 2023-11-21:
Fixed per #262

closed 2023-11-21 by Samahu