ouster-lidar/ouster-ros #207: ROS bags stay at 4096 bytes

bjornsyse, 2023-09-05:
Hi

I'm trying to record data using ROS2 iron, a bag file of imu + lidar data. But I end up with rosbags 4096 bytes of data only. I've tried both this command

`ros2 launch ouster_ros record.launch.xml  lidar_port:=7502 imu_port:=7503 sensor_hostname:=169.254.221.254 bag_file:=ouster_rec.bag metadata:=ouster_rec.json viz:=false udp_dest:=172.22.168.21 timestamp_mode:=TIME_FROM_ROS_TIME `

but also the sensor mode + recording with ros2 bag like so: 

`ros2 launch ouster_ros sensor.launch.xml lidar_port:=7502 imu_port:=7503 sensor_hostname:=169.254.221.254 viz:=false udp_dest:=172.22.168.21 timestamp_mode:=TIME_FROM_ROS_TIME `

and then in separate terminal: 
`ros2 bag record /ouster/imu /ouster/points -o ros2_rec_.bag` 

The documentation states the topics are called `/ouster/imu` and `/ouster/points` in ros2, but when using the first method (ros2 launch ouster_ros record.launch.xml" I can see in the log mentions of 

> Subscribed to topic '/ouster/lidar_packets'
> Subscribed to topic '/ouster/imu_packets'

Which one is it? 

No matter which command I use I end up with small rosbags that are the same size no matter for how long I record. 
>drwxr-xr-x 2 tmgxbs tmgxbs  4096 Sep  5 11:10 ouster_rec_2023-09-05_1109.bag
>-rw-r--r-- 1 tmgxbs tmgxbs 10592 Sep  5 11:09 ouster_rec_2023-09-05_1109.json

Need some help troubleshooting this, and should also mention I'me complely new to ROS but have followed the installation instructions here: https://docs.ros.org/en/iron/Installation/Ubuntu-Install-Debians.html and this github page for ouster drivers. Might I have missed something rudymentary? 

Btw. when compiling ouster there was a lot of warnings about deprecated things in stderr. 

**Platform (please complete the following information):**
- Ouster Sensor? \[OS-0 128 Gen C\]
- Ouster Firmware Version? \[v2.5.2\]
- ROS version/distro? \[iron\]
- Operating System? \[Tried both Windows 10 WSL Ubuntu 22.04.2, but also same Ubuntu through VMWare virtual machine\]
- Machine Architecture? \[x64\]

Regards
-  Björn

bjornsyse, 2023-09-05:
Hi Samahu, thanks for prompt reply. 

I see, when trying out `ros2 launch ouster_ros sensor.launch.xml ` I'm gettings these topics listed. However i can't see any output when trying to echo them... 

`tmgxbs@SFDLSE22WR6G3:~$ ros2 topic list
/ouster/imu
/ouster/metadata
/ouster/nearir_image
/ouster/os_driver/transition_event
/ouster/points
/ouster/range_image
/ouster/reflec_image
/ouster/scan
/ouster/signal_image
/parameter_events
/rosout
/tf_static
tmgxbs@SFDLSE22WR6G3:~$ ros2 topic echo /ouster/imu
^Ctmgxbs@SFDLSE22WR6G3:~$ ros2 topic echo /ouster/points
^Ctmgxbs@SFDLSE22WR6G3:~$`

Samahu, 2023-09-05:
Being able to list the topics is not an indicator that you are streaming data successfully from the sensor. Topics are created connected or not. Since you are not getting messages on the two mentioned topics, this means that you are not getting data from the sensor.

First verify that the `dest_udp` parameter is correct (this should be the ip of the machine where you run the ROS node on) or remove it altogether and the client will figure the address automatically.

Other than that check the console logs for any problems.

bjornsyse, 2023-09-05:
I see. I might have solved my problems now, because reading up on it  a bit it seems WSL2 does not allow UDP packets to reach the guest OS that easily through it's NAT, even though portproxy settings are applied. 

I tried a different approach with VMWare where I put the lidar and windows computer on a LAN through a router, and also here I had the possibility to Bridge the guest os so that Ubuntu get's it's own IP through the router dhcp. This way, I was able to record packages and seeing them with `ros2 topic echo`. 

Seems it was a network issue on my end then, thanks for troubleshooting!

closed 2023-09-05 by bjornsyse