ouster-lidar/ouster-ros #217: ROS ouster Point struct meaning, and Autoware implmentation for enhancing data field of Point

TZECHIN6, 2023-09-15:
**Describe your question**
Whats are the current available Point data field, and their physical meaning?

I am trying to implement ouster lidar into autoware project, using this official ouster lidar driver. According from the document [link](https://autowarefoundation.github.io/autoware-documentation/main/design/autoware-architecture/sensing/data-types/point-cloud/#point-cloud-fields), autoware expect to have Point structure in this way:
```
struct PointXYZIRCAEDT
{
  float x;
  float y;
  float z;
  std::uint8_t intensity;
  std::uint8_t return_type;
  std::uint16_t channel;
  float azimuth;
  float elevation;
  float distance;
  std::uint32_t time_stamp;
};
```

Base on my observation, `return_type`, `azimuth` and `elevation` are not directly being published right now. For `channel`, does this has the same meaning to `ring`? For `distance`, does this has the same meaning of `range`? For `time_stamp`, i believe its same as `t` when selected ROS_TIME.

If I have to enhance the data field from exist driver, which file should I focusing on modification? I believe those data is actually exist from where (might be need to reverse calculation to extract them out, say `ring` can be decouple into `azimuth` and `elevation`?) Any hints and guidance would be appreciate.

**Screenshots**
Below is the struct of ouster Point now, I am not so familiar with some of the data field meaning in physical world, which are `reflectivity`, `ring` and `ambient`, for ring I believe its referring which beam belongs to for that points, starting from the very top should be "0", I guess? Any document I can refer to about the meaning of those data field?
```
struct EIGEN_ALIGN16 Point {
    PCL_ADD_POINT4D;
    float intensity;
    uint32_t t;
    uint16_t reflectivity;
    uint16_t ring;
    uint16_t ambient;
    uint32_t range;
    EIGEN_MAKE_ALIGNED_OPERATOR_NEW
};
}  // namespace ouster_ros
```

**Platform (please complete the following information):**
- Ouster Sensor? OS1-64 (i have a hard time to identify between OS1 and OS2)
- Ouster Firmware Version? 2.4
- ROS version/distro? humble
- Operating System? Ubuntu 22.04
- Machine Architecture? x64
- git commit hash (if not the latest). The latest ros2 branch

Samahu, 2023-09-15:
- `return_type` we don't have this value implemented but the driver generates one point cloud on the topic `/ouster/points` and if then sensor supports dual returns and the dual return mode was selected then you would get another point cloud published on the topic `/ouster/points2`. In short, the data is there but we don't publish create this field, it would be very easy thing to add.
- ` channel`: is the same as `ring`
- `distance`: I think this coincidence with the `range` field
- `time_stamp`: seems same as `t` field
- `azimuth` and `elevation` are not implemented currently; there are probably different ways you could go about implementing these fields, I would consider using `x,y,z` instead.

Please note that I am currently working on a feature that allows users to choose a specific representation of the point cloud type (checkout: #216)  .. I am planning to accommodate few other standard representations

TZECHIN6, 2023-09-18:
Would you mind telling me hows the `intensity` and ` reflectivity` are difference? And also whats the meaning of `ambient`? Great thanks

Samahu, 2023-09-18:
You can learn about the different channels and their meaning from [here](https://github.com/ouster-lidar/ouster-ros/blob/471402bb60f3df94516b1d1307e2837056e2098c/ouster-ros/ouster-sdk/ouster_client/include/ouster/types.h#L747)

closed 2023-09-18 by Samahu

szepilot, 2025-01-17:
We fixed it this way: build our fork, changed the `os_cloud` `point_type` to `xyzir`, and it should work. Note that we have not tested it extensively yet.

https://github.com/jkk-research/ouster-ros/issues/1

https://github.com/jkk-research/ouster-ros/commit/672b9600e532cafe9376f76a460d87d6070125ac

Samahu, 2025-01-17:
@szepilot thanks for sharing this info, consider opening a PR against ouster-ros repo with your changes so other who might be interested can benefit from it.

hect95, 2025-01-18:
@szepilot I'd be great if you can open PR with this feature :)