ouster-lidar/ouster-ros #218: No messages published without reinit of Rev 7 sensor

FranzAlbers, 2023-09-15:
**Describe the bug**
We have two Ouster Lidar sensors: An OS1-64 Rev 7 (Firmware 3.0.1) and an OS2-64 Rev D (Firmware 2.4.0). We use similar configs according to the community_driver_config.yaml. The only differences are namespaces, lidar_ip, imu_port, lidar_port, and frames, which are specific for each sensor. 

Launching the OS2 with the driver_launch.py script works without any issues. Launching the OS1 shows the same log output with no errors, but no messages are published. For some reason, messages are published only after the config in the web interface is applied (without any changes). I've observed this behavior over multiple reboots of the sensor. 

```
albers@leaf-pc:~$ ros2 launch ouster_ros driver_launch_os1.py
[INFO] [launch]: All log files can be found below /home/albers/.ros/log/2023-09-15-17-08-04-688271-leaf-pc-8782
[INFO] [launch]: Default logging verbosity is set to INFO
/home/albers/autoware/install/ouster_ros/share/ouster_ros
/home/albers/autoware/install/ouster_ros/share/ouster_ros/config/viz.rviz
/home/albers/autoware/install/ouster_ros/share/ouster_ros
/home/albers/autoware/install/ouster_ros/share/ouster_ros/config/viz.rviz
[INFO] [os_driver-1]: process started with pid [8792]
[os_driver-1] [sensing.lidar.os1.ouster_driver]: Will send UDP data to 192.168.0.30
[os_driver-1] [sensing.lidar.os1.ouster_driver]: Sensor 192.168.0.21 configured successfully
[os_driver-1] [sensing.lidar.os1.ouster_driver]: Starting sensor 192.168.0.21 initialization...
[os_driver-1] [2023-09-15 17:08:05.148] [ouster::sensor] [info] initializing sensor: 192.168.0.21 with lidar port/imu port: 7502/7503
[os_driver-1] [sensing.lidar.os1.ouster_driver]: No metadata file was specified, using: 192.168.0-metadata.json
[os_driver-1] [sensing.lidar.os1.ouster_driver]: Wrote sensor metadata to 192.168.0-metadata.json
[os_driver-1] [sensing.lidar.os1.ouster_driver]: ouster client version: 0.8.1+d730798-release
[os_driver-1] product: OS-1-64-U13, sn: 122325000484, firmware rev: v3.0.1
[os_driver-1] lidar mode: 1024x20, lidar udp profile: RNG19_RFL8_SIG16_NIR16
[os_driver-1] [sensing.lidar.os1.ouster_driver]: reset service created
[os_driver-1] [sensing.lidar.os1.ouster_driver]: get_metadata service created
[os_driver-1] [sensing.lidar.os1.ouster_driver]: get_config service created
[os_driver-1] [sensing.lidar.os1.ouster_driver]: set_config service created
[INFO] [launch.user]: os_driver activating...
```

**To Reproduce**
We only experience this issue with our OS1-64 Rev 7, not with the OS2-64 Rev D. 

1. `ros2 launch ouster_ros driver_launch.py`
2. `ros2 topic echo /points`  --> Nothing is shown
3. Apply config / reinit without any changes in the webinterface
4. `ros2 launch ouster_ros driver_launch.py`
5. `ros2 topic echo /points`   --> Messages are published correctly

**Platform (please complete the following information):**
- Ouster Sensor? OS1-64 Rev7
- Ouster Firmware Version? 3.0.1
- ROS version/distro? Humble
- Operating System? Linux / Ubuntu
- Machine Architecture? 
- git commit hash 87d6d49 (latest)

Samahu, 2023-09-15:
That's kinda weird .. I will investigate this issue and get back to you.

FranzAlbers, 2023-09-15:
Thanks for the quick answer and for your support! 

That's also what I thought. I've tried this multiple times, rebooted the sensor and the computer repeatedly, but came each time to the same results. No messages are published until we reinit the sensor.

FranzAlbers, 2023-09-15:
I've reset the sensor configuration and it's now working as it should. I did not change any parameters. 

Not sure what the problem was, but this solves the issue for me.

FranzAlbers, 2023-09-19:
Unfortunately, I was a little bit too optimistic and closed this issue too soon. I experience the behavior described in the issue again each time after powering on the lidar.

Samahu, 2023-09-19:
@FranzAlbers I assume that you do have other sensors, right? and the other sensor didn't exhibit this behavior. If it is only the REV7 sensor that having this issue please reach out [customer support](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/45/create/168) and let them know as it might be a problem with the sensor itself.

FranzAlbers, 2023-09-20:
Thanks for the hint! I'll reach out to the customer support. 
Correct, our OS2 Rev D does not exhibit this behaviour.

Indeed, this does not seem like a ROS driver issue. 
Visualizing the point cloud in Ouster Studio works only after reinitializing the driver over the web interface as well.

closed 2023-09-20 by FranzAlbers