ouster-lidar/ouster-ros #219: Fatal error when using driver.lauch

HelieD, 2023-09-20:
Fatal error when using driver.lauch :
-----------
When launching driver.launch with `roslaunch ouster_ros driver.launch sensor_hostname:=os-122146000978`, a fatal error is raised.

 [FATAL] [1695200156.014513350]: Failed to load nodelet '/ouster/os_driver` of type `ouster_ros/OusterDriver` to manager `os_nodelet_mgr'

The driver shuts down before any lidar data is shown.

* Platform : OS-0 rev. D
* FW : 2.5.2
* ROS Noetic
* Linux Ubuntu 20.04 on WSL2, x64, Nvidia Quadro P520
* Latest version of ROS_Ouster, installed with catkin using the provided guide

---

Reproduced after clean install of ros and ros ouster on a new Ubuntu 20.04 on WSL2.

---

[log.txt](https://github.com/ouster-lidar/ouster-ros/files/12671343/log.txt)

Samahu, 2023-09-20:
sensor_hostname should be passed as `os-122146000978.local`

Samahu, 2023-09-20:
Oh, just noticed that you are using WSL, I haven't tried the driver on WLS, but consider using the ip address instead of the hostname.

HelieD, 2023-09-21:
The error message is the same when using the ip. It seems that the program stops in any case before any connection is made with the sensor, because the error is the same when I use a random sensor_hostname (123456789 for example)

HelieD, 2023-09-21:
Using the local IP worked ! Thanks !

closed 2023-09-21 by HelieD