ouster-lidar/ouster-ros #220: No points are received when using any launch file on OS-0

HelieD, 2023-09-21:
No points are published when using driver.launch, sensor.launch or record.launch

The program connects succesfully to the sensor and all the lidar infos are found 
`([ INFO] [1695292577.959556491]: ouster client version: 0.8.1+d730798-release product: OS-0-128, sn: 122146000978, firmware rev: v2.5.2 lidar mode: 1024x20, lidar udp profile: RNG15_RFL8_NIR8)`, but no points are sent.

The lidar is not defective, as points collection works on Python using the Ouster library.

Command used : `roslaunch ouster_ros driver.launch sensor_hostname:=169.254.198.165`

Also tested with the right udp ports, without change.

The RViz vizualisation is empty :

[image]

The rosbag files produced when using record.launch only contains some sensor info (beam_intrinsics, imu_intrinsics, etc...) but no point data.
When using `rostopic echo` to view the messages sent on the different topics created, no message is seen.

    Platform : OS-0 rev. D
    FW : 2.5.2
    ROS Noetic
    Linux Ubuntu 20.04 on WSL2, x64, Nvidia Quadro P520
    Latest version of ROS_Ouster, installed with catkin using the provided guide

--------
[log.txt](https://github.com/ouster-lidar/ouster-ros/files/12683537/log.txt)

Samahu, 2023-09-21:
@HelieD Didn't you just confirm that using local ip worked in the other  #219

HelieD, 2023-09-22:
That was an other issue, so i opened a new one. Now, the lidar is reached but no points are received.

Samahu, 2023-11-24:
Closing this issue due to no inactivity. Feel free to re-open if you still having the issue.

closed 2023-11-24 by Samahu