ouster-lidar/ouster-ros #222: Failed to load nodelet '/ouster/os_driver` of type `ouster_ros/OusterDriver` to manager

Vineet-Pandey, 2023-09-28:
I am using Ouster OS0-32 on Ubuntu 18 OS (ROS Melodic). When I launch driver.launch, i receive the following error and the node fails to load:

```  
[ INFO] [1695933697.697372216]: Loading nodelet /ouster/os_driver of type ouster_ros/OusterDriver to manager os_nodelet_mgr with the following remappings:

[ INFO] [1695933697.703015200]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] has not been advertised, waiting...

[ INFO] [1695933699.749443205]: Initializing nodelet with 12 worker threads.

[ INFO] [1695933699.765939364]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] is now available.

[ INFO] [1695933699.822747573]: TIME_FROM_ROS_TIME timestamp mode specified. IMU and pointcloud messages will use ros time

[ WARN] [1695933699.822785774]: lidar port set to zero, the client will assign a random port number!

[ WARN] [1695933699.822804838]: imu port set to zero, the client will assign a random port number!

[ INFO] [1695933699.822837060]: Will use automatic UDP destination

[ INFO] [1695933702.958399408]: rviz version 1.13.30

[ INFO] [1695933702.958462396]: compiled against Qt version 5.9.5

[ INFO] [1695933702.958484836]: compiled against OGRE version 1.9.0 (Ghadamon)

[ INFO] [1695933702.970869200]: Forcing OpenGl version 0.

[ INFO] [1695933703.708634056]: Stereo is NOT SUPPORTED

[ INFO] [1695933703.708788444]: OpenGL device: GeForce GTX 1660 Ti/PCIe/SSE2

[ INFO] [1695933703.708851086]: OpenGl version: 4.6 (GLSL 4.6).

[FATAL] [1695933705.346283393]: Failed to load nodelet '/ouster/os_driver` of type `ouster_ros/OusterDriver` to manager `os_nodelet_mgr'

================================================================================REQUIRED process [ouster/os_driver-3] has died!

process has died [pid 5912, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load ouster_ros/OusterDriver os_nodelet_mgr __name:=os_driver __log:=/home/bob/.ros/log/6aa42476-5e3f-11ee-ba13-673aca74fe6f/ouster-os_driver-3.log].

log file: /home/bob/.ros/log/6aa42476-5e3f-11ee-ba13-673aca74fe6f/ouster-os_driver-3*.log

Initiating shutdown!

================================================================================

[rviz-4] killing on exit

[ouster/os_driver-3] killing on exit

[ouster/os_nodelet_mgr-2] killing on exit

[rosout-1] killing on exit

[master] killing on exit

shutting down processing monitor...

... shutting down processing monitor complete

done
``` 
How do we fix this?

Samahu, 2023-09-28:
unfortunately the logs does not show any useful information .. so you may want to enable additional level of log information.

Vineet-Pandey, 2023-09-29:
Could you please let me know if that could be done through the launch file? (i saw a few closed issues in which  it was done for the ros2 package)

Samahu, 2023-09-29:
You may want to enable [GDB debugging](http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB) If this is too much then consider reaching out to our customer support.

Vineet-Pandey, 2023-09-29:
I tried this with in the driver.launch file:
```
<group ns="$(arg ouster_ns)">
    <node pkg="nodelet" type="nodelet" name="os_nodelet_mgr"
      output="screen" required="true"
      launch-prefix="xterm -e gdb --args"
      args="manager"/>
  </group>
```
but the terminal closes as soon as the nodelet crashes.

Samahu, 2023-10-02:
Please reach our customer support for help with this.

rami1891, 2023-11-08:
@Vineet-Pandey Were you able to figure out the problem?

LinusNEP, 2024-01-02:
@Vineet-Pandey I am having the same issue. Were you able to fix it?

Samahu, 2024-01-02:
Please reach to our customer support and they would assist you with this.

Vineet-Pandey, 2024-01-02:
> @Vineet-Pandey I am having the same issue. Were you able to fix it?

I did! In my case it was incorrect lidar address (i had a typo of using ```.``` instead of ```-```).

Samahu, 2024-01-02:
@Vineet-Pandey thanks for updating the issue. I am closing this issue with a "nothing to fix".

For @LinusNEP and @rami1891 consider opening a new issue if you had a different error log otherwise as I noted earlier you may need to contact [Ouster support team](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86)

closed 2024-01-02 by Samahu