ouster-lidar/ouster-ros #227: Changing the FOV of the Lidar in ROS

Vineet-Pandey, 2023-10-10:
**Describe your question**
I want to change the FOV of the Ouster OS0-32 from [0-360] to [0, 180000] and in the Software User guide i see  the use of ``` set_config_param azimuth_window``` How can i set this param in ros? I don't see this particular rosparam in rosparam list.

**Platform (please complete the following information):**
- Ouster Sensor: OS-0-32
- ROS version/distro Melodic
- Operating System? Ubuntu 18

Samahu, 2023-10-12:
We don't expose the ability to set azimuth window via ros launch file. We have `/ouster/set_config/` (described [here](https://github.com/ouster-lidar/ouster-ros#setconfig-experimental) which could be used for this purpose but due to sensor re-initialization this is unlikely to work for ROS1 driver. It does work though for the ROS2 implementation since it implements Node LifeCycle.

imdsafi09, 2024-03-14:
Hi @Samahu 
Could you please further elaborate how we can change the configuration to set the azimuth window up to 180 ° in ROS-2?

Samahu, 2024-03-14:
Sure, create a config file in some known path and add the following content:
```json
{
    "azimuth_window": 
    [
        0,
        180000
    ],
}
```
This config would set the azimuth window to 180 degrees. Let's name the file os-xxx.json. Now to apply this config file you can do so by invoking the command:
```bash
ros2 service call /ouster/set_config ouster_sensor_msgs/srv/SetConfig "{config_file: /path/to/file/os-xxx.json}"
```
Make sure to set the correct path of your file. If the call was successful you should get the following response:
```bash
ouster_sensor_msgs.srv.SetConfig_Response(config='{\n    "azimuth_window": \n    [\n        0,\n        180000\n    ]\n}')
```
The driver will stop momentarily and and restart with new the new azimuth window configuration.

imdsafi09, 2024-03-15:
Hi @Samahu,

Thank you for your response. I would like to know whether the above configuration will only limit the field of view of the point cloud or image information  (signal image, etc.) as well?

Best regards,

Samahu, 2024-03-16:
It will affect both

Samahu, 2024-08-21:
addressed in #357 and #358

PR #357 (2024-08-21): ROS-227: Set LIDAR FOV on startup and add an option to persist the config [HUMBLE/IRON]
## Related Issues & PRs
- related #138
- closes #227
- relatd #356 
- related #358 

## Summary of Changes
* Set LIDAR FOV on startup
* Add an option to persist the config

## Validation
* Run the node with azimuth start and stop as follows 
```bash
ros2 ouster_ros sensor.launch \
    azimuth_window_start:=150000 \
    azimuth_window_end:=250000
```
Verify that the azimuth window is cropped to selection 

* Run the node with `persist_config` set **true** as follows 
```bash
ros2 ouster_ros sensor.launch.xml [... cut]
PR #358 (2024-08-21): ROS-227: Set LIDAR FOV on startup and add an option to persist the config [FOXY]
## Related Issues & PRs
- related #138
- closes #227
- relatd #356
- related #357 

## Summary of Changes
* Set LIDAR FOV on startup
* Add an option to persist the config

## Validation
* Run the node with azimuth start and stop as follows 
```bash
ros2 ouster_ros sensor.launch \
    azimuth_window_start:=150000 \
    azimuth_window_end:=250000
```
Verify that the azimuth window is cropped to selection 

* Run the node with `persist_config` set **true** as follows 
```bash
ros2 ouster_ros sensor.launch.xml \
   <p [... cut]
closed 2024-08-21 by Samahu