ouster-lidar/ouster-ros #231: Consider changing `ptp_utc_tai_offset` default value

lowellausen, 2023-10-13:
**Describe your question**
A clear and concise description of your question.

Hello! Opening this issue here as an idea/suggestion :pray: 

Should  the default value for the new `ptp_utc_tai_offset` parameter (introduced here https://github.com/ouster-lidar/ouster-ros/pull/195)  be set to `0.0`?

We had our PTP synchronization working nicely (I think we followed this at some point https://github.com/ouster-lidar/ouster_example/issues/144), and after pulling the latest updates we had to look into the changelog to figure that we needed to change the aforementioned parameter from `-37.0` to `0.0` for our synchronization to work properly again.

My idea would be to set this to `0.0` so
1. It doesn't automatically "break" the system for people who pull the latest changes 
2. It removes the need to change all the `-37.0`s when a new leap second is added   

I can open a PR if we agree to some possible change on this issue!

Thanks for the attention!

**Platform (please complete the following information):**
- Ouster Sensor? OS-0 32 Rev D (we also have some devices with newer hw revisions, but I have not yet gotten to test this on them)
- Ouster Firmware Version?  2.0.0 and 2.5.2.
- ROS version/distro? Humble
- Operating System? Ubuntu 22.04 based ros:humble Docker image
- Machine Architecture? x64
- git commit hash: https://github.com/ouster-lidar/ouster-ros/pull/195

lowellausen, 2023-11-15:
Hi @Samahu, thanks for your answer! 

I understand your point and I agree, it is easy to adapt to this change as it is. With the upcoming breaking change indication in the changelog this will be quite clear! :ok_hand: 

Will close this one, since the proposed solution is underway. Thanks!

closed 2023-11-15 by lowellausen

Hommus, 2024-04-30:
@Samahu I came across this issue and would like to suggest adding a note to the [Ouster PTP Quickstart Guide](https://static.ouster.dev/sensor-docs/image_route1/image_route2/appendix/ptp-quickstart.html) to help those in a similar situation in the future.

Samahu, 2024-04-30:
That's a good note. I will check with the page editor if they can add such a note.

PR #254 (2023-11-15): SW-5466: Support Velodyne point type in the ROS driver amendments
## Related Issues & PRs
- Follow ups to #216
- resolves #231 (adds a note to CHANGLOG.rst indicating that the default ptp_utc_tai_offset forms a breaking change).

## Summary of Changes
- Add support to control `point_type` through `*.launch.xml` files
- Added a note to CHANGLOG.rst indicating that the default ptp_utc_tai_offset forms a breaking change.

## Validation
Same steps to validate in #216 but utilize `*.launch.xml` format to set the `point_type` parameter.
PR #253 (2023-11-16): SW-5657: Back port SW-5466 to ROS-1
## Related Issues & PRs
- parent #216 
- resolves #15 
- resolves #29
- resolves #210 
- resolves #97
- related #217
- related #223
- resolves #231 (adds a note to CHANGLOG.rst indicating that the default `ptp_utc_tai_offset` forms a breaking change).  

## Summary of Changes
- added the ability to customize the published point clouds(s) to velodyne point cloud format and
  other common pcl point types.
- made ouster_image_nodelet operate separately from ouster_cloud_nodelet
- added a note to `CHANGLOG.rst` indicating that the default `ptp_utc_tai_offset` fo [... cut]