ouster-lidar/ouster-ros #235: Add a laser_scan_frame

lowellausen, 2023-10-16:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Previously the scan and point cloud had the same orientation when setting the point cloud frame as the value of the sensor frame. After the previous change (https://github.com/ouster-lidar/ouster-ros/pull/195), they are off by 180 degrees regardless of which frame you set as the point cloud frame.

Lidar frame as the point cloud frame:

[image]

Sensor frame as the point cloud frame:

[image]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Adding a separate `laser_scan_frame` parameter, so we can have by default the current behavior, where the scan data will be stamped with lidar frame. But it also gives us the possibility to stamp it with the sensor frame (as we do for the point cloud) and they keep being aligned.

I'm also opening a PR with this change, to be evaluated!

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Of course, also adding the transformation into the given frame would be ideal.

**Targeted Platform (please complete the following information only if applicable, otherwise dot N/A):**
- Ouster Sensor? N/A
- Ouster Firmware Version?  N/A
- ROS version/distro? Humble
- Operating System? N/A
- Machine Architecture? N/A

lowellausen, 2025-02-20:
The main problem with this (orientation mismatch between scan and point cloud) seems to have been fixed!

closed 2025-02-20 by lowellausen

Samahu, 2025-02-20:
Thanks for the feedback!