ouster-lidar/ouster-ros #238: `RAW` option for proc_mask

Grisly00, 2023-10-17:
**Is your feature request related to a problem? Please describe.**
At the moment I am switching back and forth between the different XML and Python launch files. As far as I understand the preferred approach in the future would be to use `driver.launch.py`. The problem for me is that it does not publish raw lidar_packets which I like to record.

**Describe the solution you'd like**
Would it be possible to add another key to proc_mask like `RAW` to publish raw lidar_packets?

**Describe alternatives you've considered**
At the moment I simply use `sensor.independent.launch.py`. (`sensor.independent.launch.xml` gave me recently some errors as described in this issue #159.)

**Targeted Platform (please complete the following information only if applicable, otherwise dot N/A):**
- ROS version/distro? ROS2 humble
- Operating System? Jammy
- Machine Architecture? ARM

Samahu, 2023-10-17:
Yes, we do plan on providing a python versions for record/replay launch files..

Grisly00, 2023-10-31:
@Samahu: Thanks, but this does not really answer my question. 
To rephrase, is it possible to provide a `RAW` option to `proc_mask` so we can record lidar_packets with `driver.launch.py` launch file, or will `sensor.independent.launch.py` be the way to go also in future?

Samahu, 2023-11-08:
Got it. It is the same as #251

PR #363 (2024-09-19): [NOETIC] Further reduce overhead in the ros driver
## Related Issues & PRs
- resolves #10 
- resolves #152 
- related #175 
- resolves #238 
- resolves #251 
- related #259
- resolves #304
- related #306 
- related #352

## Summary of Changes
- Add Jazzy to the build status
- Update to SDK client 0.11.1
- Switch to using LidarPacket and ImuPacket
  - Get's rid of the deprecation warnings
- Publish raw packets for the `os_driver`
- Add a flag to control the publish of organized point cloud vs non-organized
- Add a flag to control the publish of destaggerd point cloud vs staggered
- Add paramete [... cut]
PR #369 (2024-09-20): [HUMBLE|IRON|JAZZY] Port ROS-363 to ROS2
## Related Issues & PRs
- ported_from #363 
- related #10 
- related #152 
- related #175 
- resolves #238 
- resolves #251 
- resolves #259
- resolves #304
- related #306 
- resolves #352

## Summary of Changes
- Update to SDK client 0.11.1
- Switch to using LidarPacket and ImuPacket
  - Get's rid of the deprecation warnings
- Publish raw packets for the `os_driver`
- Add a flag to control the publish of organized point cloud vs non-organized
- Add a flag to control the publish of destaggerd point cloud vs staggered
- Add parameter to set the min/max l [... cut]
closed 2024-09-19 by Samahu