ouster-lidar/ouster-ros #239: laserscan is not correctly

adridelta, 2023-10-24:
**Describe the bug**
Laserscan does not publish data correctly. there are only a few point when using /ouster/scan. PointCloud data is present and published correctly

**To Reproduce**
ros2 launch ouster_ros driver.launch.py

**Screenshots**
[image]
[image]

**Platform (please complete the following information):**
- Ouster Sensor? OS1-32-U
- Ouster Firmware Version? 2.4
- ROS version/distro? \foxy
- Operating System? Ubuntu 20.04
- Machine Architecture? x64
- git commit hash (if not the latest).

Samahu, 2023-11-03:
@adridelta just to make sure, did you set the `scan_ring` to an appropriate value? For an OS1-32-U you should be using a value of 16?

Samahu, 2023-11-29:
Resolved by #204

closed 2023-11-29 by Samahu