ouster-lidar/ouster-ros #240: Missing lidar chunk os1-128-u

JonasTietz, 2023-10-24:
**Describe the bug**
We are using a Nvidia Jetson AGX xavier on Auvidea X221-AI carriert board and we experience some missing pointcloud chunks on many of the pointclouds we receive. The missing chunk is always in the same spot, but some times larger and sometimes smaller. This happens on all udp profiles except for the Low Data Rate Profile. On the dual return profile it is more extreme. 

We are on commit 6a7693c551a34fc9f8dc47c8dda09cc64e2c57c0 and had nothing running in the background on a fresh ubuntu 20.04 install. The lidar is connected to a "RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller". Lidar firmware version is v2.5.2

Here is a video of the behavior.
https://github.com/ouster-lidar/ouster-ros/assets/17435797/555eb98b-b7be-4171-927e-d0ae7420444a

**To Reproduce**
Steps to reproduce the behavior (steps below are just an example):
1. source ros environment
2. ouster-ros
3. source the project workspace
4. roslaunch ouster_ros driver.launch ...
5. observe the issue ...

**Platform:**
- Ouster Sensor: OS1-128-U
- Ouster Firmware Version: v2.5.2
- ROS version: noetic
- Operating System: Ubuntu 20.04
- Machine Architecture: arm64
- git commit: 6a7693c551a34fc9f8dc47c8dda09cc64e2c57c0

FabianThomsen, 2023-10-31:
I have the same issue with an OS1 128 in ROS2 humble/iron (tested both). The higher the resolution, the more the PointCloud in RVIZ2 is "flickering" and missing chunks. The fix from [here](https://github.com/ouster-lidar/ouster-ros/issues/190#issuecomment-1675273573) - increasing the UDP receive buffer size - seemed to work at first, but after a day, the flickering started again, so it seems to have been pure luck.

I can add, that even in the time, the fix seemed to have worked, it only did so while the ouster driver was running in a docker container started with `--network host`, while in a container with a dedicated `ipvlan` for the OS1, it did never work. On the other hand, an older version of the driver (version 0.8.3 in the `package.xml`, for ROS noetic) does work without any missing chunks.

shubham-shahh, 2023-11-02:
any update on this issue? I have noticed it on os-32 as well, running latest driver.

lionator, 2023-11-17:
Any Update? I have the same with OS2-128. When i subscribe to it more often it gets even worse. Also it depends on the lidar_mode. Below 1024x10 it works with rviz as a single subscriber. When i want to do a recording with the record.launch file I get chunks missing again.
[Screencast from 17.11.2023 10:26:20.webm](https://github.com/ouster-lidar/ouster-ros/assets/12926762/68948ebe-1016-46bc-9797-3d89c0e27597)

CPU Resources are fine too:
[image]

I hope we get some advice soon.

javierAraluce, 2023-11-17:
We are encountering the same issue when configuring our OS1-128 with a resolution of 2048x10. It functions properly in OusterStudio, but when utilizing the ROS2 driver, it experiences flickering.
Currently, we are using:

- ROS2 Humble
- Ubuntu 22
- It is running in a Docker container.

Thanks

Samahu, 2023-11-18:
@lionator and @javierAraluce I have just pushed a new release [towards ros2 (rolling,humble, iron)](https://github.com/ouster-lidar/ouster-ros/releases/tag/ros2-v0.12.0) which has several fixes and improvements. Could you try the updated version 0.12.0 and let me know whether this fixes the problem you are observing or not? Thanks

lionator, 2023-11-20:
> @lionator and @javierAraluce I have just pushed a new release [towards ros2 (rolling,humble, iron)](https://github.com/ouster-lidar/ouster-ros/releases/tag/ros2-v0.12.0) which has several fixes and [...]

I have done it. It works now with 2048x10 with one subscriber, but if 2 or more subscriber are added it stops working. See here:
[Screencast from 20.11.2023 11:50:09.webm](https://github.com/ouster-lidar/ouster-ros/assets/12926762/35311da7-e943-421e-a254-18ec560b60c9)
It doesn't matter whether i subscribe with another rviz2 window or in the terminal, it does weird things. I never had this before with our previous Ouster OS1.

FabianThomsen, 2023-11-20:
> @lionator and @javierAraluce I have just pushed a new release [towards ros2 (rolling,humble, iron)](https://github.com/ouster-lidar/ouster-ros/releases/tag/ros2-v0.12.0) which has several fixes and [...]

I also tried the new release and it doesn't change anything on my machine. I can also confirm that with more subscribers, a larger portion of the pointcloud is missing on average.

Samahu, 2023-11-21:
> I have done it. It works now with 2048x10 with one subscriber, but if 2 or more subscriber are added it stops working. See here: [Screencast from 20.11.2023 [...]

Thanks for the feedback, I don't usually test with multiple subscribers. I will look into the issue.

lionator, 2023-11-22:
If you test it with the record.launch.xml file and rviz visualisation is active you have 2 subscribers and it does not work.

javierAraluce, 2023-11-28:
Hello all, 
We have tested the new release with our lidar, and it still suffers from flickering.

tom-bu, 2023-11-28:
Does this have to do with the socket buffer size? https://github.com/ros-drivers/ros2_ouster_drivers/issues/89

Samahu, 2023-11-29:
@JonasTietz I tried to reproduce this issue under ROS1 on NVIDIA AGX with a sensor connected to a switch but I couldn't generate the problem with the latest ouster-ros release. Are you able to generate the problem when connected directly to the sensor or maybe try a different network interface? I remember one of [our users noted that there were issues with certain network hardware](https://github.com/ouster-lidar/ouster_example/issues/323#issuecomment-1172385208) which could be the cause of the problem. I am running a long term test to check for any degradation in the performance.

I was able to re-produce the issue with ROS2 and looking further into it, but since the issue is specific to ROS2 it might be what @tom-bu had mentioned.

Samahu, 2023-11-29:
> @JonasTietz I tried to reproduce this issue under ROS1 on NVIDIA AGX with a sensor connected to a switch but I couldn't generate the problem with the latest ouster-ros release. Are you able to [...]

@JonasTietz I ran 12 hours long test on my setup and I didn't observe any degradation in the driver performance using the ROS1 driver. I used 1024x10 lidar mode and the RNG19_RGL8_SIG16_NIR profile and the frame rate was at the expected 10 Hz. I am going to argue that your network setup has to do with the issue for ROS1. Could you please check if you can produce the problem without using the described network switch?

I am still investigating the issue with the ROS2 driver.

Samahu, 2023-12-01:
@JonasTietz thanks for the feedback, will keep the ticket open untill we have it resolved.

Imaniac230, 2024-02-07:
Hi, we have been encountering the same issues using our `OS-0-128` ousters (fw: `v3.0.1`). I've been doing some quick experiments to see what's going on. I haven't found any obvious root cause, but maybe this information will still be somewhat helpful.

Most of my following assumptions are coming from the fact that I have a hacked-up version of the old community driver (with ported functionality for the new firmware) [in this fork branch](https://github.com/Imaniac230/ros2_ouster_drivers/tree/ros2-fw3-refactor-follow-up) that does not appear to have any of these issues (or they're at least a lot less prevalent). So, while there might be some hardware-related differences that influence this behavior, that would still not explain why there is one software implementation that works and one that has problems, when using the same hardware. It would be good if someone else could also confirm this.

Here, I am always using the single ouster driver node, so the raw packets must be getting processed directly between the `connection` and `processing` threads and are not being sent anywhere through the ros interfaces.

First, by counting up the the accumulated packets in `ScanBatcher::operator()()` calls and printing them out once the accumulation condition is met, it is visible that the final cloud is never actually accumulated from all of the expected packets.
* with 1024 columns at 16 columns per packet we should expect a full cloud accumulated from 64 packets,
* with 4096 columns at 16 columns per packet we should expect a full cloud accumulated from 256 packets.

But those counts are never reached, hence we're missing chunks of the cloud. Going further and printing out the individual `frame_id` and `measurement_id` numbers being processed in `ScanBatcher::operator()()`, there is indeed a gap and some `measurement_id`s are missing.

I performed various experiments with different lidar output modes and subscriber counts to compare the behavior of the two driver implementations. To make sure I'm doing the counting as close as possible to the actual moment any new packet reaches the socket, I've added the counting directly into the lambda call in `OusterSensor::handle_lidar_packet()`:
```
lidar_packets->write_overwrite([this, &cli, pf](uint8_t* buffer) {
        ...

        const auto timeStamp = std::chrono::high_resolution_clock::now().time_since_epoch().count();
        if (!success) std::cout << "[" << timeStamp << "] ERROR reading lidar packet!" << std::endl;
        const uint16_t f_id = pf.frame_id(buffer);
        const uint16_t fIDDiff = f_id - lastFrameID;
        if (fIDDiff > 1) {
            std::cout << "[" << timeStamp << "] missing " << (fIDDiff - 1) << " whole frames (last f_id: " << lastFrameID << ", new f_id: " << f_id << ")" << std::endl;
        }
        for (int icol = 0; icol < pf.columns_per_packet; icol++) {
            const uint8_t* col_buf = pf.nth_col(icol, buffer);
            const uint16_t m_id = pf.col_measurement_id(col_buf);
            if (f_id == lastFrameID) {
                const uint16_t mIDDiff = m_id - lastMeasID;
                if (mIDDiff > 1) {
                  std::cout << "[" << timeStamp << "] missing " << (mIDDiff + 1) / 16 << " packets (last m_id: " << lastMeasID << ", new m_id: " << m_id << ")" << std::endl;
                }
                if (mIDDiff < 1) {
                  std::cout << "[" << timeStamp << "] got the same packet again, (last m_id: " << lastMeasID << ", new m_id: " << m_id << ")" << std::endl;
                }
            }
            lastMeasID = m_id;
        }
        lastFrameID = f_id;
});
```

I've also added the same to my modified version of the community driver, so I can compare the outputs from both, like for like. All of the subscribers were run on the same machine that was running driver node. I haven't performed any tests with subscribers being on a different machine.

Testing configuration 1:
* CPU: `AMD Ryzen 7 3700X @ 4GHz`
* MOBO: `Gigabyte B450 AORUS PRO-CF` (NIC: `Intel I211` )
* connection: `lidar -> router (MikroTik RBM33G) <-> switch <- PC`

1. mode: `1024x20 @ RNG19_RFL8_SIG16_NIR16` (network data rate: `~260Mbps`)
    <details>
    <summary>ouster_ros (this)</summary>

    * zero subs: no missing packets
    * one sub: cpu ~10%, continuous missing packets - missing chunk
        <details>
        <summary>output sample</summary>

        ```
        [os_driver-1] [1707163269239530629] missing 24 packets (last m_id: 191, new m_id: 576)
        [os_driver-1] [1707163269290930295] missing 26 packets (last m_id: 191, new m_id: 608)
        [os_driver-1] [1707163269331508908] missing 14 packets (last m_id: 191, new m_id: 416)
        [os_driver-1] [1707163269381766988] missing 14 packets (last m_id: 191, new m_id: 416)
        [os_driver-1] [1707163269431606906] missing 14 packets (last m_id: 191, new m_id: 416)
        [os_driver-1] [1707163269481805224] missing 14 packets (last m_id: 191, new m_id: 416)
        [os_driver-1] [1707163269531601038] missing 14 packets (last m_id: 191, new m_id: 416)
        [os_driver-1] [1707163269582329121] missing 15 packets (last m_id: 191, new m_id: 432)
        [os_driver-1] [1707163269631583257] missing 14 packets (last m_id: 191, new m_id: 416)
        [os_driver-1] [1707163269681783229] missing 14 packets (last m_id: 191, new m_id: 416)
        ```
[... 793 lines cut ...]
    * five subs: cpu ~70% whole cloud flickers every ~6s, but doesn't break
[... cut ...]
    * Refactoring the connection and processing thread initializations in `OusterSensor::start_packet_processing_threads()` and `OusterSensor::start_sensor_connection_thread()` from unique pointers with lambdas to the old style `std::bind()` calls made no difference.
    * Compiling the whole driver node as a classic standalone executable instead of the cool ros2 component macros also made no difference.

What I see for now is:
1. There is something different in the community implementation that doesn't create these missing packet events.
2. The events somehow relate to cpu load and data rate - maybe some system calls are being handled differently?

Samahu, 2024-02-08:
@Imaniac230 thanks for your detailed report, this very helpful. When porting the ROS1 driver to ROS2 and implementing the combined node I haven't tested its behavior with multiple subscribers at work hence I haven't noted this issue. I think one notable difference is that the community driver utilize a lock-free ring buffer which definitely has better advantages over the current ThreadSafeRingBuffer implementation, it was my intention to switch to lock-free implementation per [my comment](https://github.com/ouster-lidar/ouster-ros/blob/5a08f89e21dabfb78ae359f9bfc4ba869a275104/src/os_sensor_nodelet.h#L128) but I considered this an optimization and deferred it to later. This may not be the sole problem but a lock-free implementation - if done right - would perform better since it doesn't need to hold a lock on every push or pop of items.

Another thing is I never actually checked or performed a benchmark on the queue length and how often it gets filled. As you have noted in your report that with a single subscriber there isn't any missing packets so the allocated buffers seem to work just fine.

There is also one factor which I would be curious if you took a look during the comparison, when comparing my implementation vs the community driver - using a single subscriber - it was note worthy to me that the community driver had less CPU utilization but at the same time it had less frequency of publishing point clouds as the LidarMode increased [link to plots](https://github.com/ouster-lidar/ouster-ros/discussions/205).

andre-nguyen, 2024-02-09:
Adding a data point here, I have two Lenovo Thinkpad laptops (different models) running ubuntu 22 with ROS Humble. I can see missing packets on one but not the other. Funnily enough, the laptop where everything works is an older model with a weaker processor.

I wonder if this is related to the network adapter in use? Maybe just a fluke?

Samahu, 2024-02-12:
Thanks all for the feedback, I have identified the problem and will be working on a fix soon. The issue stems when the buffer read operation takes more time than available window before we receive another packet. If this happens then we'd run into the problem. Since I perform packet processing and publishing in the same thread that reads packets that problem will be exacerbated when having more than one subscriber to the point clouds node. I will work on a fix to address the issue very soon.

Imaniac230, 2024-02-14:
> There is also one factor which I would be curious if you took a look during the comparison, when comparing my implementation vs the community driver - using a single subscriber - it was note worthy [...]

Regarding the topic rates, I've taken a quick look at the pointcloud topic from both implementations (just with the `ros2 topic hz` tool, waiting for at least 100 samples).  First I had just the `hz` tool being the only subscriber, and then also with an rviz subscription running. In all cases, both implementations always seemed to reach the desired rates (for all the lidar modes, and using profile `RNG19_RFL8_SIG16_NIR16`). So as far as my testing goes, I haven't seen any of the huge drops to `5 Hz` with the community version. The rates do start to drop as more subscribers are being added, but it is the same for both implementations.

RYO0115, 2024-02-16:
This issue also happend with OS1-128 and OS2-128 Rev7 with Ubuntu 20.04 & ROS noetic.

I have changed harness and power supply but won't fix.
But weird thing is that sometimes this data loss won't happen.

Seems like thread timing issue so I estimate that CPU usage is the key, but any updates or way to deal with this?

Imaniac230, 2024-02-19:
After another batch of experiments, I think there two separate problems.
1. The missing incoming raw packets don't even get written into the buffer, so they are missing completely from the accumulated cloud. We're never overflowing the buffer because we're not even writing the packets into it (probably caused by the lock creation overhead, as you mention here https://github.com/ouster-lidar/ouster-ros/issues/240#issuecomment-1934735107).
2. If the `read` callback operations are substantially slower that the `write` operations, then we should start overflowing the buffer, what I think is what you mention here https://github.com/ouster-lidar/ouster-ros/issues/240#issuecomment-1939514177. But this will only happen if the first problem is resolved, otherwise this never happens.

If we keep track of the read and write operations and print their difference for ex. like this:
```
lidar_packets->read([this](const uint8_t* buffer) {
                readCounter += 1;
                uint64_t counterMax = UINT64_MAX;
                readCounter.compare_exchange_strong(counterMax, 0);
                const uint64_t counterDiff = writeCounter.load() - readCounter.load();
                if (counterDiff > 0) {
                  std::cout << "unprocessed packets: " << counterDiff << " (written: " << writeCounter.load() << ", read: " << readCounter.load() << ")" << std::endl;
                  //std::cout << "buffer active items: " << lidar_packets->size() << std::endl;
                }

                on_lidar_packet_msg(buffer);
            });
```
> It wouldn't be possible to call the `lidar_packets->size()` directly, since it would cause a dead-lock. But the resulting difference gives us the same thing.

and, again, add the same thing to the community implementation, then we get the following output:
* CPU: `AMD Ryzen 7 3700X @ 4GHz`
* mode: `1024x20 @ RNG19_RFL8_SIG16_NIR16` (network data rate: `~260Mbps`)
  * ouster_ros (this)
    <details>
    <summary>zero subs, </summary>

    ```
    [os_driver-1] unprocessed packets: 1 (written: 1491, read: 1490)
    [os_driver-1] unprocessed packets: 5 (written: 1497, read: 1492)
    [os_driver-1] unprocessed packets: 4 (written: 1497, read: 1493)
    [os_driver-1] unprocessed packets: 3 (written: 1497, read: 1494)
    [os_driver-1] unprocessed packets: 2 (written: 1497, read: 1495)
    [os_driver-1] unprocessed packets: 1 (written: 1497, read: 1496)
    [os_driver-1] unprocessed packets: 2 (written: 1556, read: 1554)
    [os_driver-1] unprocessed packets: 1 (written: 1556, read: 1555)
    ```
    </details>
    <details>
    <summary>one sub, </summary>

    ```
    [os_driver-1] unprocessed packets: 1 (written: 3482, read: 3481)
    [os_driver-1] [1708368385378405204] missing 14 packets (last m_id: 191, new m_id: 416, time diff:     450404 ns)
    [os_driver-1] unprocessed packets: 1 (written: 3483, read: 3482)
    [os_driver-1] [1708368385428436651] missing 14 packets (last m_id: 191, new m_id: 416, time diff:       7674 ns)
    [os_driver-1] unprocessed packets: 8 (written: 3533, read: 3525)
    [os_driver-1] unprocessed packets: 7 (written: 3533, read: 3526)
    [os_driver-1] unprocessed packets: 6 (written: 3533, read: 3527)
    [os_driver-1] unprocessed packets: 5 (written: 3533, read: 3528)
    [os_driver-1] unprocessed packets: 4 (written: 3533, read: 3529)
    [os_driver-1] unprocessed packets: 3 (written: 3533, read: 3530)
    [os_driver-1] unprocessed packets: 2 (written: 3533, read: 3531)
    [os_driver-1] unprocessed packets: 1 (written: 3533, read: 3532)
    [os_driver-1] unprocessed packets: 8 (written: 3582, read: 3574)
    [os_driver-1] unprocessed packets: 7 (written: 3582, read: 3575)
    ```
    </details>
    <details>
    <summary>two subs, </summary>

    ```
    [os_driver-1] unprocessed packets: 1 (written: 1916, read: 1915)
    [os_driver-1] [1708368460842251588] missing 44 packets (last m_id: 191, new m_id: 896, time diff:     260444 ns)
    [os_driver-1] unprocessed packets: 1 (written: 1917, read: 1916)
    [os_driver-1] unprocessed packets: 7 (written: 1936, read: 1929)
    [os_driver-1] unprocessed packets: 6 (written: 1936, read: 1930)
    [os_driver-1] unprocessed packets: 5 (written: 1936, read: 1931)
    [os_driver-1] unprocessed packets: 4 (written: 1936, read: 1932)
[... 750 lines cut ...]
    [os_driver-1] unprocessed packets: 1123 (written: 4354, read: 3231)
    [os_driver-1] buffer active items: 1016
    [os_driver-1] unprocessed packets: 1122 (written: 4354, read: 3232)
    [os_driver-1] buffer active items: 1015
    [os_driver-1] unprocessed packets: 1121 (written: 4354, read: 3233)
    [os_driver-1] buffer active items: 1014
    [os_driver-1] unprocessed packets: 1120 (written: 4354, read: 3234)
    [os_driver-1] buffer active items: 1013
    [os_driver-1] unprocessed packets: 1119 (written: 4354, read: 3235)
    [os_driver-1] buffer active items: 1012
    [os_driver-1] unprocessed packets: 1118 (written: 4354, read: 3236)
    [os_driver-1] buffer active items: 1011
    [os_driver-1] unprocessed packets: 1117 (written: 4354, read: 3237)
    [os_driver-1] buffer active items: 1010
    [os_driver-1] unprocessed packets: 1116 (written: 4354, read: 3238)
    [os_driver-1] buffer active items: 1009
    [os_driver-1] unprocessed packets: 1115 (written: 4354, read: 3239)
    [os_driver-1] buffer active items: 1008
    [os_driver-1] unprocessed packets: 1114 (written: 4354, read: 3240)
    ```
    </details>

[... cut ...] s the community version - we're not missing any packets, but the buffer will overflow with too many subscribers).

Since the writing thread is no longer holding back on the packets, they are actually written into the buffer, which will overflow if the reading thread can not keep up. I think you mention separating out the reading and processing into different threads (https://github.com/ouster-lidar/ouster-ros/issues/240#issuecomment-1939514177), which could be a  solution for the second problem.

mwhannan74, 2024-03-29:
I would like to comment that I am also having this issue with my Ouster OS-2-128, Firmware v2.5.2, UDP Profile LEGACY, pulled ROS2 branch today. 

If I use 2048x10 mode, than I see lost data from about 270 deg to 360 deg (https://static.ouster.dev/sensor-docs/_images/lidar-frame-top-down.svg). Sometimes less, sometimes flickers to almost 240 deg. This makes it practically unusable.

I can confirm that if I switch to 1024x10 that I do NOT see this issue. This is an okay work around for now, but not ideal for sure.

I can also confirm that this is NOT an issue when using Ouster studio.

mwhannan74, 2024-04-01:
I did some more testing today, and though the flickering and lost data is a lot less, I can still see it at 1024x10.

shubham-shahh, 2024-04-01:
> I did some more testing today, and though the flickering and lost data is a lot less, I can still see it at 1024x10.
> 
> Is this an issue with me using an older model of the OS-128 or the UDP LEGACY mode?
> 
> I am a bit baffled that everyone is not seeing this issue and that there is not a clear understanding why. It is easily reproducible for my setup. 

Are you building the ouster driver in Release mode?

mwhannan74, 2024-04-02:
Yes I am building in release.
>> colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release

Here is a video (I hope) showing the issue at 2048x10.
[Screencast from 04-01-2024 08:51:12 PM.webm](https://github.com/ouster-lidar/ouster-ros/assets/15201656/49fc58df-81cc-48d3-9b9e-a2431e6ddeb0)

I ran Ouster Studio (https://ouster.com/products/software/ouster-studio) and I do not have any issues, I am assuming that this is a ROS2 issue.

I checked the CPU usage, and it is not high.

mwhannan74, 2024-04-02:
I switched the UDP profile from LEGACY to RNG19_RFL8_SIG16_NIR16, but still the same flickering.

mwhannan74, 2024-04-02:
Here it is at 1024x10. It is better, but there is still a small flicker occasionally. 
[Screencast from 04-01-2024 09:06:42 PM.webm](https://github.com/ouster-lidar/ouster-ros/assets/15201656/3c7d0d07-7e1e-4aa1-b6e4-1fb8fe3aca22)

shubham-shahh, 2024-04-02:
> Here it is at 1024x10. It is better, but there is still a small flicker occasionally. [Screencast from 04-01-2024 09:06:42 PM.webm](https://github.com/ouster-lidar/ouster-ros/assets/15201656/3c7d0d07-7e1e-4aa1-b6e4-1fb8fe3aca22)

what's the platform on which you're testing this? did you try out with some powerful CPUs to eliminate performance throttling?

mwhannan74, 2024-04-02:
Laptop with i9-9980HK (2.4GHzx16) with 32GB ram. Not the fastest, but should be more than plenty. I don't have a faster computer with ROS2 available. I was processing a Velodyne HDL64 using ROS1 over 8 years ago on a computer a fraction of this capability without issue. So, I hope that the hardware is not an issue. Again, Ouster studio works perfectly and I did not see any obvious CPU limiting when looking at the system monitor.

It could be something that I have done wrong with ROS2 or a driver configuration issue. ROS2 has a lot more nuance in setting it up then ROS1, especially with QoS.

I am launching it with.
> ros2 launch ouster_ros driver.launch.py

Here is the driver_params.yaml file that I am using. Maybe there is something wrong in here?
[driver_params.txt](https://github.com/ouster-lidar/ouster-ros/files/14839267/driver_params.txt)

Thanks for you help. Sorry if I seem a bit frustrated. I finally have a project to use this amazing lidar, and with it not working correctly in ROS2 is disappointing. I don't really have the time/money to start from scratch with the Ouster SDK.

shubham-shahh, 2024-04-03:
> Laptop with i9-9980HK (2.4GHzx16) with 32GB ram. Not the fastest, but should be more than plenty. I don't have a faster computer with ROS2 available. I was processing a Velodyne HDL64 using ROS1 over 8 years ago on a computer a fraction of this capability without issue. So, I hope that the hardware is not an issue. Again, Ouster studio works perfectly and I did not see any obvious CPU limiting when looking at the system monitor.
> 
> It could be something that I have done wrong with ROS2 or a driver configuration issue. ROS2 has a lot more nuance in setting it up then ROS1, especially with QoS.
> 
> I am launching it with.
> 
> > ros2 launch ouster_ros driver.launch.py
> 
> Here is the driver_params.yaml file that I am using. Maybe there is something wrong in here? [driver_params.txt](https://github.com/ouster-lidar/ouster-ros/files/14839267/driver_params.txt)
> 
> Thanks for you help. Sorry if I seem a bit frustrated. I finally have a project to use this amazing lidar, and I have spent a lot of time this last week or two trying to get it to work. I don't really have the time/money to start from scratch with the Ouster SDK.

no worries. @Samahu are there any updates concerning solving this issue on ROS2?

mwhannan74, 2024-04-03:
Here is what I see in Ouster Studio with 2048x10. It looks pretty good with no large regions of data mission. I do see some very minor flickering here and there (really only noticeable in the image), which is worrying. However, overall it looks significantly better than the ROS2 version.
[Screencast from 04-03-2024 11:36:49 AM.webm](https://github.com/ouster-lidar/ouster-ros/assets/15201656/b8970db2-071f-40d5-8a61-7bc7899d57db)

Samahu, 2024-04-16:
Hi All, please consider taking a look at any of the three following PRs (#319, #320, #321) whichever applies to your current ROS distribution and check if it helps resolve this issue without a regression in performance. I did take close look at #302 approach solution and I think it does help too but I couldn't verify it completely because under ROS2-foxy which is what @Imaniac230 targeted for the solution I don't seem to be to reproduce the issue on my laptop no matter how many subscribers I added. In the fix that I am submitting I use less synchronization overall. I have provided my analysis of the issue and the motivation behind this approach.

I do have one follow up item before I consider the solution complete but what I have pushed so far still resolves this problem.

Please if you can give it a try and provide feedback whether this helps with your situation or not (or any relevant feedback). :pray:

mwhannan74, 2024-04-16:
@Samahu, I tested PR #321. I have good and bad news. I really appreciate your help with this. 

The good news is that it does indeed perform better, a lot better. I can now run 1024x10 while actually processing the point cloud with voxel filtering, normal estimation and filter, and ICP. The flickering and lost data seems to be gone. 

However, the bad news is that at 2048x10 I still have issues. It is significantly better where I am only seeing minor drops of data and not huge chunks, but I still see the flickering. It is minimal when just running the lidar driver alone, but it does exist. If I run my same processing, then the flickering is consistently an issue. Also, I see the output rate of the driver drop from a solid 10Hz at 1024x10 to more like 8.5Hz at 2048x10.

I would expect that a gaming laptop or desktop grade processor may be able to successfully run at 2048x10. I am not sure what exactly is happening in the processing of scan packets to a point cloud that requires so much CPU. My more moderate computer and laptop that I use for my USV clearly struggles. It is an understandable and workable issue for me, but clearly disappointing.

Samahu, 2024-04-16:
@mwhannan74 Thanks for sharing the feedback, while waiting for further feedback for others here.. have you considered increasing your system's recv buffer size by performing?

```bash
sudo sysctl -w net.core.rmem_default=26214400
sudo sysctl -w net.core.rmem_max=26214400
```

Also if you only need the point clouds consider setting the proc_mask launch parameter to `proc_mask:="PCL"` this would reduce the amount spent for processing LidarScan's and reduce the amount data published by the driver.

Samahu, 2024-04-16:
awesome, thanks for sharing!

Imaniac230, 2024-04-18:
> Hi All, please consider taking a look at any of the three following PRs (#319, #320, #321) - whichever applies to your current ROS distribution - and check if it helps resolve this issue without a regression in performance. I did take close look at #302 approach solution and I think it does help too but I couldn't verify it completely because under ROS2-foxy which is what @Imaniac230 targeted for the solution I don't seem to be able to reproduce the issue on my laptop no matter how many subscribers I add. In the fix that I am submitting I use less synchronization overall. I have provided my analysis of the issue and the motivation behind this approach.
> 
> I do have one follow up item before I consider the solution complete but what I have pushed so far still resolves this problem.
> 
> Please if you can give it a try and provide feedback whether this helps with your situation or not (or any relevant feedback). 🙏

@Samahu I'm using foxy in my default dev system, but if it helps with comparisons and evaluations in any way I can also quickly create target PRs with the same changes for humble (and possibly noetic).

Samahu, 2024-04-23:
> @Samahu I'm using foxy in my default dev system, but if it helps with comparisons and evaluations in any way I can also quickly create target PRs with the same changes for humble (and possibly noetic).

I can do a comparison using the foxy branch I created once I get back to it.

PR #319 (2024-05-28): Implement lock free ring buffer with throttling [NOETIC]
## Related Issues & PRs
- Closes #240
- Closes #272
- Related #302
- ROS2[FOXY]: #320
- ROS2[HUMBLE/IRON]: #321

## Problem Analysis
- When adding multiple subscribers or when having a bad network or low performing system, LidarScan composition and processing could eventually take more time, this leads to a build up in the packets buffer which will eventually be exhausted and the current mechanism of the thread_safe_ring_buffer::write_overwrite made things only worse. The solution can be in one of two things either ensure that the LidarS [... cut]
PR #321 (2024-05-28): Implement lock free ring buffer with throttling [HUMBLE/IRON]
## Related Issues & PRs
- Closes #240
- Closes #272
- Related #302
- ROS1[NOETIC] #319 
- ROS2[FOXY] #320 

## Problem Analysis
- When adding multiple subscribers or when having a bad network or low performing system, LidarScan composition and processing could eventually take more time, this leads to a build up in the packets buffer which will eventually be exhausted and the current mechanism of the thread_safe_ring_buffer::write_overwrite made things only worse. The solution can be in one of two things either ensure that the LidarS [... cut]
PR #320 (2024-05-28): Implement lock free ring buffer with throttling [FOXY]
## Related Issues & PRs
- Closes #240
- Closes #272
- Related #302
- ROS1[NOETIC] #319 
- ROS2[HUMBLE/IRON] #321 

## Problem Analysis
- When adding multiple subscribers or when having a bad network or low performing system, LidarScan composition and processing could eventually take more time, this leads to a build up in the packets buffer which will eventually be exhausted and the current mechanism of the thread_safe_ring_buffer::write_overwrite made things only worse. The solution can be in one of two things either ensure that the LidarS [... cut]
closed 2024-05-28 by Samahu