ouster-lidar/ouster-ros #247: Increase RCVBUF_SIZE to prevent package loss

tik0, 2023-11-01:
We experience package losses when it comes to some system load on our vanilla Ubuntu 20.04.
We reproducibly tested it with maxing out the system's load with `$ yes` while running:
```
roslaunch ouster_lidar sensor.launch sensor_hostname:=$OUSTER_OS1_SENSOR_HOSTNAME udp_dest:=$OUSTER_OS1_UDP_DESTINATION lidar_port:=$OUSTER_OS1_UDP_PORT_LIDAR 
``` 
The applications were compiled with `-O3` and `release` flags beforehand.
However, adapting the system's `rmem_default` and `rmem_max` to e.g., 20MB did not help, since the ouster_client sets the receive-buffer explicitly to 256kB in the [client.cpp](https://github.com/ouster-lidar/ouster_example/blob/d73079865ebadd961d7af81b235bb34747f7cb6a/ouster_client/src/client.cpp#L56)

Increasing `RCVBUF_SIZE` to 20MB as well (this value was just a random guess BTW!) solved all our problems and we don't experience any frame drops from now on.

However, I think that just increasing `RCVBUF_SIZE` is a bad practice.
I think, that this value should be exposed as an argument such that it can be configured concerning the user's demand.

**What do you think?** 

Example with package loss:
[image]

Example with increased `RCVBUF_SIZE ` buffer and no package loss:
[image]

tik0, 2023-11-06:
> I also had problems with packet loss in Ouster Studio and tshark. I got frustrated and just build a custom udp capture tool just with recv + fwrite for capturing and fread+sendto for replaying.

Hi @KenthJohan , this sounds nice. What other optimizations have done to the system? Have you increased the OS's socket buffers or NIC ring buffers?

closed 2023-11-09 by Samahu

KenthJohan, 2024-01-16:
> > I also had problems with packet loss in Ouster Studio and tshark. I got frustrated and just build a custom udp capture tool just with recv + fwrite for capturing and fread+sendto for replaying.
> 
> Hi @KenthJohan , this sounds nice. What other optimizations have you done to the system? Have you increased the OS's socket buffers or NIC ring buffers?

Hi sorry for late answer. I have not touched Ubuntu default network settings. But there is a difference between UDP recv and tshark. tshark captures at data link layer (2), UDP recv captures at transport layer (4). I believe more things can go wrong when using tshark because its capturing more layers and headers where as UDP recv only captures size and data.

PR #249 (2023-11-09): SW-5622: Bump up ouster_client to 20231031 release
## Related Issues & PRs
- related #10 
- related #240
- resolves #247 

## Summary of Changes
- Bump up ouster_client to 20231031 release
- Remove ouster-ros melodic from build checks since it is no longer compatible (due to spdlog)

## Validation
Ensure all functionality sustained with no drawbacks per testing procedures for ouster-ros
PR #262 (2023-11-18): SW-5623: Bump up ouster_client to 20231031 release
## Related Issues & PRs
- related #240
- resolves #247 

## Summary of Changes
- Bump up ouster_client to 20231031 release

## Validation
Ensure all functionality sustained with no drawbacks per testing procedures for ouster-ros
PR #263 (2023-11-18): Bump up ouster_client to 20231031 release
## Related Issues & PRs
- rolling/humble/iron equivalent #262
- related #240
- resolves #247 

## Summary of Changes
- Bump up ouster_client to 20231031 release

## Validation
Ensure all functionality sustained with no drawbacks per testing procedures for ouster-ros