ouster-lidar/ouster-ros #248: Filtering out dust/fog/smoke using DUAL RETURNS

Pallav1299, 2023-11-02:
1. We're using Ouster OS0-32 Rev-06 Lidar. We're using the latest ouster_ros driver in LEGACY mode. We recently realized that dust is being detected in the pointcloud. I know that we can get two pointcloud rostopics (points1, points2) by setting to DUAL RETURN mode in ROS driver, and leverage the second pointcloud topic to get pointcloud data without dust particles detection. But, how should I use the second pointcloud? 
2. Shouldn't the pointcloud comprise of points with second return whenever there are dual returns? Please provide reference documentation for the same.
3. Could you please also tell, which resolutions support DUAL RETURN mode?

**Platform:**
- Ouster Sensor? OS-0
- Ouster Firmware Version? v2.5.2
- ROS version/distro? melodic 
- Operating System? Linux
- Machine Architecture? arm64
- git commit hash (if not the latest).

Samahu, 2023-11-06:
@Pallav1299 I agree there is a lack of documentation/support on how to utilize the second return to filter smoke or dust and other particles. But in principle when dual mode is enabled the first return penetrates through fog/smoke etc. The second return has measurements that correspond to less dense objects such as tree leaves, smoke, maybe glass, etc. You could enable the dual return in the ROS driver and check what type of objects gets captured in the second point cloud and how it differs from the first point cloud (both point clouds are visualized with different colors in RVIZ). It is currently left open to the user on how they want to utilize the second point cloud.

Pallav1299, 2023-11-16:
@Samahu, thanks for your response. I saw this [**THREAD**](https://forum.ouster.at/d/130-introducing-the-l2x-chip-supporting-dual-returns) and this [**VIDEO**](https://www.youtube.com/watch?v=ofO6ybzHmzY) of Ouster's Dirt drive performance. It seems to be performing quite good. 

Questions:
1. Can you please tell what approach was used in the video to filter out the dirt/dust from the raw pointcloud? Was only dual return used for this?
2. Is it possible to share this Dirt Drive, Smoke and Fog test data(ROSBAG) and ouster configuration to test it ourselves? It would be very helpful.
3. Is there a measurement metric for fog/smoke/dust density, under which Ouster performs good? Or is it general?

Samahu, 2023-11-20:
@Pallav1299 I am asking around if we can publicly share access to the raw data of the recording that you referenced. According to one of my colleagues the capture didn't utilize any specific technique to filter out the dust, this was the raw output of the sensor.

I could also refer you to some additional literature about the subject (which may or may not apply to our sensors):
- [Design of Dust-Filtering Algorithms for LiDAR Sensors Using Intensity and Range Information in Off-Road Vehicles](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC9185448/)
- [Mapping with Reflection - Detection and Utilization of Reflection in 3D Lidar Scans](https://arxiv.org/pdf/1909.12483.pdf)
- [Full Waveform LiDAR for Adverse Weather Conditions](https://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=9076331)

Pallav1299, 2023-11-25:
@Samahu, sorry for not replying sooner.
I'm attaching link to compressed ROSBAG data that we collected using OS-32 in Dual_Return profile.
[**DATA_LINK**](https://www.dropbox.com/scl/fi/fz3mtjpm5mq9qv619sxiw/os32_dust_dual_return.bag?rlkey=9m8u1odim19h1lhgduv9icchk&dl=0)

I want to understand what’s leading to different results in our data and the DIRT DRIVE data from Ouster related to more dust particles being detected.

We seek your insights regarding dust/smoke/fog filtering to handle complex use-cases.

Samahu, 2023-11-27:
@Pallav1299 You should take advantage of the signal/intensity channel in both returns since it response is weak to smoke and fog. The PointCloud(s) generated by ouster-ros driver directly embeds this information for each point (refer to https://github.com/ouster-lidar/ouster-ros/blob/master/include/ouster_ros/os_point.h). Utilizing this information and picking the right threshold will enable you to filter unwanted particles from the scene. Simply relying on the raw xyz data from the range data won't be sufficient.

Pallav1299, 2023-12-04:
@Samahu, Thanks for sharing the resources and insights. 

- In dual return mode we get two pointclouds, strongest and second strongest? What's the measure of strongest return? Is it the intensity/signal channel?
> Utilizing this information and picking the right threshold will enable you to filter unwanted particles from the scene. Simply relying on the raw xyz data from the range data won't be sufficient.

- Does it mean for different type of environment we've to do intensity/signal channel thresholding to filter out unwanted particles? How can we make it more general?

> * Design of Dust-Filtering Algorithms for LiDAR Sensors Using Intensity and Range Information in Off-Road Vehicles

- We're tried this but it runs offline for us. Trying to make it run realtime for our use case.

> 3\. Is it possible to share this Dirt Drive, Smoke and Fog test data(ROSBAG) and ouster configuration to test it ourselves? It would be very helpful.

- Can this data be shared publicly for testing purposes?

mitch-galea, 2024-12-22:
My findings for filtering out dust with dual returns using furthest return priority is that it works if there is something behind the dust that gives off a return. 
Our ouster is mounted around 2m off the ground and is pitched down at 20 degrees. On a relatively flat surface this resulted in dust close to the ground not being `filtered out` in the `/ouster/points` cloud. However dust that was near the sensor height with nothing behind it to give off a second return was not filtered out. This makes complete conceptual sense
Here is an image showing this with `/ouster/points` in purple and `/ouster/points2` in green  
[image]