ouster-lidar/ouster-ros #251: Bring back optional publishing of lidar and imu packets to new single nodelet driver

skohlbr, 2023-11-08:
**Is your feature request related to a problem? Please describe.**
We recently upgraded to use the new single nodelet driver (started via driver.launch as described in https://github.com/ouster-lidar/ouster-ros#sensor-mode ). This shows significantly reduced CPU consumption on Jetson embedded systems compared to the "old" multi-nodelet setup. However, it appears the support to simultaneously publish the lidar and imu packets has been dropped. This is a problem because for data recording purposes, those are the most useful data to record (as also remarked by others in https://github.com/ouster-lidar/ouster-ros/issues/110 ).

**Describe the solution you'd like**
It would be great to have the single driver nodelet also optionally (i.e. `if numSubscribers() > 0` ) publish the raw packets, so they can be recorded for later postprocessing/analysis.

**Describe alternatives you've considered**
An alternative is using the old multi-nodelet setup, but that one causes high CPU load on Jetson embedded systems, so not a good option.

**Targeted Platform (please complete the following information only if applicable, otherwise dot N/A):**
- ROS1, but likely applies to ROS2 also (?)

Samahu, 2023-11-08:
Same as #238

skohlbr, 2024-02-20:
@Samahu can you provide an indication if this is in the works or what the plans are wrt this feature?

PR #363 (2024-09-19): [NOETIC] Further reduce overhead in the ros driver
## Related Issues & PRs
- resolves #10 
- resolves #152 
- related #175 
- resolves #238 
- resolves #251 
- related #259
- resolves #304
- related #306 
- related #352

## Summary of Changes
- Add Jazzy to the build status
- Update to SDK client 0.11.1
- Switch to using LidarPacket and ImuPacket
  - Get's rid of the deprecation warnings
- Publish raw packets for the `os_driver`
- Add a flag to control the publish of organized point cloud vs non-organized
- Add a flag to control the publish of destaggerd point cloud vs staggered
- Add paramete [... cut]
PR #369 (2024-09-20): [HUMBLE|IRON|JAZZY] Port ROS-363 to ROS2
## Related Issues & PRs
- ported_from #363 
- related #10 
- related #152 
- related #175 
- resolves #238 
- resolves #251 
- resolves #259
- resolves #304
- related #306 
- resolves #352

## Summary of Changes
- Update to SDK client 0.11.1
- Switch to using LidarPacket and ImuPacket
  - Get's rid of the deprecation warnings
- Publish raw packets for the `os_driver`
- Add a flag to control the publish of organized point cloud vs non-organized
- Add a flag to control the publish of destaggerd point cloud vs staggered
- Add parameter to set the min/max l [... cut]
closed 2024-09-19 by Samahu