ouster-lidar/ouster-ros #256: Ouster Data Quality Issues

satyajitghana, 2023-11-15:
I have a OS1-32

Today i was just collecting some static data, no movement, just static

[image]

[image]

[image]

[image]

The Point Cloud is really weird, it like streaks of points. instead of a single point.

Ideally we expect noise in the points, but not like this, it should have been a grid of points, not a grid of streak lines. Its like a smudged Image.

this was resulting in a very thick point cloud if we do any sort of SLAM.

So we were just taking a a look at the RAW data.

Is this normal?

Samahu, 2023-11-16:
What kind of visualizer are you using to represent the LidarScan? Is this coming through RVIZ

satyajitghana, 2023-11-16:
Yes this is coming from RVIZ. So we've kept the LiDAR static and recorded bag file. Converted to PCD. I was thinking this could be due to vibrations. But looking at the pattern it doesn't seem so. It looks like something is wrong in the LiDAR output

Samahu, 2023-11-18:
I have never seen such an output on RVIZ, I advise you to open a [support ticket](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86)  because there could be some hardware issue at play. (and share the bag file for in-depth analysis of the problem).

Samahu, 2023-11-28:
@satyajitghana I don't see this to be a software problem but could a hardware issue. Please open a support ticket and link this PR to it.

closed 2023-11-28 by Samahu