ouster-lidar/ouster-ros #259: Image gaps while LidarScan from LidarPackets

rmbsmit, 2023-11-16:
**Describe your question**
I have recorded a bagfile containing LidarPackets (type PacketMsg.msg). From this data, i am trying to batch them, recreate a LidanScan using the ScanBatcher and view the Intensity/Reflectivy/Range images in python. However, when looking at the final images, there seems to be a problem in either the batching of the LidarPackets, or in the destaggering. The Range output image looks like this: 

[image]

The image itself seems fine, however there are gaps in between the packages. 

The sample code look like this: 
```
        self.batcher = client._client.ScanBatcher(self.metadata)
        self.ls = client.LidarScan(height,
                          width,
                          self.metadata.format.udp_profile_lidar)

def Callback(self, msg):
packet = client.LidarPacket(msg.buf, self.metadata, _raise_on_id_check=True)
if self.batcher(packet, self.ls):
            ref_field = self.ls.field(client.ChanField.SIGNAL)
            ref_val = client.destagger(self.metadata, ref_field, inverse=False)
            ref_img = (ref_val / np.max(ref_val) * 255).astype(np.uint8)
```

**Platform (please complete the following information):**
- Ouster OS2
- ROS version: noetic
- Operating System: Ubuntu 20.04

Can somebody help me solve the issue?

Samahu, 2023-11-16:
Do you get the same gaps when replaying the bagfile through the driver?

rmbsmit, 2023-11-17:
No, replaying the bagfile, the images look fine.

[image]

rmbsmit, 2023-12-04:
@Samahu do you have any update on this issue?  Thanks in advance

Samahu, 2023-12-04:
I don't have an update but looking briefly at your code I am wondering if you actually filter out ImuPacket before trying to parse the packet as a LidarPacket.

bexcite, 2023-12-05:
Hi @rmbsmit,

There are two things that I would check:

1) BAG lidar_packets content has no "holes" - i.e. all lidar_packets were received without drops and the network/system wasn't experiences the load or other form of disruptions that prevented the HOST machine to reveive all packets.
(to verify it you can use `ptudes viz` tool that I'll descrive below)

2) The ScanBatching routine that you are using is not broken. Usually using the `ouster.client.Scans` interface is a good starting point that will give you a lot of mileage (see in the example snippet below for the code sample).

I've recently solved the Ouster raw packets reading from BAGs (needed to get the Newer College 2021 dataset up) and here is my resulting `ptudes.bag.OusterRawBagSource` packet source reader: https://github.com/bexcite/ptudes-lab/blob/main/src/ptudes/bag.py#L16

You can try it with your bag by:

1) Install `ptudes-lab` from pypi (just random tools that I'm developing for my own experiments):

```
pip install ptudes-lab
```

2) Visualize the bag (for quickly check packets drops, keep attention on top 2d images):
```
ptudes viz path_to_the_bag.bag --meta path_to/metadata.json
```

3) Build you own processing using the snippet below:
```
from typing import Optional

import ouster.client as client
import ouster.viz as viz
from ouster.sdk.util import resolve_metadata

from ptudes.utils import read_metadata_json
from ptudes.bag import OusterRawBagSource

def main(bag_path: str, meta_path: Optional[str] = None):
    """Example on reading Ouster raw data ROS bags"""

    meta_path = resolve_metadata(bag_path, meta_path)
    # NOTE: some hack inside this method for enabling Newer College 2020
    # dataset :(. (you may use a vanilla Ouster SDK approach if don't need it)
    info = read_metadata_json(meta_path)

    packet_source = OusterRawBagSource(bag_path, info)
    scans = client.Scans(packet_source)

    # Option 1: Iterate over scans and get client.LidarScan object
    #           and process to you liking
    # for ls in scans:
        # range_field = ls.field(client.ChanField.RANGE)
        # # process, viz, etc the range_field data
        # print("ls = ", ls)  # also shows a good output for what is inside

    # Option 2: Visualize scans from the BAG to quickly spot the
    # drops by reviewnig 2d range images (top)
    viz.SimpleViz(info, rate=1.0).run(scans)

if __name__=="__main__":
    bag_path = "/home/2021-07-01-11-31-35_0-quad-medium.bag"
    meta_path = "/home/beam_intrinsics_os0-128.json"
    main(bag_path, meta_path)
```

Hope it helps,

-- Pavlo

PR #369 (2024-09-20): [HUMBLE|IRON|JAZZY] Port ROS-363 to ROS2
## Related Issues & PRs
- ported_from #363 
- related #10 
- related #152 
- related #175 
- resolves #238 
- resolves #251 
- resolves #259
- resolves #304
- related #306 
- resolves #352

## Summary of Changes
- Update to SDK client 0.11.1
- Switch to using LidarPacket and ImuPacket
  - Get's rid of the deprecation warnings
- Publish raw packets for the `os_driver`
- Add a flag to control the publish of organized point cloud vs non-organized
- Add a flag to control the publish of destaggerd point cloud vs staggered
- Add parameter to set the min/max l [... cut]