ouster-lidar/ouster-ros #269: pointcloud2 data and LaserScan data at the same time

7DoF, 2023-11-23:
Hi, I'm using Ouster OS0-32 Rev-06 Lidar. I'd like to know if there is a possibility that we can output both pointcloud2 3d data and LaserScan data using pro_mask field.

Thanks and regards.

**Platform (please complete the following information):**
- Ouster Sensor? OS-0
- Ouster Firmware Version? v2.5.2
- ROS version/distro? melodic
- Operating System? Linux
- Machine Architecture? arm64
- git commit hash (if not the latest).

Samahu, 2023-11-23:
@7DoF It is already possible to enable/disable both **PointCloud2** and **LaserScan** using the `proc_mask` flag through launch file arguments which you could get a list of by following the instructions noted in this section: https://github.com/ouster-lidar/ouster-ros#launch-files-arguments and you can find the proc_mask is on of the arguments along with its use.

It is worth noting that the main branch for ROS1 has an issue in which  the published **LaserScan msg** has some [misalignment issue](#241) with the produced point cloud which will be addressed once PR #224 is reviewed and merged. Which seems great but I see that you are on ROS melodic which we have stopped supporting starting with version [v0.10.0](https://github.com/ouster-lidar/ouster-ros/releases#:~:text=%5BBREAKING%5D%3A%20with%20this%20release%20the%20ouster%2Dros%20driver%20is%20no%20longer%20compatible%20with%20ROS%20melodic). You would need to upgrade to using noetic distro to take advantage of the pending fix.

closed 2023-11-23 by Samahu

7DoF, 2023-11-27:
Hi, @Samahu , thank you for the clarification.