ouster-lidar/ouster-ros #270: How to replay lidar data recorded under ros1 with ros2

rdouguet, 2023-11-27:
**Describe your question**

I'm trying to replay lidar data recorded under ros1 (melodic) in the ros2 (Humble) environment with the Ouster driver for ROS2. So far, I still haven't succeeded but I don't know if I'm going in the right direction.

Here are the step for replaying a recording :
First, I convert my ROS1 recording file to ROS2 using the rosbags tool (https://pypi.org/project/rosbags/)
Next, I compile the ros ouster package so that normally ROS is able to recognize lidar custom messages.
But when I start the replay of this new file, the ouster topics are not recognized and I have the following error :

**Screenshots**
[image]

I work the same way with other sensors and have never had this problem. 

**Platform**
- Ouster Sensor: OS1-64
- (Logger) ROS version/distro/package: melodic / Ubuntu 18.04 / ouster_ros=0.2.1
- (replay) ROS version/distro/package: humble / Ubuntu 22.04 / ouster_ros=0.12.0

Samahu, 2023-11-27:
@rdouguet I didn't try to convert from ROS1 to ROS2 but when doing the opposite I had to add and register the message definitions within rosbags then I was successful with the conversion. I believe you have to do the same thing.

rdouguet, 2023-11-30:
Finally is not really simple as problem, I did not find easy way...

Maybe we can use "rosbags tools" to add custom message and analyzing the rosbag with python scripts... but my goal is to use old rosbag(ros1) to test new fusion algorithm (develop under ros2) between some sensors such as Lidar, camera and radar.

One solution that I haven't yet had time to finalize is to follow the steps explain at the start (convert ros1 to ros2 and next compile ros ouster package). Finally, you can change manually the type of message in new dB3 log... to replace the type "/ouster_ros/msg/PacketMsg" by "/ouster_sensor_msgs/msg/PacketMsg". In this way, you can publish raw lidar data.

Now I must find how the new ros ouster node can subscribe to these topics to generate the point cloud ?

Samahu, 2023-11-30:
To find how to read load and read the packets from a bag file you should checkout the file `replay.launch.xml` (or `replay.launch` for ROS1).

Samahu, 2023-12-06:
@rdouguet I only suggested that you examine `replay.launch` so you could implement your solution based on it.

If you examine the file closely (using ROS2 version here):
https://github.com/ouster-lidar/ouster-ros/blob/66d6b9ff104693c1b24397dd0b66d67bd0c3d4e4/ouster-ros/launch/replay.composite.launch.xml#L65-L81

You can see that along with the **os_replay** node, we construct an **os_cloud** and **os_image** nodes (components). These two nodes process the raw lidar/imu packets when published on the default `/ouster/lidar_packets` and `/ouster/imu_packets` topics. which is what the rosbag play command is doing.

If you need to have more than one senosr then you would need to a pair of **os_cloud** and **os_image** nodes for each sensor that you intend to replay and utilize [topic remapping](https://design.ros2.org/articles/static_remapping.html) to set each pair differently for each sensor.

rdouguet, 2023-12-06:
Thanks for your help, I can finally replay the data provided by an old rosbag :)

To make it work, I removed the part that launches the ros2 bag from the launch file and deleted the "bag_file" arg. I directly renamed my topics from the *.db3 log file ("/os_node/lidar_packets" to "**/ouster/lidar_packets**" and /os_node/imu_packets" to **"/ouster/imu_packets"**).

But I suppose it would also work if I remapped these topics with the remapping function.

Samahu, 2023-12-06:
> Thanks for your help, I can finally replay the data provided by an old rosbag :)
>
> To make it work, I removed the part that launches the ros2 bag from the launch file and deleted the "bag_file" arg. [...]

The topic prefix `/os_node/` is quite old more than 18 months since I actually took over and normalized all topics under a single default namespace `/ouster/`.. I totally forgot about it. happy to learn that you managed to convert the bag files and replay them.

Rotoslider, 2024-09-21:
Is there a converter now for playing bags created with a device running Ros1 Noetic that needs to be played back on a different machine with Ros2 Humble? The machine with Ros2 does not have Ros1 on it.

Samahu, 2024-09-23:
@Rotoslider please checkout these two helper scripts that allows you to switch between ROS1/ROS2 using rosbags library https://github.com/ouster-lidar/ouster-ros-extras/pull/2. Note that I have developed these scripts for my own use and they are not production ready.

Hope this helps!