ouster-lidar/ouster-ros #272: Parts of point cloud missing

shubham-shahh, 2023-11-29:
**Describe the bug**
Hi, We are using OS-0 32-channel Rev6 lidar, We have found instances, where significant chunks of the point cloud are missing. I have attached a screen recording below which shows the anomalous behaviour. 

I have already gone through the issue mentioned [here](https://github.com/ouster-lidar/ouster-ros/issues/240)
but I am not sure If the cause is the same.

There are 3 possible points of failures 
1) Power supply - We are querying the sensor's telemetry to check for the V, A, and temperature

- status: "\"RUNNING\""
- temperature: 35
- input_voltage_mv: 1151.0
- input_current_ma: 1759.0 

The usual values of current and voltage are mentioned above, can you please mention the exact cutoff values below which the sensor would stop functioning, the docs mention 12V but even with Ouster's provided power supply we didn't get telemetry reporting 12V

2) Ethernet Switch - Can you point out some methods to detect packet loss or bandwidth throttling to isolate if the ethernet switch is the culprit?

3) Driver - Any method to isolate if there are any issues related to the driver?

**Setup**
The Lidar and the edge device are connected to a Gigabit Ethernet switch and the lidar is connected to a constant 12V power supply

**Platform (please complete the following information):**
- Ouster Sensor \[OS-0\]
- Ouster Firmware Version \[Latest\]
- Programming Language \[C++\]
 
- Operating System \[Ubuntu 18.04\]
- Machine Architecture \[arm]

https://github.com/ouster-lidar/ouster-ros/assets/51359628/e779e0ef-1800-4142-bb8d-b36b023f466f

Samahu, 2023-11-29:
@shubham-shahh I don't know much about power requirements of the sensor and how it could affect the sensor operation, but I will ask around. The sensor does produce some diagnostics and other telemetry but so far I hadn't had the chance to integrate that into the driver.

Samahu, 2023-11-29:
Is this ROS1 or ROS2?

Samahu, 2023-11-29:
The sensor does not limit or has any cut off based on Power according to the feedback I got.
I have a question however, when the sensor exhibits this behavior do you see any warnings or errors listed on the diagnostics page?

shubham-shahh, 2023-11-30:
> Is this ROS1 or ROS2?

Hi, Thanks for the response, we are using ROS1 Melodic

shubham-shahh, 2023-11-30:
> The sensor does not limit or has any cut off based on Power according to the feedback I got. I have a question however, when the sensor exhibits this behavior do you see any warnings or alerts listed on the diagnostics page?

I have not noticed it, but I will reproduce the issue and see if I get any warnings on the diagnostics page

Samahu, 2023-11-30:
> > Is this ROS1 or ROS2?
> 
> Hi, Thanks for the response, we are using ROS1 Melodic any inputs on point 2? such that I can isolate if the issue is related to the switch?
> 
> thanks

I am not sure if you have seen this [Ouster Network Analysis Guide](https://static.ouster.dev/sensor-docs/image_route1/image_route3/appendix/Analyzing-Linux.html)

TimeEscaper, 2023-12-06:
Hi all, I noticed this issue few days ago too. In my case, I faced it when had multiple subscribers on /ouster/points topic (2 RVIZ instances, 1 3D object detection node).

Samahu, 2023-12-06:
> Hi all, I noticed this issue few days ago too. In my case, I faced it when had multiple subscribers on /ouster/points topic (2 RVIZ instances, 1 3D object detection node).

@TimeEscaper the issue you are describing seems more like this one #240. other users have reported missing parts of the point cloud when the point cloud topic is consumed by more than one subscriber.

shubham-shahh, 2024-02-07:
can I use the method mentioned [here](https://github.com/ouster-lidar/ouster_example/issues/323#issuecomment-1802193989) to check if the scan is incomplete? if yes, it'd be great if you could share an example of how I can leverage it to check if the scan is complete or not before publishing.

thanks

shubham-shahh, 2024-02-07:
Hi @Samahu 
adding  `SET(CMAKE_BUILD_TYPE "Release")` in `ouster-ros/CMakeLists.txt` solved the issue for me. Is there any reason why it doesn't exist in the default config? 

thanks

Samahu, 2024-02-08:
> Hi, @Samahu can I use the method mentioned [here](https://github.com/ouster-lidar/ouster_example/issues/323#issuecomment-1802193989) to check if the scan is incomplete? if yes, it'd be great if you could share an example of how I can leverage it to check if the scan is complete or not before publishing.
> 
> thanks

The [client_example](https://github.com/ouster-lidar/ouster_example/blob/1f73d012c489d71ba6c88f56c6726369e83c8b57/examples/client_example.cpp#L123) in our C++ examples shows how to invoke this method to check if a scan is complete or not:

Samahu, 2024-02-08:
> Hi @Samahu adding `SET(CMAKE_BUILD_TYPE "Release")` in `ouster-ros/CMakeLists.txt` solved the issue for me. Is there any reason why it doesn't exist in the default config?
> 
> thanks

Our [Getting Started](https://github.com/ouster-lidar/ouster-ros?tab=readme-ov-file#getting-started) section clearly states to build with `Release` option. I don't want to force users into a specific build option as it is a common that they want to alternate `Release` and the two other options `Debug` or `RelWithInfo`.

But yeah glad the issue has been resolved. Will try to find some way to hint to users that `Release` is almost needed for the driver(s) to run normally.

PR #319 (2024-05-28): Implement lock free ring buffer with throttling [NOETIC]
## Related Issues & PRs
- Closes #240
- Closes #272
- Related #302
- ROS2[FOXY]: #320
- ROS2[HUMBLE/IRON]: #321

## Problem Analysis
- When adding multiple subscribers or when having a bad network or low performing system, LidarScan composition and processing could eventually take more time, this leads to a build up in the packets buffer which will eventually be exhausted and the current mechanism of the thread_safe_ring_buffer::write_overwrite made things only worse. The solution can be in one of two things either ensure that the LidarS [... cut]
PR #321 (2024-05-28): Implement lock free ring buffer with throttling [HUMBLE/IRON]
## Related Issues & PRs
- Closes #240
- Closes #272
- Related #302
- ROS1[NOETIC] #319 
- ROS2[FOXY] #320 

## Problem Analysis
- When adding multiple subscribers or when having a bad network or low performing system, LidarScan composition and processing could eventually take more time, this leads to a build up in the packets buffer which will eventually be exhausted and the current mechanism of the thread_safe_ring_buffer::write_overwrite made things only worse. The solution can be in one of two things either ensure that the LidarS [... cut]
PR #320 (2024-05-28): Implement lock free ring buffer with throttling [FOXY]
## Related Issues & PRs
- Closes #240
- Closes #272
- Related #302
- ROS1[NOETIC] #319 
- ROS2[HUMBLE/IRON] #321 

## Problem Analysis
- When adding multiple subscribers or when having a bad network or low performing system, LidarScan composition and processing could eventually take more time, this leads to a build up in the packets buffer which will eventually be exhausted and the current mechanism of the thread_safe_ring_buffer::write_overwrite made things only worse. The solution can be in one of two things either ensure that the LidarS [... cut]
closed 2024-05-28 by Samahu