ouster-lidar/ouster-ros #273: Invoking "make -j4 -l4" failed error is occurred while I follow the Getting Started.

CYJ00, 2023-12-08:
**Describe the bug**
The error below is occurred while I was doing the step(catkin_make --cmake-args -DCMAKE_BUILD_TYPE=Release).

[ 65%] Building CXX object ouster-ros/ouster_example/ouster_client/CMakeFiles/ouster_client.dir/src/parsing.cpp.o
[ 68%] Building CXX object ouster-ros/ouster_example/ouster_client/CMakeFiles/ouster_client.dir/src/sensor_http.cpp.o
/home/unitree/Documents/OSDome_ws/src/ouster-ros/ouster-sdk/ouster_client/src/parsing.cpp: In member function ‘uint32_t ouster::sensor::packet_format::frame_id(const uint8_t*) const’:
/home/unitree/Documents/OSDome_ws/src/ouster-ros/ouster-sdk/ouster_client/src/parsing.cpp:443:50: warning: ‘uint16_t ouster::sensor::packet_format::col_frame_id(const uint8_t*) const’ is deprecated: Use frame_id instead [-Wdeprecated-declarations]
         return col_frame_id(nth_col(0, lidar_buf));
                                                  ^
In file included from /home/unitree/Documents/OSDome_ws/src/ouster-ros/ouster-sdk/ouster_client/include/ouster/impl/packet_writer.h:8:0,
                 from /home/unitree/Documents/OSDome_ws/src/ouster-ros/ouster-sdk/ouster_client/src/parsing.cpp:18:
/home/unitree/Documents/OSDome_ws/src/ouster-ros/ouster-sdk/ouster_client/include/ouster/types.h:1102:53: note: declared here
     [[deprecated("Use frame_id instead")]] uint16_t col_frame_id(
                                                     ^~~~~~~~~~~~
In file included from /home/unitree/Documents/OSDome_ws/src/ouster-ros/ouster-sdk/ouster_client/src/sensor_http.cpp:5:0:
/home/unitree/Documents/OSDome_ws/src/ouster-ros/ouster-sdk/ouster_client/src/curl_client.h: In member function ‘std::__cxx11::string CurlClient::execute_get(const string&, int, int) const’:
/home/unitree/Documents/OSDome_ws/src/ouster-ros/ouster-sdk/ouster_client/src/curl_client.h:95:43: error: no matching function for call to ‘spdlog::logger::warn(std::__cxx11::basic_string<char>, const string&, long int&, std::__cxx11::string&)’
                     url, http_code, buffer);
                                           ^
In file included from /home/unitree/Documents/OSDome_ws/src/ouster-ros/ouster-sdk/ouster_client/src/logging.h:3:0,
                 from /home/unitree/Documents/OSDome_ws/src/ouster-ros/ouster-sdk/ouster_client/src/curl_client.h:8,
                 from /home/unitree/Documents/OSDome_ws/src/ouster-ros/ouster-sdk/ouster_client/src/sensor_http.cpp:5:
/usr/include/spdlog/logger.h:43:53: note: candidate: template<class Arg1, class ... Args> void spdlog::logger::warn(const char*, const Arg1&, const Args& ...)
     template <typename Arg1, typename... Args> void warn(const char* fmt, const Arg1&, const Args&... args);
                                                     ^~~~
/usr/include/spdlog/logger.h:43:53: note:   template argument deduction/substitution failed:
In file included from /home/unitree/Documents/OSDome_ws/src/ouster-ros/ouster-sdk/ouster_client/src/sensor_http.cpp:5:0:
/home/unitree/Documents/OSDome_ws/src/ouster-ros/ouster-sdk/ouster_client/src/curl_client.h:93:54: note:   cannot convert ‘std::operator+(std::__cxx11::basic_string<_CharT, _Traits, _Alloc>&&, const _CharT*) [with _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>](((const char*)"[{}] with the code: [{}] - and return: {}"))’ (type ‘std::__cxx11::basic_string<char>’) to type ‘const char*’
                     std::string("Re-attempting CurlClient::execute_get after "
                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                 "failure for url: ") +
                                 ~~~~~~~~~~~~~~~~~~~~~^
                         "[{}] with the code: [{}] - and return: {}",
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/unitree/Documents/OSDome_ws/src/ouster-ros/ouster-sdk/ouster_client/src/logging.h:3:0,
                 from /home/unitree/Documents/OSDome_ws/src/ouster-ros/ouster-sdk/ouster_client/src/curl_client.h:8,
                 from /home/unitree/Documents/OSDome_ws/src/ouster-ros/ouster-sdk/ouster_client/src/sensor_http.cpp:5:
/usr/include/spdlog/logger.h:63:32: note: candidate: template<class T> void spdlog::logger::warn(const T&)
     template <typename T> void warn(const T&);
                                ^~~~
/usr/include/spdlog/logger.h:63:32: note:   template argument deduction/substitution failed:
In file included from /home/unitree/Documents/OSDome_ws/src/ouster-ros/ouster-sdk/ouster_client/src/sensor_http.cpp:5:0:
/home/unitree/Documents/OSDome_ws/src/ouster-ros/ouster-sdk/ouster_client/src/curl_client.h:95:43: note:   candidate expects 1 argument, 4 provided
                     url, http_code, buffer);
                                           ^
ouster-ros/ouster_example/ouster_client/CMakeFiles/ouster_client.dir/build.make:254: recipe for target 'ouster-ros/ouster_example/ouster_client/CMakeFiles/ouster_client.dir/src/sensor_http.cpp.o' failed
make[2]: *** [ouster-ros/ouster_example/ouster_client/CMakeFiles/ouster_client.dir/src/sensor_http.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:3265: recipe for target 'ouster-ros/ouster_example/ouster_client/CMakeFiles/ouster_client.dir/all' failed
make[1]: *** [ouster-ros/ouster_example/ouster_client/CMakeFiles/ouster_client.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

I installed all requirements and additional dependencies.
I followed the tutorial in Noetic before, and it worked without error. How can I fix it?

**Platform (please complete the following information):**
- OS-DOME-128
- Melodic
- Ubuntu 18.04
- x64/Jetson Nano

Samahu, 2023-12-08:
@CYJ00 starting with v0.10.1 under ROS1, the ouster-ros driver [dropped support for melodic](https://github.com/ouster-lidar/ouster-ros/discussions/252#:~:text=%5BBREAKING%5D%3A%20with%20this%20release%20the%20ouster%2Dros%20driver%20is%20no%20longer%20compatible%20with%20ROS%20melodic). Melodic is EOL, consider upgrading to noetic and this error will not take place.

closed 2023-12-08 by Samahu

CYJ00, 2023-12-09:
Oh.. I got it. Thank you.