ouster-lidar/ouster-ros #274: OS0 showing strange ghost ring on latest master branch?

alexjunholee, 2023-12-12:
Hi,
We're currently using a OS0-128 LiDAR on Ubuntu 20.04.
And we've experienced some weird LiDAR behaviors on this repository.

In ouster studio and latest release (0.10.0), it does not happens and everything looks fine.
However if i run the driver.launch with the provided master branch, it shows an angel ring like this:

and by the way, it also shows a strange reflections incurred by the cable itself.
Would this will be fixed in the next release? Thanks!

alexjunholee, 2023-12-12:
1. The angel ring
[image]

2. Removed angel ring, but now the cable 
[image]

Samahu, 2023-12-12:
@alexjunholee this is rather strange and unexpected, could you try turning off LaserScan message and see if that's something related? You can enable/disable the the LaserScan message topic through the [proc_mask](https://github.com/ouster-lidar/ouster-ros/blob/5a08f89e21dabfb78ae359f9bfc4ba869a275104/launch/sensor.launch#L57) launch file parameter.

Samahu, 2023-12-12:
With regard to the second image, it is not clear to me how the cable is contributing to the output. Can you highlight where the issue please, thank you.

Samahu, 2023-12-13:
@alexjunholee Can you add the sensor REV and FW in use to the description?

alexjunholee, 2023-12-13:
I am not sure of what REV and FW corresponds to,
but it seems like its REV7 (OS0) and we are using v3.0.1
with Image version ousteros-image-prod-bootes-v3.0.1+20230209044733
(as identical version when received the sensor)

:)

Samahu, 2023-12-13:
With regard to the interference caused by the cable connected to OS0 you could try utilizing a right angle connector, this should significantly reduce the interference.

Still waiting on your confirmation with regard to LaserScan ring

alexjunholee, 2023-12-14:
Thanks for you update, it seems like like it was a laserscan points.
It was publishing a scan ring 0 from the scan_pubs. No worries!

Samahu, 2023-12-14:
> Thanks for you update, it seems like like it was a laserscan points. It was publishing a scan ring 0 from the scan_pubs. No worries!

Cool, with regard to the artifacts around the connector area all the solutions that I could come up are dependent on your sensor setup:
* Consider choosing an azimuth window that skips the area towards the connector (this would discard an entire column)
* Consider applying a filter to remove the outlier points 
* As I suggested earlier consider using a right angle connector to reduce interference.

closed 2023-12-14 by Samahu