ouster-lidar/ouster-ros #278: Low-frequency lidar pointcloud messages with ROS2

SimonGiampy, 2024-01-11:
**Low-frequency and unstable lidar pointcloud messages transmission with ROS2 Humble and Iron**

The OS1-64 lidar with the ROS2 driver has very low frequency and unstable messages transmission. Sometimes the rate of messages transmission drops below 1Hz, but it is often unpredictable. Most of the time the behavior in data transmission is unstable: it sends messages "in a burst", meaning that it sends few messages, than drops silent for about 1 second, then restarts the transmission. Overall the lidar pointcloud transmission is not reliable at all, making the sensor unusable with ROS2. There are other times in which the lidar seems to work better, but this is unpredictable. We currently do not know whether the problem is at ROS Middleware level, or at ROS-wrapper driver level. The problem arises with both ROS2 Humble and Iron distributions, even in a fresh linux installation.

**What we tried**:
- Ouster Studio shows the Lidar working perfectly fine, even at 20Hz, so it is not a hardware malfunctioning problem
- Different middleware implementations showed the same problem. FastDDS shows missing pointcloud messages, dropping the transmission rate. CycloneDDS shows missing chunks of pointcloud, being received "in sectors" and not in their entirety.
- Changed resolution of the pointcloud: even the lowest 512x10 doesn't solve the issue
- LIDAR IMU messages received at correct rate of 100Hz
- Used realtime linux kernel but didn't change anything

**Platform**
- Ouster Sensor: OS1-64
- Ouster Firmware Version: v2.4.0
- ROS version/distro: ROS2 Humble & Iron
- Middleware DDS: FastDDs & CycloneDDS
- Operating System: Linux Ubuntu 22.04
- Machine Architecture: Intel 12th gen processor, x64
- git commit: latest
- lidar connected via ethernet to the PC

**Screenshot of the problem**
[image]

As you can see, this screenshot shows an unstable rate of reception of pointcloud messages. In this case it is around 4hz, but sometimes drops below 1Hz.

**Driver configuration parameters used**:
```yaml
ouster/os_driver:
  ros__parameters:
    
    sensor_hostname: 'os1-991942000600.local'

    udp_dest: ''
    
    mtp_dest: ''
   
    mtp_main: false
    
    lidar_mode: '1024x10'
    
    timestamp_mode: 'TIME_FROM_ROS_TIME'
    
    ptp_utc_tai_offset: -37.0
    
    udp_profile_lidar: 'RNG19_RFL8_SIG16_NIR16'
    
    metadata: ''
    
    lidar_port: 7502
    
    imu_port: 7503
    
    sensor_frame: os_sensor
    
    lidar_frame: os_lidar
    
    imu_frame: os_imu
    
    point_cloud_frame: os_lidar 
    
    proc_mask: PCL|IMU
    
    scan_ring: 0
    
    use_system_default_qos: false
    
    point_type: original
```

SimonGiampy, 2024-01-12:
**Solution for the problem found**

I'll write here the solution to the problem in case it can help someone else. I'll close the issue since I solved it.

The solution was using CycloneDDS to overcome the limitations in the transmission efficiency of the pointcloud messages, which are very heavy therefore occupy a lot of bandwidth. In our case scenario, it _initially_ didn't help because we would see only sectors of the pointcloud, as also other users pointed out in other github issues. The actual problem was the configuration of CycloneDDS, which was not optimal for our system. An important lesson learned is that the DDS matter a lot in ROS, so configuring its parameters is essential to have efficient high-bandwidth communication, especially in multi-robot scenarios.

In our case, the robot pc handles alone everything in ROS2, meaning that computation happens locally only, and no other client subscribes or publishes using the ros communication middleware in the local network. What happened in our case was that ROS2 checked for subscribers in the wifi network to which the robot PC was connected to via ethernet. So the communication was broadcast to another wifi access point, occupying uselessly a portion og the available bandwidth of the middleware, meaning that everything was slowed down and CycloneDDS had a hard time keeping up with the pointcloud data transmission.

The solution found in our case was to restrict the ROS middleware communication to localhost only, without broadcasting data to other networks. Using `export ROS_LOCALHOST_ONLY=1` didn't work in our case, so we found this configuration for CycloneDDS that does exactly that. Thanks to this fix, the lidar pointcloud streaming works flawlessly, even at 20Hz (also very stable).

```xml
<?xml version="1.0" encoding="UTF-8" ?>
<CycloneDDS xmlns="https://cdds.io/config" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="https://cdds.io/config https://raw.githubusercontent.com/eclipse-cyclonedds/cyclonedds/master/etc/cyclonedds.xsd">
  <Domain id="any">
    <General>
      <Interfaces>
        <NetworkInterface name="lo"/>
      </Interfaces>
      <AllowMulticast>false</AllowMulticast>
    </General>
    <Discovery>
      <ParticipantIndex>auto</ParticipantIndex>
      <Peers>
        <Peer Address="localhost"/>
      </Peers>
      <MaxAutoParticipantIndex>120</MaxAutoParticipantIndex>
    </Discovery>
  </Domain>
</CycloneDDS>

```

closed 2024-01-12 by SimonGiampy

Samahu, 2024-01-12:
@SimonGiampy thanks for the feedback..definitely a helpful insight that you provided! Yes, it is unfortunate but the driver performance is very much dependent on the underlying RMW/DDS and the machine you use, otherwise I would got the same performance on my development machine (and I would believe many of the driver users).

I do understand that you have a working solution but with regard to "pointcloud messages being heavy and occupy a lot of bandwidth". That's absolutely true, Users/Developers who are still struggling with the bandwidth overuse could try one or more the following:
* Choose a lower rate or lower size LIDAR_MODE (i.e choose 512x10 instead of 1024x10 or 1024x10 instead of 1024x20)
* Optimize the type of the published point cloud through `point_type` launch file parameter. For example if in your use only care about xyz values and the intensity then you could opt for `point_type=xyzi or xyzir`, which would significantly bring down the required bandwidth (almost 75% reduction). Another option is to choose `point_type=native` and set the sensor `udp_lidar_profile` to the low data mode `RNG15_RFL8_NIR8` which will also bring down the required bandwidth by about 33% (this profile doesn't have intensity however). For more information consult [Point Cloud Customization](https://github.com/ouster-lidar/ouster-ros/discussions/260).