ouster-lidar/ouster-ros #280: Point Cloud Recording Issue: Frame Skipping and Reduced Frequency During Rosbag Playback

Risbo6, 2024-01-18:
Hello,

While recording a point cloud with a rosbag, I've noticed that some frames are being skipped. I'm uncertain if this issue is related to quality of service settings, as I'm still relatively new to ROS2.

When I launch the driver and check the frequency with ros2 topic hz, I observe a rate of around 10 Hz. However, during the playback of the rosbag, the frequency drops to approximately 5-6 Hz.

I'm using an OS-0-32 on ROS2 Foxy, and I've tested it on both arm and x64 architectures.

Samahu, 2024-01-18:
Are you using the provided `record.launch.xml` launch file to capture the raw packet or are you using recording the `/ouster/points(s)` topics through rosbag functionality.

Julien2109, 2024-01-26:
Hi,

We tried both options, but the result is similar. You can find an example recorded with the provided  `record.launch.xml`, and another recorded through the rosbag functionality on this google drive: [https://drive.google.com/drive/folders/1CDsDUO-_0VUns5zw2WFIlMCnPdpe9jUW?usp=sharing](url).

Samahu, 2024-01-26:
Which ROS2 distro are you targeting?

Julien2109, 2024-01-26:
We are testing with foxy, and we tried to replay the data on both foxy and humble.

Samahu, 2024-01-27:
@Julien2109 I have confirmed the problem, I can see the point cloud blinking on RVIZ

[image]

and I did see the point cloud runs at half the frame rate. If you used the provided `record.launch.xml` (for the raw record) then the launch script automatically sets the proper QoS for record/replay scenario. This would eliminate the theory that the issue is due to QoS.

Do you observe this problem during live mode? Or does it only happen when it involves rosbag?

I can't produce the same problem on the computer I use for development. My intuition tells me this boils down to the underlying computer arch and dds middleware used. However you tried a different RMW?

Julien2109, 2024-01-27:
During live mode, it seems that it is not happening. I couldn't verify on RVIZ, as our target has no screen, but the topic rate was arround 10Hz in live mode as well as in recording mode. The rate is droping only when replaying the data (I tried to replay on same target and on another computer after having exported the files).  

We also tried to record the data from another computer and it was working well (also with foxy). So yes, it is possible that the problem is coming from the target...

Risbo6, 2024-01-29:
We're using a jetson AGX Xavier. The issue may indeed comes from the ARM architecture. I will try to connect a laptop to the network, play the pointcloud with the jetson and record it with the laptop to confirm this.

Most of our other vehicles are still running on ROS 1, and we never had any issue with a rosbag. So far, I've only had problems with ROS 2. I hope we can find a solution.

Samahu, 2024-01-29:
Since you mentioned Jetson AGX I can confirm that other users of our driver have indeed experienced some partial or complete frame drops (checkout out https://github.com/ouster-lidar/ouster-ros/issues/240). I do think it the issue applies to all ARM processor running with the ROS driver in general and is not specific to the NVIDIA Jetsons. The issue exist even with the latest updates done to the driver and needs a closer look.