ouster-lidar/ouster-ros #281: Build error on Humble, Ubuntu 22.04

johanneswiig, 2024-01-19:
**Describe your question**
I created a new blank ros2_ws and tried to build it according to instructions on ros2 humble on ubuntu 22.04,  cmake version 3.22.1. The ouster msgs package build without errors, however, the ouster_ros fails to build. Have you been able to build this on a similar setup? Do you have any advice to resolve the below errors? Please see the output below:

**Platform (please complete the following information):**
- Ouster Sensor? \[OS-1]
- Ouster Firmware Version? \[2.4]
- ROS version/distro? \[humble]
- Operating System? \[Linux]
- Machine Architecture? \[x64]

- Build error message:

trust@trust-nuc:~/ouster/ros2_ws$ source /opt/ros/humble/setup.bash
trust@trust-nuc:~/ouster/ros2_ws$ colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> ouster_sensor_msgs
Finished <<< ouster_sensor_msgs [4.95s]                     
Starting >>> ouster_ros
[Processing: ouster_ros]                             
[Processing: ouster_ros]                                     
--- stderr: ouster_ros                                         
/home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/ouster-sdk/ouster_client/src/parsing.cpp: In member function ‘uint32_t ouster::sensor::packet_format::frame_id(const uint8_t*) const’:
/home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/ouster-sdk/ouster_client/src/parsing.cpp:443:28: warning: ‘uint16_t ouster::sensor::packet_format::col_frame_id(const uint8_t*) const’ is deprecated: Use frame_id instead [-Wdeprecated-declarations]
  443 |         return col_frame_id(nth_col(0, lidar_buf));
      |                ^~~~~~~~~~~
In file included from /home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/ouster-sdk/ouster_client/include/ouster/impl/packet_writer.h:8,
                 from /home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/ouster-sdk/ouster_client/src/parsing.cpp:18:
/home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/ouster-sdk/ouster_client/include/ouster/types.h:1102:53: note: declared here
 1102 |     [[deprecated("Use frame_id instead")]] uint16_t col_frame_id(
      |                                                     ^~~~~~~~~~~~
In file included from /home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/src/os_ros.cpp:17:
/opt/ros/humble/include/tf2_eigen/tf2_eigen/tf2_eigen.h:32:2: warning: #warning This header is obsolete, please include tf2_eigen/tf2_eigen.hpp instead [-Wcpp]
   32 | #warning This header is obsolete, please include tf2_eigen/tf2_eigen.hpp instead
      |  ^~~~~~~
/home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/src/os_replay_node.cpp: In member function ‘void ouster_ros::OusterReplay::declare_parameters()’:
/home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/src/os_replay_node.cpp:99:50: error: no matching function for call to ‘ouster_ros::OusterReplay::declare_parameter(const char [9])’
   99 |     void declare_parameters() { declare_parameter("metadata"); }
      |                                 ~^~
In file included from /home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/include/ouster_ros/os_sensor_node_base.h:13,
                 from /home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/src/os_replay_node.cpp:10:
/opt/ros/humble/include/rclcpp_lifecycle/rclcpp_lifecycle/lifecycle_node.hpp:342:3: note: candidate: ‘template<class ParameterT> auto rclcpp_lifecycle::LifecycleNode::declare_parameter(const string&, const ParameterT&, const ParameterDescriptor&, bool)’
  342 |   declare_parameter(
      |   ^~~~~~~~~~~
/opt/ros/humble/include/rclcpp_lifecycle/rclcpp_lifecycle/lifecycle_node.hpp:342:3: note:   template argument deduction/substitution failed:
/home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/src/os_replay_node.cpp:99:50: note:   candidate expects 4 arguments, 1 provided
   99 |     void declare_parameters() { declare_parameter("metadata"); }
      |                                 ~^~
In file included from /home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/include/ouster_ros/os_sensor_node_base.h:13,
                 from /home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/src/os_replay_node.cpp:10:
/opt/ros/humble/include/rclcpp_lifecycle/rclcpp_lifecycle/lifecycle_node.hpp:355:3: note: candidate: ‘template<class ParameterT> auto rclcpp_lifecycle::LifecycleNode::declare_parameter(const string&, const ParameterDescriptor&, bool)’
  355 |   declare_parameter(
      |   ^~~~~~~~~~~
/opt/ros/humble/include/rclcpp_lifecycle/rclcpp_lifecycle/lifecycle_node.hpp:355:3: note:   template argument deduction/substitution failed:
/home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/src/os_replay_node.cpp:99:50: note:   couldn’t deduce template parameter ‘ParameterT’
   99 |     void declare_parameters() { declare_parameter("metadata"); }
      |                                 ~^~
In file included from /home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/include/ouster_ros/os_sensor_node_base.h:13,
                 from /home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/src/os_replay_node.cpp:10:
/opt/ros/humble/include/rclcpp_lifecycle/rclcpp_lifecycle/lifecycle_node.hpp:316:3: note: candidate: ‘const rclcpp::ParameterValue& rclcpp_lifecycle::LifecycleNode::declare_parameter(const string&, const rclcpp::ParameterValue&, const ParameterDescriptor&, bool)’
  316 |   declare_parameter(
      |   ^~~~~~~~~~~
/opt/ros/humble/include/rclcpp_lifecycle/rclcpp_lifecycle/lifecycle_node.hpp:316:3: note:   candidate expects 4 arguments, 1 provided
/opt/ros/humble/include/rclcpp_lifecycle/rclcpp_lifecycle/lifecycle_node.hpp:329:3: note: candidate: ‘const rclcpp::ParameterValue& rclcpp_lifecycle::LifecycleNode::declare_parameter(const string&, rclcpp::ParameterType, const ParameterDescriptor&, bool)’
  329 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/opt/ros/humble/include/rclcpp_lifecycle/rclcpp_lifecycle/lifecycle_node.hpp:329:3: note:   candidate expects 4 arguments, 1 provided
In file included from /home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/src/os_ros.cpp:17:
/opt/ros/humble/include/tf2_eigen/tf2_eigen/tf2_eigen.h:32:2: warning: #warning This header is obsolete, please include tf2_eigen/tf2_eigen.hpp instead [-Wcpp]
   32 | #warning This header is obsolete, please include tf2_eigen/tf2_eigen.hpp instead
      |  ^~~~~~~
gmake[2]: *** [CMakeFiles/os_replay_component.dir/build.make:90: CMakeFiles/os_replay_component.dir/src/os_replay_node.cpp.o] Error 1
[... 86 lines cut ...]
  233 |         if (!(scan_batcher)(lidar_buf, lidar_scan)) return false;
      |              ~^~~~~~~~~~
In file included from /home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/include/ouster_ros/os_ros.h:16,
                 from /home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/src/os_driver_node.cpp:13:
/home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/ouster-sdk/ouster_client/include/ouster/lidar_scan.h:532:5: note: declared here
  532 |     operator()(const uint8_t packet_buf, LidarScan& ls);
      |     ^~~~~~~~
In file included from /home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/src/os_driver_node.cpp:20:
/home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/src/lidar_packet_handler.h: In member function ‘bool ouster_ros::LidarPacketHandler::lidar_handler_ros_time(const ouster::sensor::packet_format&, const uint8_t)’:
/home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/src/lidar_packet_handler.h:250:29: warning: ‘bool ouster::ScanBatcher::operator()(const uint8_t*, ouster::LidarScan&)’ is deprecated: Use ScanBatcher::operator(LidarPacket&, LidarScan&) instead. [-Wdeprecated-declarations]
  250 |         if (!(*scan_batcher)(lidar_buf, lidar_scan)) return false;
      |              ~^~~~~~~~~~
In file included from /home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/include/ouster_ros/os_ros.h:16,
                 from /home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/src/os_driver_node.cpp:13:
/home/trust/ouster/ros2_ws/src/ouster-ros/ouster-ros/ouster-sdk/ouster_client/include/ouster/lidar_scan.h:532:5: note: declared here
  532 |     operator()(const uint8_t packet_buf, LidarScan& ls);
      |     ^~~~~~~~
gmake: *** [Makefile:166: all] Error 2

Failed   <<< ouster_ros [1min 14s, exited with code 2]

Summary: 1 package finished [1min 19s]
  1 package failed: ouster_ros
  1 package had stderr output: ouster_ros
trust@trust-nuc:~/ouster/ros2_ws$

Samahu, 2024-01-19:
@johanneswiig The build is failing due to the use of incompatible API `declare_parameter` for the current ROS version that you have sourced.

> Have you been able to build this on a similar setup?

The last build status passes on humble with no issues, please check out: https://github.com/ouster-lidar/ouster-ros/actions/runs/7038088158/job/19154264195
This build pulls a docker image based of ROS2/humble, so I don't see an issue in the code

> Do you have any advice to resolve the below errors?

* If you are building the driver on ros2 humble double check that you checked out the `ros2` branch and not the `ros2-foxy`.
* double check that you are sourcing only ros2 humble... sometimes if you have more that one ROS2 environment installed one of them maybe sourced automatically (double check your `.bashrc` for ros commands added)
* Install the latest updates for ROS2 setup.

Samahu, 2024-01-23:
Yes, warnings are there unfortunately but they are mostly due to deprecation within the client code. I want to address them in a better ways in the future but for the time being I recommend that you silence them via adding the flag "-DCMAKE_CXX_FLAGS="-Wno-deprecated-declarations -Wno-deprecated" to your colcon build.

closed 2024-01-23 by Samahu