ouster-lidar/ouster-ros #283: Colcon build failed on Ubuntu 22.04 ROS2 Humble

kushal2507, 2024-01-29:
**Describe your question**
I am unable to successfully build the package on ros2 humble, Ubuntu 22.04. I have installed the dependencies as listed on the page and also ran rosdep install. I have also run apt update before 

Here are the steps I am undertaking, hopefully someone can reproduce the problem : 

- cd ros2_ws/src
- git clone --recurse-submodules https://github.com/ouster-lidar/ouster-ros.git
- git checkout ros2
- colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release

Error attached below

**Screenshots**
[image]

**Platform (please complete the following information):**
- ROS version/distro \[humble]
- Operating System \[Ubuntu 22.04\]

kushal2507, 2024-01-29:
I added the dependency libglfw3 by running 

`apt install libglfw3-dev`

But now I am getting the following error : 

[image]

hehonglu123, 2024-01-29:
Just encountered the same problem, make sure you have the `ouster-sdk` [submodule](https://github.com/ouster-lidar/ouster_example/tree/2898060fb47a69b4fbf9807d3c17f6fdc8249379)  2898060 at `<ws>/src/ouster_ros/ouster_ros/ouster_sdk/`.

The full build command from Action is `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-Wno-deprecated-declarations"`

kushal2507, 2024-01-29:
@hehonglu123 thank you for your response. I was able to resolve this sometime back using the same.

closed 2024-01-29 by kushal2507