ouster-lidar/ouster-ros #287: Ouster support for Apollo?

daohu527, 2024-02-04:
**Is your feature request related to a problem? Please describe.**
We want to support ouster for Apollo. 

**Describe the solution you'd like**
We hope that Ouster will provide an SDK, and Apollo will obtain lidar point clouds through interface calls.
Of course there may be more details, such as some configuration files, etc.

**Describe alternatives you've considered**

**Targeted Platform (please complete the following information only if applicable, otherwise dot N/A):**
- Ouster Sensor? \[feature only applicable to specific ouster sensor\]
- Ouster Firmware Version? \[feature only applicable to specific fw version\]
- Operating System? \[Ubuntu 18.04]
- Machine Architecture? \[x86, arm]

Samahu, 2024-02-05:
Hi @daohu527, thanks for contacting. I haven't looked very closely into Apollo but i I may ask what kind of support you are looking for? What does Apollo require more than the published PointCloud to utilize our Sensors? Like a specific PointCloud format or is it a little more than that?

daohu527, 2024-02-06:
Many people in the community need to use Ouster lidar, so we want to integrate Ouster directly into Apollo. 

What we expect is that Ouster can provide an SDK to obtain point clouds, and then Apollo will call the API and adapt to Apollo's [point cloud message format](https://github.com/ApolloAuto/apollo/blob/master/modules/common_msgs/sensor_msgs/pointcloud.proto). 

This is very similar to ROS, but each lidar may have some special configurations, so if possible we can write a design document, then I will push forward and integrate the ouster lidar as soon as possible, of course there are better suggestions we can discuss here.

daohu527, 2024-02-11:
@Samahu Thank you for your quick solution! I'm on vacation recently, I will give you a plan on February 19th

daohu527, 2024-02-18:
@Samahu I checked your pr, but the difference is that Apollo uses cyber instead of ros, so it needs to be redesigned.

I write an design [doc](https://docs.google.com/document/d/1gAE07kdTcXDqUBaCzLJsanPv_cY4UtzDkz5mAIc3Agg/edit?usp=sharing) about it. To put it simply, lidar sdk provides a **callback interface** for Apollo registration, and provides **data format meta info** for raw data. I currently use the PCD format, and then the apollo lidar driver will register the callback and convert the data into Apollo format.

Looking forward to your reply

Samahu, 2024-02-21:
Hi @daohu527 the framework in question is not very popular on my radar. I think for such task that the team behind Apollo's could utilize [the main Ouster SDK](https://static.ouster.dev/sdk-docs/index.html) to carry out the integration.

closed 2024-02-21 by Samahu

daohu527, 2024-02-22:
Thank you for your reply! Can you tell me what aspects you consider? In fact, each Lidar API only needs to provide registration callbacks.

You know we can refer to the current framework for integration, but we would like to get official support so that during the upgrade and iteration process, the quality of these drivers will be more guaranteed. **We can do integrated development if possible**, but I would like your support, such as reviewing the code and subsequent updates.