ouster-lidar/ouster-ros #288: ROS2 crashes when standby mode is set and then set to normal

mtswisher, 2024-02-07:
**Describe the bug**
The ROS2 driver will crash if the service /ouster/set_config is called with OPERATING_MODE: STANDBY
The nodelet will no longer receive additional queries and issuing a SIGKILL to the operating shell exits with additional error codes.

**To Reproduce**
Steps to reproduce the behavior (steps below are just an example):
1. Setup working ROS2 environment from README
2. `ros2 launch ouster_ros sensor.launch.xml sensor_hostname:=os-122319000734.local` launch node
3. Validate node is publishing data through RVIZ
4. Issue the following service command: ros2 service call /ouster/set_config ouster_sensor_msgs/srv/SetConfig "{config_file: 'test
.json'}" [test_3.json](https://github.com/ouster-lidar/ouster-ros/files/14187644/test_3.json) where test.json contains operating mode **STANDBY**
[image]

5. Issue the following service command: ros2 service call /ouster/set_config ouster_sensor_msgs/srv/SetConfig "{config_file: 'test
.json'}" [test_3.json](https://github.com/ouster-lidar/ouster-ros/files/14187644/test_3.json) where test.json contains operating mode **NORMAL**

6. Query a service for a response: 
[image]
This query will hang indefinitely

7. Issuing SIGTERM to the original node fails with the following error: [ERROR] [os_driver-1]: process[os_driver-1] failed to terminate '10.0' seconds after receiving 'SIGTERM', escalating to 'SIGKILL'
[image]

- Ouster Sensor? OS1
- Ouster Firmware Version? 3.0.1
- ROS version/distro?humble
- Operating System? Ubuntu 22.04
- Machine Architecture? x86x64
- git commit hash: commit 66d6b9ff104693c1b24397dd0b66d67bd0c3d4e4

Samahu, 2024-02-08:
@mtswisher thank you for the detailed report. I have identified the root cause of the problem and will submit a fix soon.

Samahu, 2024-02-09:
Resolved by #292 and #293

PR #293 (2024-02-09): ROS-288: ROS2 crashes when standby mode is set and then set to normal [HUMBLE]
## Related Issues & PRs
- foxy PR: #292
- closes: #288

## Summary of Changes
- Use timeout when waiting for packets to be proceed in case no packets are coming

## Validation
- driver functionality sustained
- launch a ros2 node then perform a set_config call with STANDBY mode, see the driver idles
- perform a follow set_config call this time with NORMAL mode and verify that driver resumes
closed 2024-02-09 by Samahu