ouster-lidar/ouster-ros #289: PTP Time is not integrated into the IMU Topic Header

GeneSysTobiasWagner, 2024-02-08:
**Question**
I am synchronizing the Ouster OS-1 with a GNSS System of us. The OS-1 Status information tells me it is synchronized as Slave and it looks like the sensor is synching to the Master. 

I am launching the driver with "ros2 launch ouster_ros sensor.launch.xml sensor_hostename:=<IP>"

My sensor.launch.xml is configured with <arg name="timestamp_mode" default = "TIME_FROM_PTP_1588" . Somehow the header time stamp is still starting to count from zero. When i configure TIME_FROM_ROS instead, the time stamps look valid to the current time. 

I am using the gptp profile. 

This is the /api/v1/time/ output :
system	
realtime	1266876336.099259
monotonic	1503.525721687
tracking	
frequency	534.371
root_delay	1e-9
rms_offset	0.000001705
leap_status	"normal"
system_time_offset	-1e-9
skew	0.103
last_offset	3.13e-7
remote_host	"ptp"
root_dispersion	0.002236648
stratum	1
residual_frequency	0.156
reference_id	"70747000"
ref_time_utc	1266874897.2536337
update_interval	2
sensor	
timestamp	
mode	"TIME_FROM_PTP_1588"
time	1483.394498112
time_options	
internal_osc	1492
ptp_1588	1483
sync_pulse_in	1492
multipurpose_io	
sync_pulse_out	
pulse_width_ms	10
angle_deg	360
polarity	"ACTIVE_HIGH"
frequency_hz	1
nmea	
polarity	"ACTIVE_HIGH"
diagnostics	
decoding	
date_decoded_count	0
utc_decoded_count	0
not_valid_count	0
last_read_message	""
io_checks	
start_char_count	0
char_count	0
bit_count_unfiltered	1
bit_count	1
baud_rate	"BAUD_9600"
locked	0
ignore_valid_char	0
leap_seconds	37
mode	"INPUT_NMEA_UART"
sync_pulse_in	
polarity	"ACTIVE_HIGH"
locked	0
diagnostics	
last_period_nsec	0
count	0
count_unfiltered	1
ptp	
current_data_set	
offset_from_master	1483056887613
mean_path_delay	0
steps_removed	1
parent_data_set	
grandmaster_identity	"42d855.fffe.0ef000"
parent_stats	0
observed_parent_offset_scaled_log_variance	65535
grandmaster_priority2	248
gm_offset_scaled_log_variance	17258
grandmaster_priority1	248
parent_port_identity	"42d855.fffe.0ef000-1"
observed_parent_clock_phase_change_rate	2147483647
gm_clock_accuracy	35
gm_clock_class	248
time_status_np	
ingress_time	1483181920861
cumulative_scaled_rate_offset	-0.000976562
scaled_last_gm_phase_change	0
gm_present	true
master_offset	1483056887613
last_gm_phase_change	"0x0000'0000000000000000.0000"
gm_time_base_indicator	0
gm_identity	"42d855.fffe.0ef000"
port_data_set	
port_identity	"bc0fa7.fffe.001d36-1"
log_min_delay_req_interval	0
peer_mean_path_delay	-28814
log_announce_interval	0
log_min_pdelay_req_interval	0
port_state	"SLAVE"
delay_mechanism	2
announce_receipt_timeout	3
version_number	2
log_sync_interval	-3
time_properties_data_set	
leap61	0
leap59	0
ptp_timescale	1
time_source	160
time_traceable	1
current_utc_offset	37
frequency_traceable	0
current_utc_offset_valid	0
profile	"gptp"

**Platform:**
- Model: OS-1
- Firmware: v2.5.3
- ROS Distro: Humble
- System: Ubuntu 20.04
- Machine structure: x64

GeneSysTobiasWagner, 2024-02-14:
@Samahu Thanks for the feedback. I reset the complete ROS Driver Repo to the last commit, changed nothing in the configuration files and executed the exact same command with the IP of my lidar. Still, the Lidar says he is synced as "slave" and in the headers of the ROS topics, i got 600 sec instead of 1707913480. 

When im starting the driver with TIME_FROM_ROS_TIME instead, the time stamps are correct...

Samahu, 2024-02-14:
Can you examine the point cloud topic and compare it to the imu topic, do they show same or different times when you alter the timestamp mode?

GeneSysTobiasWagner, 2024-02-15:
Nevermind, somehow the PTP seems not to work correctly at the Lidar even if he shows "Slave" state. Thank you for your thoughts anyways!

closed 2024-02-15 by GeneSysTobiasWagner