ouster-lidar/ouster-ros #291: Obtaining an organized point cloud from OS0 sensor

erminkev1, 2024-02-08:
Hello! 

I would like to preface my question by saying that I am somewhat of a beginner in ROS, and that I hope you can assist me here.

In one of my ROS nodes, I am processing the point cloud coming from an OS0 sensor, and in one part I am using a functionality of PCL library to generate a _spherical_ range image, however it is very slow. The method in question is _pcl::RangeImage::createFromPointCloud_. Now, according_ to one discussion on their GitHub, if the point cloud is "organized", meaning in a grid of HxW, then the algorithm can process it faster, however when I read the H of the ouster point cloud I get **1**.

May you please provide me directions or hints whether it is possible to obtain the grid-like representation of a point cloud so that I can use it in my processing pipeline? 

Best,
Ermin

Samahu, 2024-02-08:
Hi Ermin, The point cloud that by ouster-ros driver is organized. You can double check that by echoing the `/ouster/points` topic and verify that both width and height are set properly. For example when I examine the header on my machine I get the following:
```
header: 
  seq: 1350
  stamp: 
    secs: 1707414371
    nsecs: 853034304
  frame_id: "os_sensor"
height: 128
width: 1024
```
 I am not sure how you are getting a PointCloud with a height of 1. It might be that you are invoking a PCL method in your pipeline that flattens our organized point cloud.

erminkev1, 2024-02-08:
Thank you for your quick response :) 

I thought so too. Inspecting my bagfile in FoxGlove I also see that height and width are as such. I can also see that the message type is sensor_msgs:PointCloud2.

Therefore, in my callback function arguments, I define it as follows:

`void callback(_const boost::shared_ptr<const sensor_msgs::PointCloud2>& in_pc2_ , const ImageConstPtr& in_image) {stuff}`

and once I inspect the values of height or width it via:

```
  if (in_pc2->height > 1) {
      // The point cloud is organized
      std::cout << "Received an organized point cloud." << std::endl;
  } else {
      // The point cloud is unorganized
      std::cout << "Received an unorganized point cloud." << std::endl;
  }
```

the output is "Received an unorganized pc"

Best,
Ermin :)

Samahu, 2024-02-08:
Ermin, I am not sure how or why you are getting a point cloud with a height of one. This definitely something has to do with part of your pipeline. I see that your callback has an image as the second parameter which suggests that you are passing the point cloud that the driver publish through some sort of a filter first. This filter that you are using maybe the one responsible for changing the structure of the point cloud. Do you get this issue when you directly subscribe to the `/ouster/points` topic? Are you applying any filter to the points before you examine the point cloud?

Samahu, 2024-02-08:
Ermin, I am afraid the issue you are experiencing has nothing to do with the driver itself but your point cloud processing pipeline. I am going to mark this with "no fix" and close the issue. You could still follow on with further questions if you still think the issue has to do with the ouster-ros driver itself but I am trying to narrow down the list of issues that require any action or investigation on our part.

closed 2024-02-08 by Samahu

erminkev1, 2024-02-08:
Thank you for your prompt responses @Samahu. I will keep testing and if the issue persists then I will provide in detail on how to reproduce it. Otherwise consider it solved :) 

p.s the second argument is just an Image message coming from a camera stream and doesn't affect the point cloud.