ouster-lidar/ouster-ros #294: Unexpected udp packet length

gauravkuppa, 2024-02-12:
### **Describe the bug**
Unexpected udp packet length: 49

**To Reproduce**
We have tried many different things to consistently reproduce this bug, but have not been able to identify any reliable procedure to find this bug.

**Screenshots**
```
[os_driver-1] [INFO] [1707685258.160491401] [os_cloud_node.os_driver]: on_activate() is called.
[os_driver-1] [2024-02-11 16:00:58.226] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.227] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.227] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.227] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.227] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.227] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.227] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.227] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.227] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.228] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.228] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.228] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.228] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.229] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.229] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.229] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.230] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.230] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.230] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.231] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.231] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.231] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.231] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.231] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.231] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.231] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.231] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.231] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.231] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.232] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.275] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.275] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.275] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.276] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.276] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.276] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.276] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.276] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.277] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.277] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.277] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.277] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.277] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.277] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.277] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.277] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:00:58.278] [ouster::sensor] [warning] Unexpected udp packet length: 49
[lidar_obstacle_map_node-1] [INFO] [1707685259.080868699] []: ----> Publisher Status Thread Started.
[lidar_obstacle_map_node-1] [INFO] [1707685259.282003737] []: ----> Publisher Status Thread Started.
[os_driver-1] [2024-02-11 16:00:59.286] [ouster::sensor] [warning] Unexpected udp packet length: 49
[lidar_obstacle_map_node-1] [WARN] [1707685259.484422288] []: Started Using Ouster
[lidar_obstacle_map_node-1] [INFO] [1707685259.494672701] []: ----> Obs Map Thread Started.
[lidar_obstacle_map_node-1] [INFO] [1707685260.525952018] []: new path recieved size: 15 ind: 0
[mpc_node-1] [WARN] [1707685260.540535842] [mpc_gps_obs_node]: Got valid points
[os_driver-1] [2024-02-11 16:01:01.444] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:01:02.222] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:01:03.230] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:01:03.254] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:01:03.305] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:01:03.469] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:01:03.686] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:01:03.853] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:01:03.857] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:01:04.109] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:01:04.251] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:01:04.308] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [2024-02-11 16:01:04.391] [ouster::sensor] [warning] Unexpected udp packet length: 49
[os_driver-1] [ERROR] [1707685264.391222819] [os_cloud_node.os_driver]: maximum number of allowed errors from sensor::read_imu_packet() reached, reactivating...
[os_driver-1] [INFO] [1707685264.393500073] [os_cloud_node.os_driver]: on_deactivate() is called.
[os_driver-1] [INFO] [1707685264.416670877] [os_cloud_node.os_driver]: on_activate() is called.
```
**Platform (please complete the following information):**
- Ouster Sensor? OS-1-32
- Ouster Firmware Version? Rev7
- ROS version/distro? Galactic
- Operating System? Ubuntu
- Machine Architecture? x86
- git commit hash (if not the latest).

Samahu, 2024-02-12:
This isn't a real bug but I presume you have more than one sensor or other devices on the network that might be using the same port?

Try to initialize the targeted sensor with a different port or just pass zeros and the ouster-driver will pick random ports for you.

closed 2024-02-12 by Samahu

gauravkuppa, 2024-02-12:
We have this Ouster connected directly to the computer, without any other sensors sharing this ethernet connection or network/subnet. The port is already initialized to 0. Would it be better to assign a specific port here or is the standard behavior of `lidar_port: 0` preferred?

Samahu, 2024-02-12:
This isn't about the lidar port the issue is about the IMU port. You could try to assign the imu port directly to a specific value that doesn't overlap with the lidar port. Also if you don't care IMU values, you can also set the port to zero on the sensor config page this will effectively stop the IMU from sending packets.

gauravkuppa, 2024-02-13:
Is there a method to set IMU port to zero via code? The IMU port and LiDAR port are always different, by default. I am not sure why that would be interfering with lidar packets.

Samahu, 2024-02-13:
The ROS driver doesn't support this out of the box (zero port value is interpreted as assign a random port value). But you can design a function that use the `set_config` in which you set imu port to zero. Note that you can not call the function while the ROS1 driver is running. It doesn't support _Sensor Node Lifecycle Management_ as the ROS2 driver does.

Hope this helps

gauravkuppa, 2024-02-13:
This error is causing several, unpredictable issues in our system. It would be great if you could advise on what is causing the error, why it is happening, and how to fix this issue on a system where only LiDAR is connected without IMU interference. There are no other sensors, connected to this port. Your advice thus far has been predicted on the assumption on sensor interference, which does not hold in our system. I have pasted our driver_params yaml file below for your reference.

```
os_cloud_node/os_driver:
  ros__parameters:
    sensor_hostname: '192.XXX.YYY.ZZZ'
    udp_dest: ''
    mtp_dest: ''
    mtp_main: false
    lidar_mode: ''
    timestamp_mode: 'TIME_FROM_ROS_TIME'
    udp_profile_lidar: 'RNG19_RFL8_SIG16_NIR16'
    metadata: ''
    lidar_port: 0
    imu_port: 0
    sensor_frame: os_sensor
    lidar_frame: os_lidar
    imu_frame: os_imu
    point_cloud_frame: os_sensor 
    proc_mask: IMG|PCL #|IMU|SCAN
    scan_ring: 0
    use_system_default_qos: True
```

Samahu, 2024-02-14:
Try to assign to a certain value to the `lidar_port` such as the default `7502`, don't set the imu port from config but rather fill `0` in the sensor config page. Use some tool (such as wireshark) to see what is the source of packets that have 49 bytes in size.