ouster-lidar/ouster-ros #298: OS1-28 /ouster/signal_image  /camera_info

imdsafi09, 2024-02-15:
Hello,
I am attempting to utilize the OS1-128 lidar as a camera for various computer vision tasks, and I have been successfully using the "ouster/signal\_image" subscription for object detection, etc. However, in order to use this information for other applications, we require the "/camera\_info" topic which is currently missing for the ouster-128 since it is not a standard camera. My question is whether there is no standard method to generate the "/camera\_info" for the OS1-128 or any alternative solution to overcome this issue. Without this, the utility of the OS1-lidar as a camera is limited.

Thank you.

LukasBergs, 2024-06-11:
Any updates on this?

Samahu, 2024-06-12:
Hi @LukasBergs, no update yet.