ouster-lidar/ouster-ros #300: Ouster launch in docker

lidarmansiwon, 2024-02-20:
root@3391c15320e2:/home/ros2_ws# ros2 launch ouster_ros driver.launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-02-20-04-12-03-969697-3391c15320e2-424
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [os_driver-1]: process started with pid [436]
[os_driver-1] [WARN] [1708402324.263024168] [ouster.os_driver]: imu port set to zero, the client will assign a random port number!
[os_driver-1] [INFO] [1708402324.263143057] [ouster.os_driver]: Will send UDP data to 192.168.6.12
[os_driver-1] [ERROR] [1708402364.263919948] [ouster.os_driver]: exception thrown while configuring the sensor, details: CurlClient::execute_get failed for the url: [http://192.168.6.11/api/v1/system/firmware] with the error message: Timeout was reached
[ERROR] [launch_ros.actions.lifecycle_node]: Failed to make transition 'TRANSITION_CONFIGURE' for LifecycleNode '/ouster/os_driver'
[os_driver-1] [WARN] [1708402364.264021881] []: Error occurred while doing error handling.
[INFO] [launch.user]: Failed to communicate with the sensor in a timely manner.
[INFO] [os_driver-1]: sending signal 'SIGINT' to process[os_driver-1]
[os_driver-1] [INFO] [1708402365.270315596] [rclcpp]: signal_handler(signal_value=2)
[ERROR] [os_driver-1]: process has died [pid 436, exit code -11, cmd '/home/ros2_ws/install/ouster_ros/lib/ouster_ros/os_driver --ros-args -r __node:=os_driver -r __ns:=/ouster --params-file /home/ros2_ws/install/ouster_ros/share/ouster_ros/config/driver_params.yaml'].
root@3391c15320e2:/home/ros2_ws#

I'm trying to use the ouster through docker but I just can't connect to it. 

I tried this options for run docker

"docker run --rm --name os1-2 -p 7502:7502/udp -it"
"docker run [other-options] --network host your-docker-image"
"docker run -it --net=host --pid=host --ipc=host"

Ping is not delivered to sensorip and launch is not possible.

Operating system window 11
docker ubuntu version 20.04 foxy

Please Help me ,,...I really want to solve this problem

Samahu, 2024-02-26:
This seems like a problem with your docker container. Try connecting to your container and see you are able to perform:
```bash
curl http://192.168.6.11/api/v1/system/firmware
```

crazy-harry, 2024-02-28:
I'm experiencing the same exact issue while running the `ros2-foxy` branch while within a docker container. Earlier releases of this ROS driver have worked just fine within the same docker container.

I am able to send and receive  HTTP API responses, and also access the sensor homepage via web browser while inside the docker container.

bracoe, 2024-06-28:
Same problem here with iron in docker with the ROS2 branch of ouster-ros.

curl http://169.254.14.88/api/v1/system/firmware
{"fw": "ousteros-image-prod-bootes-v3.1.0+20240426041747", "commit_pending": false}

Samahu, 2024-07-08:
@crazy-harry I have just tested out that the driver works fine in docker for ros2 foxy and humble. Please make sure to pull the latest changes use `driver.launch.py` to start the driver.

lidarmansiwon, 2024-07-10:
@crazy-harry My environment is Windows. If you're using Windows the same way, you'll need to open the virtual switch through Hyper-v to set up an ooster Ethernet and bridge.

Samahu, 2024-09-13:
Please verify that you are able to ping the sensor from the container and and are able to invoke any of the sensor endpoint and get a valid response.

chadrs2, 2024-09-13:
I got it fixed. I had to run 'xhost +local:docker' as well as type in the actual up address of the LiDAR rather than os-XXXXXX.local

Samahu, 2024-09-13:
Yeah the default ‘Dockerfile’ doesn’t support dns resolution so only IP addresses work. Regarding the ‘net host’ option I will include in Caveats since this is common problem.

closed 2024-09-13 by Samahu