ouster-lidar/ouster-ros #303: How do I change the QOS in ROS2?

tom-bu, 2024-02-23:
**Describe your question**
Is there a good way to change the QOS to be best effort and a history of 1? Currently, it seems it's only editable by  changing the use_system_default_qos parameter, which I'm also not sure where it is being called. When use_system_default_qos is false, the QOS is best effort and history of 5 but a long history with point clouds may be consuming too much memory. When it's set to true, it's set to reliable and a history of 1 but reliable is not appropriate for sensor data. Currently, I have been inspecting the QOS by doing ros2 topic info -v /ouster/points. Is there a QOS parameter file that the driver loads to modify the publisher?

**Platform**
- OS-1
- ROS2 Humble
- Linux
- x64

yashmewada9618, 2024-03-04:
Hi @tom-bu,

The `use_system_default_qos` param is been called [here](https://github.com/ouster-lidar/ouster-ros/blob/3f9790713efb56099a7348aec6bdefa3e6807279/ouster-ros/src/os_cloud_node.cpp#L74). 

You can create your own rmw QoS policy like this:

```c++
static const rmw_qos_profile_t qos = {RMW_QOS_POLICY_HISTORY_KEEP_LAST,
                                          1, // depth
                                          RMW_QOS_POLICY_RELIABILITY_RELIABLE,
                                          RMW_QOS_POLICY_DURABILITY_VOLATILE,
                                          RMW_QOS_DEADLINE_DEFAULT,
                                          RMW_QOS_LIFESPAN_DEFAULT,
                                          RMW_QOS_POLICY_LIVELINESS_SYSTEM_DEFAULT,
                                          RMW_QOS_LIVELINESS_LEASE_DURATION_DEFAULT,
                                          false}; // RMW QOS policy for Ouster lidar, without this the sync won't work
auto lidar_qos = rclcpp::QoS(rclcpp::QoSInitialization::from_rmw(my_custom_qos_profile), my_custom_qos_profile);
```

Hope this helps.

Samahu, 2024-03-04:
> Is there a QOS parameter file that the driver loads to modify the publisher?

The only way you can control the QoS currently would be through the `use_system_default_qos` which can be set through the `driver_params.yaml`, which allow you to switch between reliable and sensor data. If you need more fine tune control then you need to follow the advice from @yashmewada9618 which requires you to modify the driver code.

closed 2024-03-04 by Samahu