ouster-lidar/ouster-ros #305: Unexpected UDP packet length

james-ward, 2024-02-26:
**Describe the bug**
When running the ROS2 node, at regular intervals the driver reports "Unexpected udp packet length".

**To Reproduce**
Steps to reproduce the behavior (steps below are just an example):
1. Build latest driver in release mode
2. Run driver

The size of the erroneous packets keeps repeating, and the interval is also regular to the millisecond.

**Screenshots**
```
ouster  | [os_driver-1] [2024-02-26 15:49:59.819] [ouster::sensor] [warning] Unexpected udp packet length: 364
ouster  | [os_driver-1] [2024-02-26 15:49:59.820] [ouster::sensor] [warning] Unexpected udp packet length: 360
ouster  | [os_driver-1] [2024-02-26 15:49:59.908] [ouster::sensor] [warning] Unexpected udp packet length: 356
ouster  | [os_driver-1] [2024-02-26 15:49:59.972] [ouster::sensor] [warning] Unexpected udp packet length: 356
ouster  | [os_driver-1] [2024-02-26 15:50:00.054] [ouster::sensor] [warning] Unexpected udp packet length: 372
ouster  | [os_driver-1] [2024-02-26 15:50:00.872] [ouster::sensor] [warning] Unexpected udp packet length: 380
ouster  | [os_driver-1] [2024-02-26 15:50:01.062] [ouster::sensor] [warning] Unexpected udp packet length: 356
ouster  | [os_driver-1] [2024-02-26 15:50:01.168] [ouster::sensor] [warning] Unexpected udp packet length: 356
ouster  | [os_driver-1] [2024-02-26 15:50:05.920] [ouster::sensor] [warning] Unexpected udp packet length: 356
ouster  | [os_driver-1] [2024-02-26 15:50:06.010] [ouster::sensor] [warning] Unexpected udp packet length: 360
ouster  | [os_driver-1] [2024-02-26 15:50:07.404] [ouster::sensor] [warning] Unexpected udp packet length: 364
ouster  | [os_driver-1] [2024-02-26 15:50:07.474] [ouster::sensor] [warning] Unexpected udp packet length: 372
ouster  | [os_driver-1] [2024-02-26 15:50:07.489] [ouster::sensor] [warning] Unexpected udp packet length: 356
ouster  | [os_driver-1] [2024-02-26 15:50:07.504] [ouster::sensor] [warning] Unexpected udp packet length: 356
ouster  | [os_driver-1] [2024-02-26 15:50:07.600] [ouster::sensor] [warning] Unexpected udp packet length: 380
ouster  | [os_driver-1] [2024-02-26 15:50:07.628] [ouster::sensor] [warning] Unexpected udp packet length: 372
ouster  | [os_driver-1] [2024-02-26 15:50:07.661] [ouster::sensor] [warning] Unexpected udp packet length: 372
ouster  | [os_driver-1] [2024-02-26 15:50:07.662] [ouster::sensor] [warning] Unexpected udp packet length: 356
ouster  | [os_driver-1] [2024-02-26 15:50:07.698] [ouster::sensor] [warning] Unexpected udp packet length: 380
ouster  | [os_driver-1] [2024-02-26 15:50:07.698] [ouster::sensor] [warning] Unexpected udp packet length: 380
ouster  | [os_driver-1] [2024-02-26 15:50:07.784] [ouster::sensor] [warning] Unexpected udp packet length: 384
ouster  | [os_driver-1] [2024-02-26 15:50:07.788] [ouster::sensor] [warning] Unexpected udp packet length: 356
ouster  | [os_driver-1] [2024-02-26 15:50:07.810] [ouster::sensor] [warning] Unexpected udp packet length: 360
ouster  | [os_driver-1] [2024-02-26 15:50:07.814] [ouster::sensor] [warning] Unexpected udp packet length: 364
ouster  | [os_driver-1] [2024-02-26 15:50:07.814] [ouster::sensor] [warning] Unexpected udp packet length: 360
ouster  | [os_driver-1] [2024-02-26 15:50:07.819] [ouster::sensor] [warning] Unexpected udp packet length: 364
ouster  | [os_driver-1] [2024-02-26 15:50:07.820] [ouster::sensor] [warning] Unexpected udp packet length: 360
ouster  | [os_driver-1] [2024-02-26 15:50:07.908] [ouster::sensor] [warning] Unexpected udp packet length: 356
ouster  | [os_driver-1] [2024-02-26 15:50:07.972] [ouster::sensor] [warning] Unexpected udp packet length: 356
ouster  | [os_driver-1] [2024-02-26 15:50:08.054] [ouster::sensor] [warning] Unexpected udp packet length: 372
```

**Platform (please complete the following information):**
- Ouster Sensor? OS-1
- Ouster Firmware Version? 3.0 (latest)
- ROS version/distro? rolling
- Operating System? Linux
- Machine Architecture? x64

Samahu, 2024-02-26:
I think this is mainly caused by another device sending data to the same port that you have initialized the sensor with,

Please double check what other network devices might be targeting the same port or use a different port number for lidar && imu.

james-ward, 2024-02-27:
Absolutely correct! Sorry for creating noise.

closed 2024-02-27 by james-ward