ouster-lidar/ouster-ros #309: The sensor's page won't open. How can I reset static IP?

GahyeonLim, 2024-03-07:
Hi, I'm trying to open the sensor's webpage (dashboard), but the link doesn't work. (http://os-xxxx.local)

There was also no response from Ping.
$ ping -c3 os-xxxx.local

I tried to delete the static IP on the sensor because there seems to be a problem with the network setup, but there is no response to the command below
$ curl -i -X DELETE http://192.168.100.10/api/v1/sensor/configuration

Is there a solution to reset the configuration?

I'm using OS-1, Ubuntu 22.04, and ROS2

- sensor IP(I set): 192.168.100.10 
- my computer IP: 10.130.107.120

GahyeonLim, 2024-03-11:
I solved this problem by modifying the network address.
Thank you.

GahyeonLim, 2024-03-19:
HI!
A new issue has arisen. Please let me know how to solve this problem.

The new problem is that when executing driver.launch.py, I cannot receive ROS topics, and data cannot be seen in Ouster Studio. However, ping responses and access to the sensor page are working fine. I used ROS2, Ubuntu 22.04, and OS1.
[image]

Here are the network settings:

- Sensor IP: 192.168.100.100
- Ethernet Address: 192.168.100.10

Another issue is that when testing the connection of the LiDAR to two computers with the same network settings, data is acquired from one computer, but not from the other.

Could you provide guidance on resolving this issue?

Samahu, 2024-04-16:
You probably need to configure your network connection to local-only. Please consult your sensor setup guide.

closed 2024-04-16 by Samahu