ouster-lidar/ouster-ros #312: Raw lidar packets

ajmorris1, 2024-03-13:
Hello,

In the ROS1 version of the driver, the documentation alludes to their being a topic with lidar_packets. The readme for the ROS2 version of the documentation still alludes to this, but there appears to be no topic created of this name. Is there a way to record the raw lidar packets? I am looking for the ipv4 payload for each column scan, not just the collated ranges and intensities such as in /ouster/scan.

Thanks

Samahu, 2024-03-14:
Have you check the [record mode](https://github.com/ouster-lidar/ouster-ros?tab=readme-ov-file#recording-mode). This saves the raw lidar packets. You can still get and save the raw lidar packets under the `/ouster/lidar_packets` topic but don't use the `driver.launch.xml` launch file.

closed 2024-03-14 by Samahu

ajmorris1, 2024-03-25:
Hi Samahu,
Apologies for delay in response, I have looked at the record launch file (on ros2 branch), this no longer has topics published for imu_packets and lidar_packets, there are only metadata, some converted images and points. Has this feature been removed or am I missing something?

ajmorris1, 2024-03-25:
> Have you check the [record mode](https://github.com/ouster-lidar/ouster-ros?tab=readme-ov-file#recording-mode). This saves the raw lidar packets. You can still get and save the raw lidar packets under the `/ouster/lidar_packets` topic but don't use the `driver.launch.xml` launch file.

Hi Samahu,
Apologies for delay in response, I have looked at the record launch file (on ros2 branch), this no longer has topics published for imu_packets and lidar_packets, there are only metadata, some converted images and points. Has this feature been removed or am I missing something?

ajmorris1, 2024-03-26:
> Hi Samahu, Apologies for delay in response, I have looked at the record launch file (on ros2 branch), this no longer has topics published for imu_packets and lidar_packets, there are only metadata, [...]

Hi Samahu,

My apologies, it appears as though it was my ros master being incorrectly configured so i did not see the raw packet nodes.

Thanks,
Andrew