ouster-lidar/ouster-ros #314: Recommended launch for recording and using lidar live.

finostro, 2024-03-20:
Which launch file is the recommended way to setup an lidar, to record and use the data live , at the same time

We are using os_driver, but it does not publish lidar and imu packets, is there a recommended way to record data using this node? Ideally minimizing the output bag size. 

Do we loose something by switching to os_sensor ? I understand this will enable the imu and lidar packets topics. 

**Platform (please complete the following information):**
- Ouster Sensor? 2 x OS-0, and 1 OS-1
- Ouster Firmware Version? latest for each lidar
- ROS version/distro?  humble
- Operating System? Linux, 
- Machine Architecture? \[e.g. x64, arm\]
- git commit hash (if not the latest).

skohlbr, 2024-03-23:
This has been requested multiple times ( see https://github.com/ouster-lidar/ouster-ros/issues/251 and https://github.com/ouster-lidar/ouster-ros/issues/238 ), but unfortunately the feature isn't yet available. What you lose with using the old setup is primarily efficiency, as the new single nodelet driver appears to consume much less CPU.

Samahu, 2024-04-24:
@skohlbr adding an option to record raw packets while using the driver.launch.xml launch file is still on the roadmap but it is being pushed back in favor of other higher priority items.