ouster-lidar/ouster-ros #316: Failed to load nodelet '/ouster/os_node`

ZhangYaoll, 2024-03-25:
**Describe your question**
I get something wrong when using ouster-ros to connect an Ouster Os1 snsor with commond line input 
"roslaunch ouster_ros sensor.launch sensor_hostname:='192.168.106.44' ".  It is successful when I ping the IP address. And the complete log follwed as this:
SUMMARY
========

PARAMETERS
 * /ouster/os_cloud_node/dynamic_transforms_broadcast: False
 * /ouster/os_cloud_node/dynamic_transforms_broadcast_rate: 1.0
 * /ouster/os_cloud_node/imu_frame: os_imu
 * /ouster/os_cloud_node/lidar_frame: os_lidar
 * /ouster/os_cloud_node/point_cloud_frame: 
 * /ouster/os_cloud_node/point_type: original
 * /ouster/os_cloud_node/proc_mask: IMG|PCL|IMU|SCAN
 * /ouster/os_cloud_node/ptp_utc_tai_offset: -37.0
 * /ouster/os_cloud_node/scan_ring: 0
 * /ouster/os_cloud_node/sensor_frame: os_sensor
 * /ouster/os_cloud_node/tf_prefix: 
 * /ouster/os_cloud_node/timestamp_mode: 
 * /ouster/os_node/imu_port: 0
 * /ouster/os_node/lidar_mode: 
 * /ouster/os_node/lidar_port: 0
 * /ouster/os_node/metadata: 
 * /ouster/os_node/sensor_hostname: 192.168.106.44
 * /ouster/os_node/timestamp_mode: 
 * /ouster/os_node/udp_dest: 
 * /ouster/os_node/udp_profile_lidar: 
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    rviz (rviz/rviz)
  /ouster/
    img_node (nodelet/nodelet)
    os_cloud_node (nodelet/nodelet)
    os_node (nodelet/nodelet)
    os_nodelet_mgr (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [17449]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a50cb754-ea8f-11ee-a2b1-4b862bce983a
process[rosout-1]: started with pid [17459]
started core service [/rosout]
process[ouster/os_nodelet_mgr-2]: started with pid [17466]
process[ouster/os_node-3]: started with pid [17467]
process[ouster/os_cloud_node-4]: started with pid [17468]
process[ouster/img_node-5]: started with pid [17469]
[ INFO] [1711361317.165474826]: Loading nodelet /ouster/os_node of type ouster_ros/OusterSensor to manager os_nodelet_mgr with the following remappings:
process[rviz-6]: started with pid [17471]
[ INFO] [1711361317.168102425]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] has not been advertised, waiting...
[ INFO] [1711361317.171318739]: Loading nodelet /ouster/os_cloud_node of type ouster_ros/OusterCloud to manager os_nodelet_mgr with the following remappings:
[ INFO] [1711361317.174207858]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] has not been advertised, waiting...
[ INFO] [1711361317.182204453]: Loading nodelet /ouster/img_node of type ouster_ros/OusterImage to manager os_nodelet_mgr with the following remappings:
[ INFO] [1711361317.183895600]: Initializing nodelet with 12 worker threads.
[ INFO] [1711361317.189074169]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] is now available.
[ INFO] [1711361317.195096629]: waitForService: Service [/ouster/os_nodelet_mgr/load_nodelet] is now available.
[ INFO] [1711361317.203758518]: OusterImage: node initialized!
[ WARN] [1711361317.212207464]: lidar port set to zero, the client will assign a random port number!
[ WARN] [1711361317.212262317]: imu port set to zero, the client will assign a random port number!
[ INFO] [1711361317.212279630]: Will use automatic UDP destination
[FATAL] [1711361317.381618388]: Failed to load nodelet '/ouster/os_node` of type `ouster_ros/OusterSensor` to manager `os_nodelet_mgr'
[FATAL] [1711361317.381678357]: Failed to load nodelet '/ouster/os_cloud_node` of type `ouster_ros/OusterCloud` to manager `os_nodelet_mgr'
================================================================================REQUIRED process [ouster/os_nodelet_mgr-2] has died!
process has died [pid 17466, exit code -11, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=os_nodelet_mgr __log:=/home/zhangyao/.ros/log/a50cb754-ea8f-11ee-a2b1-4b862bce983a/ouster-os_nodelet_mgr-2.log].
log file: /home/zhangyao/.ros/log/a50cb754-ea8f-11ee-a2b1-4b862bce983a/ouster-os_nodelet_mgr-2*.log
Initiating shutdown!
================================================================================
[rviz-6] killing on exit
[ouster/img_node-5] killing on exit
[ouster/os_cloud_node-4] killing on exit
[ INFO] [1711361317.568057062]: Unloading nodelet /ouster/img_node from manager os_nodelet_mgr
[ INFO] [1711361317.569045381]: waitForService: Service [/ouster/os_nodelet_mgr/unload_nodelet] could not connect to host [zhangyao:38315], waiting...
[ WARN] [1711361317.569124743]: Couldn't find service os_nodelet_mgr/unload_nodelet, perhaps the manager is already shut down
[ouster/os_node-3] killing on exit
[ouster/os_nodelet_mgr-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

What should I do to solve such error?

**Screenshots**
[image]

**Platform (please complete the following information):**
- Ouster Sensor?  OS1 V6 128
- Ouster Firmware Version?  2.4
- ROS version/distro? noetic
- Operating System?  Linux
- Machine Architecture? x64
- git commit hash (if not the latest).

ZhangYaoll, 2024-03-27:
[image]

ZhangYaoll, 2024-03-27:
I have identified the cause of the error. It appears to be due to my usage of a virtual machine.

Samahu, 2024-03-28:
Yes, the environment is typically behind this type of issue generally speaking.

closed 2024-03-28 by Samahu

nanbwrn, 2024-08-26:
I encountered the same issue as you. I'm not using a virtual machine; instead, I'm using a Jetson Orin. Could you please advise on how to solve this?

Samahu, 2024-08-27:
Make sure you have configured your network to use local on the sensor connection and that you are able to query (using curl) the sensor page.

nanbwrn, 2024-08-28:
The radar's IP has been configured correctly. Using the curl command, I can connect to the radar. When entering the radar's static IP, 192.168.254.203, in a browser, I can also access the radar's details page, and the radar is responsive to ping as well.
Do you have any suggestions for troubleshooting the error?
[image]
[image]

Samahu, 2024-09-03:
The driver doesn't support FW prior to 2.0

Samahu, 2024-09-03:
Please head to https://ouster.com/downloads and download a more recent image of the FW that is appropriate to your sensor